DIGImend/digimend-kernel-drivers #9: H580 Pressure Sensitivity

pauloch8, 2015-01-11:
Hello again!

I have a Huion H580. You helped me at Issue #4.
I would like to use the tablet but it doesn't have pressure sensitivity.

Could you help me to providing this?

Tks

spbnick, 2015-01-11:
Hi,

Do you use the driver? Can you send the output of "/sbin/modinfo hid-huion", and output of "dmesg" after you plug it in?

pauloch8, 2015-01-11:
There is:

$ /sbin/modinfo hid-huion
filename:       /lib/modules/3.13.0-32-generic/extra/hid-huion.ko
version:        5
license:        GPL
description:    Huion HID driver
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     507D84232B8529ABBDFF3F1
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
depends:        hid
vermagic:       3.13.0-32-generic SMP mod_unload modversions

dmesg: http://pastebin.com/w4miwjrA

spbnick, 2015-01-12:
Thank you. The driver seems to be handling the tablet.

Now, could you please send the output of "sudo evtest", where you select your tablet pen device and do a stroke with the pen. There will be three devices named after your tablet, you'll need to use the one, which produces output on pen strokes.

Thank you.

pauloch8, 2015-01-12:
http://pastebin.com/7tyX82pE

Tks

spbnick, 2015-01-12:
Thank you. The kernel reports pressure alright.
Now, could you please send the output of "xinput test" for your tablet pen device, while you do a stroke?

pauloch8, 2015-01-12:
xinput test:

http://pastebin.com/MhmzP6m5

spbnick, 2015-01-12:
On 01/12/2015 12:21 PM, pauloch8 wrote:
> xinput test:
>
> http://pastebin.com/MhmzP6m5

Thank you. Yes, pressure is there too. You'll need to setup your application
to accept it. Take a look at our HOWTOs:

     http://digimend.github.io/support/

and on the web in general.

pauloch8, 2015-01-12:
Jesus! It works!
My fault. :P
Thak you very much!

spbnick, 2015-01-12:
You're welcome :)

closed 2015-01-12 by spbnick