DIGImend/digimend-kernel-drivers #10: Configuring the buttons

welblade, 2015-02-22:
I have a huion 610 tablet, and i don't like how the buttons are configured, the hand icon buttons give a ctrl+c command, the button to increse size give ctrl+v, button to back give esc, any button give the command that I expect. 

how i can change it? Any way to configure a command to each button?

like assign ctrl+[ to button with pencil+ icon?

sorry for my bad english

spbnick, 2015-02-22:
At the moment there is no way. However, Benjamin Tissoires has started work which should make the tablet better compatible with the Wacom driver, which should also make the button shortcuts configurable. I'm awaiting a pull request from him with these changes in a week or two.

BTW, your English is good :)

welblade, 2015-02-22:
I'm using linux mint that have a wacon tablet configuration tool preinstalled, that is a good new! Thanks.

welblade, 2015-03-15:
Any progress with the wacon compatibility? :)

spbnick, 2015-03-15:
Yes. I've got [patches](https://sourceforge.net/p/digimend/mailman/message/33599283/) from @bentiss changing the driver to report buttons like Wacom tablets do, waiting for my review, and he's [working](https://sourceforge.net/p/digimend/mailman/message/33599180/) on improving the Wacom bits as well.

If everything goes well we'll have something to test in the coming week.

welblade, 2015-03-16:
Cool! :)

spbnick, 2015-03-16:
Alright, the @bentiss patches are in, you can try master, if you're brave:

https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip

Now you'll *have* to use the Wacom driver to get the frame buttons working.
Then you should be able to use at least xsetwacom to set the shortcuts. The gnome applet is not likely to work ATM.

welblade, 2015-03-17:
Not work here, I try to list to find the tablet id (xsetwacom list) and nothing listed

spbnick, 2015-03-29:
Please post the output of "evtest", without dumping any device, the Xorg.conf snippet you added to configure the tablet to use the Wacom driver, the output of "xinput list" and the output of "xsetwacom --list".

Thank you.

welblade, 2015-03-29:
evtest
http://paste.ubuntu.com/10703213/

/usr/share/X11/xorg.conf.d/50-huion.conf
http://paste.ubuntu.com/10703178/

xinput list
http://paste.ubuntu.com/10703200/

xsetwacom --list output is empty

spbnick, 2015-03-29:
Thanks! OK, first the driver seems to be behaving wrongly. The event device node names shown in evtest don't show type suffixes - they should have "Pen", "Pad", etc. appended to them. That probably means that an old version is used instead did you build and install the latest master? Can you show the output of "/sbin/modinfo hid-huion" and "/sbin/modinfo hid-uclogic"?

welblade, 2015-03-29:
I build using  this file https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip

/sbin/modinfo hid-huion
http://paste.ubuntu.com/10703382/

/sbin/modinfo hid-uclogic
http://paste.ubuntu.com/1070339

spbnick, 2015-03-29:
Thank you. The zip file is correct. However, the modinfo output shows that stock drivers are being used instead. Could you please post the contents of /lib/modules/3.19.2-031902-generic/extra and /etc/depmod.d directories?

spbnick, 2015-03-29:
I mean just the listing of files for the start, not the files themselves.

welblade, 2015-03-29:
ls -la /lib/modules/3.19.2-031902-generic/extra
total 36
-rw-r--r-- 1 root root  9622 Mar 29 17:11 hid-kye.ko
-rw-r--r-- 1 root root  6236 Mar 29 17:11 hid-polostar.ko
-rw-r--r-- 1 root root 15107 Mar 29 17:11 hid-uclogic.ko

ls -l /etc/depmod.d 
total 8
-rw-r--r-- 1 root root 171 Mar 29 17:11 digimend.conf
-rw-r--r-- 1 root root  31 Abr 10  2014 ubuntu.conf

spbnick, 2015-03-30:
Thank you. This is correct. There might be something happening with modprobe database or rebinding script.

Which version of Ubuntu are you using?

Can you post /var/log/syslog contents after plugging in the tablet?

welblade, 2015-03-30:
I'm using Linux Mint 17.1, is based on Ubuntu 14.04

I decided to redo all process from the beginning, sudo rmmod hid-huion, make uninstall, download master again, make , make install, restart. Now I think it worked.

xinput list
http://paste.ubuntu.com/10707222/

xsetwacom --list
http://paste.ubuntu.com/10707341/

/sbin/modinfo hid-uclogic
http://paste.ubuntu.com/10707241/

/var/log/syslog 
http://paste.ubuntu.com/10707254/

spbnick, 2015-03-30:
Thanks you. No, this still doesn't look right. First of all the tablet is handled by the "hid-huion" module, judging from /var/log/syslog. It seems like rebinding isn't happening, so there might be something wrong with udev rules or with the rebinding script.

When the right driver (hid-uclogic) is used with the Wacom driver, device names in "xinput list" and "xsetwacom --list" outputs will have two qualifiers (one from the kernel driver one from xf86-input-wacom), as in "HUION PenTablet Pad pad".

spbnick, 2015-03-30:
I'll try installing Linux Mint 17 on a VM and reproducing this myself.

welblade, 2015-04-20:
@spbnick, have you some progress with that problem?

welblade, 2015-04-25:
This time I removed the files hid-uclogic.ko, hid-huion.ko and hid-kye.ko from kernel drivers directory 
In my case: */lib/modules/4.0.0-040000-generic/kernel/drivers/hid*

xinput list
http://paste.ubuntu.com/10888573/

xsetwacom --list
http://paste.ubuntu.com/10888640/

/sbin/modinfo hid-uclogic
http://paste.ubuntu.com/10888647/

/var/log/syslog 
http://paste.ubuntu.com/10888668/

spbnick, 2015-04-26:
Yep, looks correct now! You should be able to change shortcuts with "xsetwacom" now. For example, this will make the first button produce "Ctrl-Z" (Undo):

    xsetwacom --set 'HUION PenTablet Pad pad' Button 1 key +ctrl +z -z -ctrl

welblade, 2015-04-28:
OK, is working

I set to use in krita:
xsetwacom --set 'HUION PenTablet Pad pad' Button 1 "key +ctrl +z -z -ctrl"
xsetwacom --set 'HUION PenTablet Pad pad' Button 2 "key e"
xsetwacom --set 'HUION PenTablet Pad pad' Button 3 "key p"
xsetwacom --set 'HUION PenTablet Pad pad' Button 8 "button 5"
xsetwacom --set 'HUION PenTablet Pad pad' Button 9 "button 4"
xsetwacom --set 'HUION PenTablet Pad pad' Button 10 "key ]"
xsetwacom --set 'HUION PenTablet Pad pad' Button 11 "key ["
xsetwacom --set 'HUION PenTablet Pad pad' Button 12 "key m"

Perfect :)

spbnick, 2015-04-28:
Great :+1:

closed 2015-04-28 by spbnick