DIGImend/digimend-kernel-drivers #12: Non-wacom tablet monitor not working properly in multiple monitor mode

springvanillarose, 2015-03-14:
I have a Yiynova MVP22U (V3). I already installed the drivers and they do work including pressure sensitivity. However, I have to put it in mirror mode or else the cursor will not coincide with the stylus tip. Is there a way to make it work in multiple monitors without needing mirror mode? If so, please help me with that. Currenly, the Yiynova's resolution is 1920x1080, while my Dell UltraSharp U2412M is 1920x1200. I do plan to buy a third monitor as well. Thank you in advanced.

spbnick, 2015-03-15:
Please try following instructions on calibrating here: http://digimend.github.io/support/howto/drivers/evdev/#calibration

springvanillarose, 2015-03-15:
I already tried to use xinput-calibrator. But, that did not work. Somehow the device could not be found either. xinput list also detects multiple UC-Logic devices, even though I only have one.

spbnick, 2015-03-15:
Then please take a look at this section:

http://digimend.github.io/support/howto/drivers/evdev/#dual-and-multimonitor-set-up

Yes, there will be several devices. To help with identifying the pen input you can try to use the current development version of the driver:

https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip

It attempts to name the inputs appropriately and differently.

springvanillarose, 2015-04-04:
Thank you. I will try to do this. If I have more questions or the problem is solved, I will let you know.

springvanillarose, 2015-05-05:
Update: I purchased another Dell U2412M monitor. The two Dells are on the left and right, while the Yiynova tablet monitor is in the middle. I would like to restrict the cursor to the Yiynova. I'm also quite new to Linux as well.

springvanillarose, 2015-05-10:
Now that I look more closely at the instructions, I have a better understanding. But, I am now just lost on what the formula would be for my set up. Thank you in advanced.

springvanillarose, 2015-05-10:
On second thought, nevermind about the formula. I managed to figure it out on my own. I did get the cursor to restrict on the Yiynova. But another problem that happened was that the cursor does not align. I tried to use xinput-calibrator, but the points go onto the other monitors. Should I have calibrated the monitor first before restricting the cursor?

springvanillarose, 2015-05-10:
Nevermind again. I managed to solve this problem by realigning my monitors in the set up. Thank you very much, though.

closed 2015-05-10 by springvanillarose

spbnick, 2015-05-17:
If you can post instructions here, they will be very useful to other users. Thank you.

springvanillarose, 2015-05-17:
I am glad to share what I learned.  Could you also make a note that it is important to only focus on the formula that applies to the monitor the user wants to restricts their cursor to?

I looked at the formula that applied to the scenario containing two monitors with different resolutions and wondered if I could apply it to the current triple monitor formula. It turns out that I could.

Three monitors: 2 with the same resolution (1920 x 1200) - on the left and right, 1 with a different resolution (1920 x 1080) - in the center

  Left                                                 Center
[ 1920/(3*1200)   0                  0 ]   [ 1920(3*1920)   0                  1920/(3*1920) ]
[ 0                        1200/1200   0 ]   [ 0                       1080/1200   0                     ]      
[ 0                        0                  1 ]   [ 0                       0                  1                     ]

Right
[ 1920/(3*1200)   0                  (2*1920)/(3*1920) ]
[ 0                        1200/1200   0                           ]
[ 0                        0                  1                           ]

Left Monitor
xinput set-prop "Device name" --type=float "Coordinate Transformation Matrix" 0.333333 0 0 0 1 0 0 0 1

Center Monitor
xinput set-prop "Device name" --type=float "Coordinate Transformation Matrix" 0.333333 0 0.333333 0 0.9 0 0 0 1

Right Monitor
xinput set-prop "Device name" --type=float "Coordinate Transformation Matrix" 0.333333 0 0.666666 0 1 0 0 0 1

spbnick, 2015-05-18:
Thanks a lot!

ohsnapitsjoel, 2017-05-16:
I'm also a little lost on the description of the math here. Not my strong suit. I have a three-monitor setup as well: Yiynova MSP19U+ and two other monitors. My Yiynova's on the far left. Could someone kindly help me with the transformation matrix formula?  My setup would be, from left to right, 1440x900, 1920x1080, 1920x1080.

springvanillarose, 2017-05-16:
Due to your Yiynova being on the far left, you only need to worry about the formula for the left monitor. The chosen formula will restrict the cursor to that monitor.

The resolution is important. Since you monitor resolution is 1440x900, it would be this:
[1440/(3*1440)      0                 0]
[0                           900/900      0]
[0                           0                 1]

ohsnapitsjoel, 2017-05-17:
Thank you for that! I'm close to solving my alignment issues. I can't calibrate though - I try to use xinput_calibrator and it stretches across all three screens. Can anyone tell me if there's a configuration file I can manually input the calibration min_x, max_x, etc., into?

springvanillarose, 2017-05-17:
You need to put this into the terminal: xinput set-prop "Device name" --type=float "Coordinate Transformation Matrix" 0.333333 0 0 0 1 0 0 0 1

Afterward, hit enter in the terminal.

Have you checked if all of the top edges of the displays are aligned too?

But, do keep it written down because it is only temporary. I am not too sure how to make the effect permanent.

Carlito-s-Way, 2022-03-04:
Hi, is there a work around for this issue on Windows, I have a similar issue... when I try to right click with the pen the point jumps on the same yiynova screen (like if I was dragging it on the screen) and opens the options where it thinks the pointer started (jumped)... I found a "fix" that is to just let Windows, which really doesn't work as I loose the side buttons functions, and I need them obviously, I can olny have it working properly cloning the screen or just using the yiynova screen