DIGImend/digimend-kernel-drivers #13: Patchy and (really) slow strokes with H610Pro

RaphaBarros, 2015-03-28:
Hello there!

I have a H610Pro, and some time ago it was working perfectly with your driver (thx a lot!), but after some time and some updates on the Linux Kernel I decided to compile it again following the general procedures. Sadly, the input of the tablet has been acting weirdly. The right button of the pen act as the left one, when I use the tip of the pen (on my desktop or in non-painting softwares) it works like a "turbo mode" and it selects and unselects stuff non-stop, like I was clicking with it multiple times. In MyPaint it seems to have somehow recognized the pressure, but the lines become patchy. In Krita the pressure is also working ok, but the strokes become reeaaaallyy slow.

I'm on Ubuntu 14.04 64-bit, Linux 3.13.0-48-generic, using a Huion H610Pro.

Thx a lot!

spbnick, 2015-03-29:
Hi Rapha,

Can you post "dmesg" output after you plug in your tablet?
The "lsmod hid-uclogic" and "lsmod hid-huion" would be useful.

Thank you.

RaphaBarros, 2015-03-29:
Hi spbnick!

Here's the output for dmesg:

http://www.pasteall.org/57555/text

And for lsmod:

http://www.pasteall.org/57556/text

Hope it helps, thx!

spbnick, 2015-03-29:
Ah, sorry, I meant "/sbin/modinfo hid-uclogic" and "/sbin/modinfo hid-huion", instead of "lsmod ...".
Thanks!

RaphaBarros, 2015-03-30:
/sbin/modinfo hid-uclogic
filename:       /lib/modules/3.13.0-48-generic/extra/hid-uclogic.ko
version:        6
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     5AE0FCF75FF4215B49E87D5
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid
vermagic:       3.13.0-48-generic SMP mod_unload modversions 
------------------------------------------------------------------
/sbin/modinfo hid-huion
filename:       /lib/modules/3.13.0-48-generic/kernel/drivers/hid/hid-huion.ko
license:        GPL
description:    Huion HID driver
author:         Martin Rusko
srcversion:     9FAA8737687B76D5AB43A35
alias:          hid:b0003g*v0000256Cp0000006E
depends:        hid
intree:         Y
vermagic:       3.13.0-48-generic SMP mod_unload modversions 
signer:         Magrathea: Glacier signing key
sig_key:        4E:B2:DE:24:99:17:CB:F3:9C:B8:56:92:E5:4C:EB:AD:E5:94:D6:80
sig_hashalgo:   sha512

THX! :D

RaphaBarros, 2015-04-07:
Is there any other information I can give you to help with this?

Thx :)

spbnick, 2015-04-08:
Ah, yes, sorry for the delay with this. The contents of "/var/log/syslog" after you plug the tablet in would show if it was rebound to the correct driver. Thank you.

RaphaBarros, 2015-04-08:
I guess that's the part where it's plugged in:

http://www.pasteall.org/57732/text

Thx

spbnick, 2015-04-10:
Thank you. Is that all? Are there any messages from "hid-rebind"?

RaphaBarros, 2015-04-11:
Is the command complete, just hid-rebind?
If so, there's no message displayed after I do this command.

spbnick, 2015-04-12:
Thank you. There seems to be some issue with udev on Ubuntu Trusty. I'm currently investigating it.

spbnick, 2015-04-19:
Alright, I spent a few hours on this and here is a possible workaround for you while I think on how to fix this properly:

    sudo mv /lib/udev/rules.d/{70,90}-hid-rebind.rules

Execute the above, followed by this:

    sudo service udev restart

And then reconnect your tablet.

RaphaBarros, 2015-04-19:
Well, something strange happened: I tried what you said and it still didn't work, even after rebooting. So I tried uninstalling and reinstalling the driver and now it works fine!

Thx a lot. If there's any more info I can provide to help, don't hesitate to tell me! :)

spbnick, 2015-04-19:
This is strange. Can you check if /lib/udev/rules.d/90-hid-rebind.rules file is still there? Thank you.

RaphaBarros, 2015-04-19:
Yes, and here's what's inside:

SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:0000099A:00002620", \
	RUN+="/sbin/hid-rebind"
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:0000256C:0000006E", \
	RUN+="/sbin/hid-rebind"
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:00005543:*", \
	RUN+="/sbin/hid-rebind"
SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:00000458:*", \
	RUN+="/sbin/hid-rebind"

spbnick, 2015-04-19:
Ah, thank you. Can you move it somewhere else, reboot and try using your tablet?

RaphaBarros, 2015-04-21:
Sorry for the late response. Moving /lib/udev/rules.d/90-hid-rebind.rules to another place made the error come back again.

spbnick, 2015-04-21:
Ah, good. I assume moving it back fixes it then. Could you please do another test to complete verification? Please remove /lib/udev/rules.d/70-hid-rebind.rules, reboot, and check your tablet again? It should still work. Thank you.

RaphaBarros, 2015-04-21:
Should I remove it after moving /lib/udev/rules.d/90-hid-rebind.rules back to its original place, or still keep it in another folder and remove the 70-hid-rebind.rules?

spbnick, 2015-04-21:
After moving /lib/udev/rules.d/90-hid-rebind.rules back. Thank you.

RaphaBarros, 2015-04-25:
Sure, after deleting the mentioned file and rebooting it's still working :+1: 

Thx a lot!

spbnick, 2015-04-25:
Great, thank you!

commit a24eacf2 (2015-04-26): Reduce priority of udev rules

Move hid-rebind udev rules down the execution list, after the module
loading with kmod. Otherwise hid-rebind doesn't execute.

Fixes #13
closed 2015-04-26 by spbnick