DIGImend/digimend-kernel-drivers #15: Button Mapping for Huion 610 pro

happly, 2015-05-05:
This issue is very similar to the one closed issue by welblade.

the syslog showed that hid-huion was assuming control over the table and the re-bind was never occurring.

I executed `sudo update-initramfs -u` to update the drivers cache. after this when I plug in the tablet, syslog shows hid-uclogic assigned to the tablet but then is rebound to hid-huion. I got around this by just commenting out rebind.sh and everything seems to work ok now with the wacom driver.

I am not sure if this is the right way to bring this information to you, but I thought it might help in some way.

lsb_release -a
http://paste.ubuntu.com/10991423/

uname -r
`3.13.0-29-generic`

/sbin/modinfo hid-huion
http://paste.ubuntu.com/10991441/

/sbin/modinfo hid-uclogic
http://paste.ubuntu.com/10991452/

syslog before `sudo update-initramfs -u` upon plug in
http://paste.ubuntu.com/10991493/

syslog after `sudo update-initramfs -u` upon plug in
http://paste.ubuntu.com/10991498/

syslog after entirely commenting out /sbin/rebind.sh
http://paste.ubuntu.com/10991505/

sudo evtest (before)
http://paste.ubuntu.com/10991530/

sudo evtent (after)
http://paste.ubuntu.com/10991538/

spbnick, 2015-05-05:
Hello. It is likely that the reason is hid-huion from an old driver version is present in /lib/modules/3.13.0-29-generic/extra/. Please try to remove it and run "depmod -a" afterwards.

In general, please run "make uninstall" with the old driver before installing the new driver. It will try to take care of that.

happly, 2015-05-05:
Thank you for the prompt response.

Everything seem to function as expected now.

spbnick, 2015-05-06:
You're welcome!

closed 2015-05-06 by spbnick