DIGImend/digimend-kernel-drivers #17: Having some trouble with xsetwacom

treydb, 2015-05-25:
Hello! 
I'd like to give the maintainer[s] a heartfelt thanks for the work done here.

First, some info:
`modinfo hid-uclogic`

spbnick, 2015-05-25:
Hello Treydb,

The kernel driver seems to be working fine. You need to set up the Wacom X.org driver - xf86-input-wacom. You need to install the package and add a configuration file.

For the start try putting this into /etc/X11/xorg.conf.d/10-huion.conf:

    Section "InputClass"
            Identifier "huion tablet"
            MatchUSBID "256c:006e"
            MatchDevicePath "/dev/input/event*"
            Driver "wacom"
    EndSection

and then restart your X server or simply the whole machine.

treydb, 2015-05-25:
Hey spbnick.
That did the trick! Thanks for the fast response!

closed 2015-05-25 by treydb

spbnick, 2015-05-25:
Glad it worked for you :)

matteomartelli, 2016-05-21:
Hello, this worked for me too. My huion tablet Pen now it is recognized by the Wacom X.org driver. But it recognizes the Pen only, it does not recognize the other devices like the Pad and Mouse. I think that is the reason why the side buttons do not trigger any keyboard combination.
I am running arch linux with 4.5.3 kernel and kernel uc-logic drivers.

My tablet is a H610 (not Pro) but the usb vendor and product ids are the same.

`sudo lsusb -v -d 256c:006e`
http://pastebin.com/raw/YmMXe0iH

`modinfo hid-uclogic`
http://pastebin.com/raw/YZQSKmfg

`xsetwacom --list devices`
`HUION PenTablet Pen stylus      	id: 9	type: STYLUS ` 

`xinput --list`
http://pastebin.com/raw/DphNDXFy

Anyway raw input of the buttons seem working. This is the output of `sudo usbhid-dump -es -m 256c:006e` pressing each side button once from the top to the bottom:
http://pastebin.com/raw/fKXYFE9z

Any suggestion or any other test I can perform?

Thank you.

matteomartelli, 2016-05-21:
Sorry for the comment here, I have just found the open issue thread that relates to my problem:
https://github.com/DIGImend/digimend-kernel-drivers/issues/26