DIGImend/digimend-kernel-drivers #18: Huion 1060 support

benhbell, 2015-06-27:
Hi there,

I tried to get my Huion 1060Pro+ using your driver and it is working flawelessly. The only issue is that I do not see where to configure anything for it.

I believe the process would be for me to set the tablet up with the wacom driver, and then using xwacom to set the preferences/sensitivity, but is there a way to verify that anything else needs to change given my model?

Thanks

spbnick, 2015-07-01:
Looking at the tablet, I don't think there's any need to change anything. Except that perhaps not all of the frame buttons will be supported, but we can work on that.

For the start switching to xf86-input-wacom should give you much more functionality. Although the gnome tablet applet is not likely to work at the moment, xsetwacom will let you configure a lot.

ThePreviousOne, 2015-09-11:
I am running 4.1.6 kernel with Huion 1060Pro+, I got it working too... Just an FYI
Also I'll try working on a Gui for settings

RJQuiralta, 2016-03-03:
Hi Nikolai, Thank you so much for all the effort around the digimend drivers, my little mousepad like tablet worked without issues out the box, now is finishing its life and I got a 1060pro+ model to replace it. Also works just fine out the box.

As the OP, I also wonder if the buttons are re-map-able, I tried the master branch and it seems that it behaves the same. I am on a Archlinux box, up to date. I also installed the xf86-input-wacom after seeing that it was recommended in other posts, I added the 50-huion.conf file with the suggested entries also. so far still I can seem to get any configuration from it, there is no lines with "Pad pad" on the output of "xinput list" and "xsetwacom --list".

I would be glad to work this out if it is possible, just let me know what do I have to do, would be nice to add this tablet to the list, it seems to me that internally is quite similar to the h610, I got this model simple because is newer.

Thanks!
RJ Quiralta

didli, 2016-03-29:
I'm quite interested in this. Maybe I can give a hand, I've just received the Huion 1060pro+ too, working with Fedora 23. I'm trying to emulate mouse wheel scroll with two of the expresskeys, so far without success. Note : I can add a 52-tablet.conf and get the tablet recognized as a wacom device, but everything else apart from the pen/pressure stopped working. Same for the digimend-kernel-drivers. I must be missing something because while the driver installed just fine, the module is not loading with the kernel (and so the tablet stopped working ... at all).

spbnick, 2016-06-26:
Hi everyone, I already asked Sebastian on the digimend-devel maillist for some diagnostics to help support Huion 1060: https://sourceforge.net/p/digimend/mailman/message/35154103/

However, he's not reply so far, so perhaps you could help.

Could you please post the output of uclogic-probe, and after running that, without rebooting or reconnecting your tablet, also the output of usbhid-dump, where you would press each frame button in order, left-to-right, top-to-bottom?

RJQuiralta, 2016-06-26:
Here I posted some of the diagnostics, including the uclogic-probe, I am not sure if the "frame buttos" output I have there is the same as usbhid-dump, let me know if the information is of any use, or should I try again, if so just let me know. Thanks!
https://sourceforge.net/p/digimend/mailman/message/34902234/

didli, 2016-06-26:
As I'm not using the digimend driver (for now), uclogic-probe returns nothing for me (I'm guessing this particular command is related to the digimend driver ?). 
Here's the usbhid-dump output for my 1060pro+  :
[http://didli.free.fr/Divers/usbhid1060pro+](url)

RJQuiralta, 2016-08-18:
I finally got a chance to get back to this, the latest git version of the driver now works with the 1060pro+, xsetwacom now sees the Pad pad, and thus the buttons are remap-able, whether we could manage to divide the first four buttons, I don't know if will be possible, this is practically the double buttons in the tablet act as one, if I set the "'button 1" it will set the two contiguous buttons, thus the following are the numbers as I found them

| button1   | button1   |
| button2   | button2   |
| button3   | button3   |
Huion Logo/light
| button8   | button8   |
| button9   | button11 |      this and
| button10 | button12 |      this actually work individually

Is obvious that the missing numbers 4,5,6,7, correspond to the second buttons 1,2,3,4, yet xsetwacom returns
"Unsupported offset into 'Wacom Button Actions' property"

every time the 4-7 numbers are used.

Is a very good thing that most of the tablet is working now, and missing those four buttons is not a big issue in my opinion, as a matter of fact, I don't even know if they are disabled on the hardware since my tablet is the PRO+ and not the most expensive PLUS+, I just hope this lands on the kernel some time near to prevent the manual installation.

didli, 2016-08-19:
Your report is very interesting RJQuiralta. For now, I couldn't remap the buttons so I will give a try to the latest git commit. Also, as far as I know, I don't think we could divide the first four buttons : The huion website describes the tablet with "8 physical express-keys on the left and 16 hot cells" so I supposed its hardware limited. 
Finally, I'm using Fedora 24 x64, I'm going to try compilation, is there anything I should know before starting please ?

RJQuiralta, 2016-08-19:
@didli Indeed I don't think much can be done with those "extra" buttons and I don't use windows at all, thus can't tell what the "full" functionality of the tablet is.

Any way, for this to work I just followed the comments around here, practically what's been done on the 610h model, which is manually install the driver, add a huion.conf for xorg, rmmod the modules, re-connect the device.

Unfortunately I haven't manage to keep it after a reboot, it seems like the original kernel module (hid-uclogic) get loaded again and I don't know how to prevent it (need to read more on that) if I figure this out, the last step would be to add a script with all the xsetwacom commands since putting them in a xorg type of file doesn't looks like it works.

Would like to hear if you got better luck with all this.

didli, 2016-08-19:
@RJQuiralta Same problem here. I'm thinking about something like blacklisting the original driver (adding rdblacklist=hid-uclogic in grub) and loading the digimend driver via /etc/modules.conf.

RJQuiralta, 2016-08-19:
@didli After trying a few times I'm realizing that is not the original kernel hid-uclogic that is reloading, is the git driver (that works) that is not detecting the tablet correctly, thus if the tablet is unplugged, reboot, plug it, then all works normal again, but if the tablet is kept plugged upon reboot, only the stylus gets detected, then rmmod and re-plugging is needed. maybe a bug in the latest git driver? or something that hasn't been sorted out.

Note, maybe you shouldn't blacklist the module, in my case it blocks the one we're installing manually then the tablet would not be detected at all, (after a re-plug) at least that's whats happening in my box.

didli, 2016-08-20:
Hmm... I think it's the same for me with some drawbacks. I can get the driver working, but as soon as it is enable, only the pen is listed with xsetwacom (meaning I can't use the tablet at all). Well I'm not entirely sure what is going on, since it works at least once on my first try.

RJQuiralta, 2016-08-20:
This is what I think is happening, the git driver gets installed and replaces (with the same name) the one that comes with the kernel.

Now, after installing it, it should be the only one loading at boot, (you can check on /lib/modules/yourkernels/extra/) to see that there is only one set installed. there after replugging the tablet it should show the Pad device.

After rebooting the only device showing is again only the stylus. (if the tablet is kept plugged). reloading the module works though, by rmmod, then re-plugging the tablet, there the Pad shows again.

If the tablet is unplugged while the computer is off, then boot the system, and plug the tablet, the Pad would show with out rmmod the module. (kind of normal behavior) And thus the buttons can be changed.

All this makes me think that the module loads correctly only if there is no tablet connected to the computer at boot, otherwise this same module loads but if the tablet is attached on boot, it will not detect the pad, I can reproduce this all the time now. Would be interesting to see if also happens on your box the same way, if that is the case we may be up to something here and then reported as a bug.

RJQuiralta, 2016-08-27:
@spbnick @didli 
Update: I think I finally found the culprit of the last issue (lost of pad device after boot). Excuse my lack of knowledge, don't know much of what happens "under the hood" on the system even if I'm been using linux for years now. Thus I realize that after an nvidia update the module works even after reboots, I know that some type of command follows nvdia driver installation (like rebuilding the kernel modules?) but I don't know what is it exactly. Soon after that, I got a linux kernel update (Archlinux here) and as expected I had to reinstall the digimend git driver, then the same issue occurs: after a reboot the pad device is lost, I then reload the module, replug the tablet and after I make sure the pad device is available, I reinstalled nvidia, and the module once again loads correctly at boot.

I'm not sure if installing nividia process is followed by mkinitcpio or something similar, but now I know (and can reproduce it) that it makes the module able to see the pad device besides the stylus at boot.

Thus:
-Installing the digimend-kernel-driver from here (git) following the instructions line by line.
-Installing the wacom driver (xf86-input-wacom)
-Adding a xorg conf file (e.g. 50-huion.conf)
-Rebuilding the kernel image with the proper modules loaded? (just re/installing my graphic driver, in my case nvidia)
-Using a script to load the tablet buttons remaping that loads at boot.
Makes this tablet fully functional, with the advertised buttons (eight).

Hopefully this works for everybody, then we could add this model to the list and the full steps instructions to make it work fully.

spbnick, 2016-09-03:
I have to regretfully inform you, that I (the project maintainer) am [leaving the project](http://spbnick.github.io/2016/07/31/Wrapping-up-DIGImend-work.html). I'm no longer working on supporting new tablets, nor am I supporting users. I'm still accepting pull requests and work on HOWTOs and syncing with upstream, but only until November 2016. Members of the community may still help other users. Anyone desiring to step in is welcome to contact me.

giuliopaci, 2018-02-24:
I also own a Huion 1060PLUS (the box reports "New 1060PLUS", but I guess that this is not very meaningful).
The tablet is working since more than one year using the digimend drivers in git on Debian, using the dkms package: the tablet support the whole drawing area, the two buttons on the pen and pressure levels. It also support 9 out of 12 frame buttons and it is possible to remap them using xsetwatcom.

The remaining 3 buttons are not showed at all.

I wonder if there is anything that can be done in order to support the remaining 3 buttons.
I have tried to press all the buttons (in left-to-right, top-to-bottom, order) while running
```
usbhid-dump -es -m 256c:006e
```

And this is the result:

```
003:019:000:STREAM             1519476934.698648
 07 E0 01 01 01 00 00 00

003:019:000:STREAM             1519476934.848658
 07 E0 01 01 00 00 00 00

003:019:000:STREAM             1519476935.012667
 07 E0 01 01 02 00 00 00

003:019:000:STREAM             1519476935.180675
 07 E0 01 01 00 00 00 00

003:019:000:STREAM             1519476935.800627
 07 E0 01 01 04 00 00 00

003:019:000:STREAM             1519476935.934646
 07 E0 01 01 00 00 00 00

003:019:000:STREAM             1519476936.110642
 07 E0 01 01 08 00 00 00

003:019:000:STREAM             1519476936.200643
 07 E0 01 01 00 00 00 00

003:019:000:STREAM             1519476936.662659
 07 E0 01 01 10 00 00 00

003:019:000:STREAM             1519476936.804658
 07 E0 01 01 00 00 00 00

003:019:000:STREAM             1519476936.954649
 07 E0 01 01 20 00 00 00

003:019:000:STREAM             1519476937.058649
 07 E0 01 01 00 00 00 00

003:019:000:STREAM             1519476937.682642
 07 E0 01 01 40 00 00 00

003:019:000:STREAM             1519476937.816600
 07 E0 01 01 00 00 00 00

003:019:000:STREAM             1519476937.996610
 07 E0 01 01 80 00 00 00

003:019:000:STREAM             1519476938.104608
 07 E0 01 01 00 00 00 00

003:019:000:STREAM             1519476938.538622
 07 E0 01 01 00 01 00 00

003:019:000:STREAM             1519476938.676617
 07 E0 01 01 00 00 00 00

003:019:000:STREAM             1519476938.866633
 07 E0 01 01 00 02 00 00

003:019:000:STREAM             1519476938.956631
 07 E0 01 01 00 00 00 00

003:019:000:STREAM             1519476939.434625
 07 E0 01 01 00 04 00 00

003:019:000:STREAM             1519476939.524626
 07 E0 01 01 00 00 00 00

003:019:000:STREAM             1519476939.718626
 07 E0 01 01 00 08 00 00

003:019:000:STREAM             1519476939.826619
 07 E0 01 01 00 00 00 00
```

By default, when pressed, the buttons reports the following in xev:

| button1 | button2 |
| button3 | button8 |
| button9 | button10 |
Huion Logo/light
| button11 | button12 |
| button13 | not-working |
| not-working | not-working |

Thanks,
Giulio

mihi4, 2018-07-11:
I just discovered a way to use all buttons on my tablet!
While trying to build a DIY-button box I installed Antimicro to try out some button mappings.
Antimicro showed me my Huion Pad as Joystick, where all the buttons are displayed (some as axis, though)!
Nevertheless, it's now easy for me to map ALL buttons to keyboard shortcuts, so I don't use xsetwacom anymore...

greetings
michi

moshpirit, 2018-07-15:
Any news for Huion 1060Plus? this is my first graphic tablet and I have no idea how to configure it

spbnick, 2018-07-15:
Hi @moshpirit, Huion 1060Plus is fully supported. Download [v8 release](https://github.com/DIGImend/digimend-kernel-drivers/releases/v8) and follow instructions in README.md.

moshpirit, 2018-07-16:
Awesome! Thank you!

closed 2019-05-05 by spbnick