DIGImend/digimend-kernel-drivers #19: Noob help (please!)

eazup, 2015-07-16:
hi, im trying to install drivers in Linux MInt Cinnamon 17, with a '3.13.0-37-generic' kernel, im tryuing to install , but im not very well versed in Kernel. i think, i hope im just making a stupid noob mistake somewhere, but im getting an error message in terminal:

laszlo@laszlo ~/kernel-package-12.036+nmu3/kernel/pkg/source $ make-kpkg clean
We do not seem to be in a top level linux kernel source directory
tree. Since we are trying to make a kernel package, that does not make
sense.  Please change directory to a top level linux kernel source
directory, and try again. (If I am wrong, and this is indeed a top
level linux kernel source directory, then I have gotten sadly out of
date with current kernels, and you should upgrade kernel-package)

-how do i get the source into a "top-level" directory?
i tried copying into the file system folder, but permissions were denied. im not even sure thats what i should do, im just learning as i go
 thanks in advance, everybody

spbnick, 2015-07-16:
From your output you don't seem to be installing digimend-kernel-drivers at all. Have you downloaded the package and unpacked it?

eazup, 2015-07-16:
yes, as far as i can tell, i did. ive done quite a bit of work in terminal with the internets help/instruction, but i often feel like i missed an important class somewhere along the way, i find myself saying "wait, hold on- what does that even mean?" fairly frequently

[image]

eazup, 2015-07-16:
for instance- heres what im certain is a dumb rookie question- am i supposed to be typing in the green all-caps MAKE, KDIR, etc. or am i supposed to replace those with something else? or something else im not getting?
sorry for my general lack of understanding, but i NEED my Huion installed and working, and ive taken as many swings at this on my own as i can.

eazup, 2015-07-16:
[image]

spbnick, 2015-07-16:
You're not supposed to do anything except mentioned in [README.md](https://github.com/DIGImend/digimend-kernel-drivers/blob/master/README.md). Basically download, unpack, run "make" and "sudo make install" in the unpacked directory.

eazup, 2015-07-16:
here what i get when i do that. please be patient with me and let me know what im doing dumb
[image]

spbnick, 2015-07-16:
When people say "do something *in* a directory" it means you have to make that directory current (or "change" into that directory) and then execute the specified commands. You change the current directory with command "cd". So, in your case, you do it like this:

    cd digimend-kernel-drivers-5
    make
    sudo make install

eazup, 2015-07-16:
yep. thank you, sir! i knew it was something incredibly basic that i just wsnt keen to,
in my defense, its alot like learning French just by being helicoptered into Lyon, i can follow the laid out step by steps (as long as versions are all current, which sometimes not, as you know), but im being led blind. 
i can repeat phrases, but i cant speak for myself, you feel me? when i look for "Dummies" versions of How To Kernel, even those assume certain things
to switch the analogy, i feel confident that i can learn my way around just fine, if i just saw a map of the whole island

spbnick, 2015-07-16:
Yeah, I know it can be overwhelming, but with this kind of enthusiasm and determination I'm sure you will learn it in no time :)

eazup, 2015-07-16:
hey now that my pad is running, id like to give you a shout out in my thanks column when my comic book gets published.

spbnick, 2015-07-16:
Thanks, will be glad to see it :)

eazup, 2015-07-16:
im still working on rebuilding my website (you know, cuz of all that ^, in part), but check me out at ahzdlarge.com, thanks again, man!

closed 2015-07-17 by spbnick