DIGImend/digimend-kernel-drivers #20: Tablet button crashes Krita

MechanizedWolf, 2015-08-14:
I have a  MonoPrice 8 x 5 Graphic Drawing Tablet (Product Number: 10593). I'm pretty sure it's a re-branded Huion tablet, so I have it working pretty well with the Digimend driver. Most of the side buttons work (although I have no idea how to reconfigure them), but the issue is the last button crashes Krita.

Is there any way to fix this? Even better, would there be a way to reconfigure the buttons myself?

Thank you for your time, and thank you for writing these drivers!

spbnick, 2015-08-14:
Hello,

Could you please send the "evtest" output for your tablet device when you press that last button?

Thank you.

spbnick, 2015-08-14:
The "xinput test" output will also help.

MechanizedWolf, 2015-08-14:
**evtest:**
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x256c product 0x6e version 0x111
Input device name: "10593"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 256 (BTN_0)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
    Event code 320 (BTN_TOOL_PEN)
    Event code 321 (BTN_TOOL_RUBBER)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max     2048
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max     2048
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     2047
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
Event: time 1439580811.746805, type 2 (EV_REL), code 8 (REL_WHEEL), value 1
Event: time 1439580811.746805, -------------- EV_SYN ------------

**xinput test:**
motion a[4]=1 
button press   4 
button release 4

spbnick, 2015-08-29:
This took me a while to get back to. I'm sorry for the delay.

I see from the dump that the DIGImend driver is not being used.
Can you make sure you installed the driver correctly?

Please try to install it again and post the output of the commands you executed to do it.

MechanizedWolf, 2015-08-29:
$ make
make -C /lib/modules/3.19.0-26-generic/build SUBDIRS=/home/marc/Downloads/digimend-kernel-drivers-5 modules
make[1]: Entering directory '/usr/src/linux-headers-3.19.0-26-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-3.19.0-26-generic'
marc@marc-GP60-2PE:~/Downloads/digimend-kernel-drivers-5$ sudo make install
make -C /lib/modules/3.19.0-26-generic/build SUBDIRS=/home/marc/Downloads/digimend-kernel-drivers-5 modules_install
make[1]: Entering directory '/usr/src/linux-headers-3.19.0-26-generic'
  INSTALL /home/marc/Downloads/digimend-kernel-drivers-5/hid-huion.ko
Can't read private key
  DEPMOD  3.19.0-26-generic
make[1]: Leaving directory '/usr/src/linux-headers-3.19.0-26-generic'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 hid-rebind.rules /lib/udev/rules.d/70-hid-rebind.rules
udevadm control --reload

Then I made sure the old drivers were removed, unplugged and reconnected my tablet. When I run evtest and xinput I get the same output.
I did notice however when I ran "sudo rmmod hid-huion" with the tablet plugged in, I did not get "rmmod: ERROR: Module hid_huion is not currently loaded" which meant the module was loaded. This happens every time I plug in the tablet.

spbnick, 2015-08-29:
Hmm. The installation looks fine. Could you please post the "dmesg" output after plugging in the tablet? Thank you.

MechanizedWolf, 2015-08-30:
[  858.566852] usb 1-1: new full-speed USB device number 4 using xhci_hcd
[  858.696412] usb 1-1: New USB device found, idVendor=256c, idProduct=006e
[  858.696415] usb 1-1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[  858.696416] usb 1-1: Product: 10593
[  858.697957] input: 10593 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:256C:006E.0009/input/input24
[  858.698166] huion 0003:256C:006E.0009: input,hidraw3: USB HID v1.0b Device [10593] on usb-0000:00:14.0-1/input0
[  858.699222] input: 10593 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:256C:006E.000A/input/input25
[  858.699428] huion 0003:256C:006E.000A: input,hiddev0,hidraw4: USB HID v1.11 Mouse [10593] on usb-0000:00:14.0-1/input1
[  858.700321] input: 10593 as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:256C:006E.000B/input/input26
[  858.807439] huion 0003:256C:006E.000B: input,hidraw5: USB HID v1.0b Keyboard [10593] on usb-0000:00:14.0-1/input2

ThePreviousOne, 2015-09-13:
the Huion Driver Seams to  be missing in the Current Build (v6) use v5 instead

artscoop, 2015-09-22:
Hi,
I've compiled v5 and it's not killing Krita anymore (Krita broke because of invalid coordinates triggered by the express buttons).

if I evtest the express keys on a Huion GT-185HD monitor tablet, I get:

```
motion a[1]=16 
motion a[1]=0 
motion a[1]=8 
motion a[1]=0 
motion a[1]=8 
motion a[1]=0 
motion a[1]=8 
motion a[1]=0 
motion a[1]=16 
motion a[1]=0 
```
(the value is 0 on release, and a bit value between 1 and 128 on press)
How can one map these commands to keyboard shortcuts ???

spbnick, 2015-09-22:
Hi everyone! Sorry, I'm very busy lately and have even less time for the project. I would recommend using the latest master with your tablets and the xf86-input-wacom X.org input drivers (not xf86-input-evdev), then remapping the buttons as necessary with xsetwacom.

MechanizedWolf, 2015-10-09:
Hi, sorry for the late response. I tried using the latest master and I still have the same issue.

ThePreviousOne, 2015-10-10:
@MechanizedWolf ( see me previous post) (master is v6 ... if that wasn't clear)

MechanizedWolf, 2015-10-10:
@ThePreviousOne I tried both and I get the same result.

spbnick, 2015-10-11:
@MechanizedWolf Did you use xf86-input-wacom with the master version?

closed 2016-06-26 by spbnick