DIGImend/digimend-kernel-drivers #21: Mint backports huion drivers to 3.13 kernel

test3333, 2015-09-07:
Just FYI
You can put in the compatibility-list that mint is supporting it even on the older kernels that noobs dont need to bother about using another kernel branch on mint.

spbnick, 2015-09-08:
Thank you. I'll leave this issue open to remember to add this to the compatibility info.

spbnick, 2016-09-03:
I have to regretfully inform you, that I (the project maintainer) am [leaving the project](http://spbnick.github.io/2016/07/31/Wrapping-up-DIGImend-work.html). I'm no longer working on supporting new tablets, nor am I supporting users. I'm still accepting pull requests and work on HOWTOs and syncing with upstream, but only until November 2016. Members of the community may still help other users. Anyone desiring to step in is welcome to contact me.