DIGImend/digimend-kernel-drivers #23: hovering pen makes cursor move erratically in horizontal direction

gruntfarm, 2015-11-26:
New driver makes the cursor move erratically on the horizontal line when pen is hovering or touching the tablet. I have a Huion H610 and running it on Ubuntu 14.04.

I've tried installing via deb package as well as "make install". Before this I had a Digimend driver that was about 5-6 months old and it worked fine.

spbnick, 2015-11-29:
Could you please post the "dmesg" output after plugging in the tablet, as well as "modinfo hid-uclogic" output? Thank you.

gruntfarm, 2015-11-30:
Here is the output of dmesg

    [  147.086050] huion 0003:256C:006E.000F: input,hidraw11: USB HID v1.11 Mouse [HUION H610] on usb-0000:00:14.0-2.1/input0

And here is the output of modinfo hid-uclogic

    filename:       /lib/modules/3.13.0-70-generic/kernel/drivers/hid/hid-uclogic.ko
    license:        GPL
    srcversion:     41F93291FE3C6DDB3BFBC7F
    alias:          hid:b0003g*v00005543p00000781
    alias:          hid:b0003g*v00005543p00000522
    alias:          hid:b0003g*v00005543p00000064
    alias:          hid:b0003g*v00005543p00000005
    alias:          hid:b0003g*v00005543p00000004
    alias:          hid:b0003g*v00005543p00000003
    alias:          hid:b0003g*v00005543p00000042
    depends:        hid
    intree:         Y
    vermagic:       3.13.0-70-generic SMP mod_unload modversions 
    signer:         Magrathea: Glacier signing key
    sig_key:        7E:CB:D7:FA:0D:A5:6C:E3:B5:83:14:83:CD:07:7A:30:F0:84:3B:39
    sig_hashalgo:   sha512

Thank you.

spbnick, 2015-11-30:
This looks like the driver wasn't installed/registered. Can you please show the output of package installation and/or "sudo make install"? Thank you.

gruntfarm, 2015-12-01:
Here is what shows up after I run "sudo make install" from the digimend-kernel-drivers directory
```
make -C /lib/modules/3.13.0-70-generic/build SUBDIRS=/home/mladen/Downloads/digimend-kernel-drivers modules_install
make[1]: Entering directory `/usr/src/linux-headers-3.13.0-70-generic'
  INSTALL /home/mladen/Downloads/digimend-kernel-drivers/hid-kye.ko
Can't read private key
  INSTALL /home/mladen/Downloads/digimend-kernel-drivers/hid-polostar.ko
Can't read private key
  INSTALL /home/mladen/Downloads/digimend-kernel-drivers/hid-uclogic.ko
Can't read private key
  DEPMOD  3.13.0-70-generic
make[1]: Leaving directory `/usr/src/linux-headers-3.13.0-70-generic'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules
udevadm control --reload
```

spbnick, 2016-06-26:
@gruntfarm Now, that's a long delay with the answer, I know.

Nevertheless, hardware problems with the cheap tablets are not unheard of. There can be problems with your USB host hardware as well.

Hope you're not getting them anymore.

closed 2016-06-26 by spbnick