DIGImend/digimend-kernel-drivers #24: Readme has no link to Debian experimental .deb package

sleekweasel, 2015-12-03:
The readme says:

> If you're using Debian or a derived distro, such as Ubuntu, please try using the experimental .deb package.

but then frustratingly doesn't hint at where to get it from. I presume I need to add something to my apt config?

sunaku, 2015-12-03:
IMHO :smirk_cat: it's easier just to [build from source](https://github.com/DIGImend/digimend-kernel-drivers#installing-source-package-manually), even if you're using Ubuntu.

sleekweasel, 2015-12-03:
True - I did this - but maybe it would be better not to tease people with the idea that there's a pre-cooked solution out there? :)

spbnick, 2015-12-04:
That was deliberate to avoid updating the link to the release in the code, but I see how that can be difficult especially now when the GitHub interface has changed. Here's the file: https://github.com/DIGImend/digimend-kernel-drivers/releases/download/v6/digimend-dkms_6_all.deb

I wouldn't recommend building from source as the next time your kernel version is updated the driver will stop working and you'll have to reinstall it.

closed 2016-06-26 by spbnick