DIGImend/digimend-kernel-drivers #25: small drawing area on the screen

gpsanino, 2016-02-01:
Hi, perhaps it has been reported before. I could not find any post alike. Please forgive me if it is the case.
I got a gtablet Huion H420 (my first) that should have been recognized by the linux kernel (OpenSuSE-Desktop 3.16.7-29-desktop) and it did it. However, the area that the "cursor" uses of the screen is very small (on the top left corner of the screen). To be more clear, I can drag the tablet's pen across the entire tabled but the cursor navigates only on a reduced area of the screen.

I found no GUI to change settings nor other help associated to this driver among KDE desktop settings nor Yast. I tried restarting the PC with the tablet connected, and had no change. Also downloaded, compiled and installed DIGIMend Kernel-drivers with no change neither. The tablet is recognized properly (dmesg shows the H420) and it works, but only on a very reduced area of my screen regardless the application (i.e. Inkscape with input device H420 set to screen mode).
This is on OpenSuSE 13.2 64. However, if I connect this tablet to a small notebook with OpenSuSE Leap 42.1, it does recognizes the tablet without needing to install Digimend drivers and it does uses the entire screen. Quite weird since in general with 13.2 I have had way lesser issues than with Leap 42.1.

Is there any way to change the configuration/settings on the driver? like a way to extend the drawing area to the full screen size, and perhaps manage the buttons of the tablet as well as the pen? More than just connecting it, I have not found anything related to changing the way this hardware is used.

Any help is very appreciated.

spbnick, 2016-02-02:
Hi,

The OpenSUSE Desktop kernel version doesn't have the driver for your tablet and I suspect that the installation of the DIGImend kernel drivers didn't work for you. I never tried it myself and haven't had any reports it is working, IIRC, so it is likely it wouldn't work. Can you post the output of "make" and "sudo make install" of the driver? Thank you. The alternative for you can be installing a fresher kernel on OpenSUSE desktop.

OpenSUSE Leap, however, has a very recent kernel, which includes the driver for your tablet, so it is no wonder it worked without a problem.

There is configuration to some extent, but shouldn't be necessary and the tablet should cover the whole screen(s) by default if it's working correctly. Fix it with configuration wouldn't make it work well.

gpsanino, 2016-02-03:
Hi Nikolai,

This is the output form "make":

> user@My-OpenSuSE-Desktop:~/Downloads/digimend-kernel-drivers-master> make
> make -C /lib/modules/3.16.7-29-desktop/build
> SUBDIRS=/home/user/Downloads/digimend-kernel-drivers-master modules
> make[1]: Entering directory '/usr/src/linux-3.16.7-29-obj/x86_64/desktop'
> make[1]: Entering directory `/usr/src/linux-3.16.7-29-obj/x86_64/desktop'
>   CC [M]  /home/user/Downloads/digimend-kernel-drivers-master/hid-kye.o
>   CC [M] 
> /home/user/Downloads/digimend-kernel-drivers-master/hid-uclogic.o
>   CC [M] 
> /home/user/Downloads/digimend-kernel-drivers-master/hid-polostar.o
>   Building modules, stage 2.
>   MODPOST 3 modules
>   CC     
> /home/user/Downloads/digimend-kernel-drivers-master/hid-kye.mod.o
>   LD [M]  /home/user/Downloads/digimend-kernel-drivers-master/hid-kye.ko
>   CC     
> /home/user/Downloads/digimend-kernel-drivers-master/hid-polostar.mod.o
>   LD [M] 
> /home/user/Downloads/digimend-kernel-drivers-master/hid-polostar.ko
>   CC     
> /home/user/Downloads/digimend-kernel-drivers-master/hid-uclogic.mod.o
>   LD [M] 
> /home/user/Downloads/digimend-kernel-drivers-master/hid-uclogic.ko
> make[1]: Leaving directory '/usr/src/linux-3.16.7-29-obj/x86_64/desktop'
> user@My-OpenSuSE-Desktop:~/Downloads/digimend-kernel-drivers-master>

The "sudo make install" post:

> user@My-OpenSuSE-Desktop:~/Downloads/digimend-kernel-drivers-master>
> sudo make install
> root's
> password:                                                                                                                                                             
>  
> make -C /lib/modules/3.16.7-29-desktop/build
> SUBDIRS=/home/user/Downloads/digimend-kernel-drivers-master
> modules_install                                                       
> make[1]: Entering directory
> '/usr/src/linux-3.16.7-29-obj/x86_64/desktop'                                                                                                    
>  
> make[1]: Entering directory
> `/usr/src/linux-3.16.7-29-obj/x86_64/desktop'                                                                                                    
>  
>   INSTALL
> /home/user/Downloads/digimend-kernel-drivers-master/hid-kye.ko                                                                                                     
>  
>   INSTALL
> /home/user/Downloads/digimend-kernel-drivers-master/hid-polostar.ko                                                                                                
>  
>   INSTALL
> /home/user/Downloads/digimend-kernel-drivers-master/hid-uclogic.ko                                                                                                 
>  
>   DEPMOD 
> 3.16.7-29-desktop                                                                                                                                                  
>  
> make[1]: Leaving directory
> '/usr/src/linux-3.16.7-29-obj/x86_64/desktop'                                                                                                     
>  
> install -D -m 0644 digimend.conf
> /etc/depmod.d/digimend.conf                                                                                                                 
>  
> depmod
> -a                                                                                                                                                                    
>  
> install hid-rebind
> /sbin/hid-rebind                                                                                                                                          
>  
> install -m 0644 90-hid-rebind.rules
> /lib/udev/rules.d/90-hid-rebind.rules                                                                                                    
>  
> udevadm control
> --reload                                                                                                                                                     
>  
> user@My-OpenSuSE-Desktop:~/Downloads/digimend-kernel-drivers-master> 

Linux 3.16.7-29-desktop
openSUSE 13.2 (Harlequin) (x86_64)

Thanks a lot
gps

spbnick, 2016-09-03:
I have to regretfully inform you, that I (the project maintainer) am [leaving the project](http://spbnick.github.io/2016/07/31/Wrapping-up-DIGImend-work.html). I'm no longer working on supporting new tablets, nor am I supporting users. I'm still accepting pull requests and work on HOWTOs and syncing with upstream, but only until November 2016. Members of the community may still help other users. Anyone desiring to step in is welcome to contact me.