DIGImend/digimend-kernel-drivers #26: Huion H610 Pro - button configuration

sunaku, 2016-02-02:
Hello,

I'm using a Huion H610 Pro graphics tablet with Linux kernel 4.3.4_1 which seems to _already_ include the wonderful DIGImend drivers because I didn't have to compile and install the source code in this digimend-kernel-drivers repository for pressure sensitivity to work properly: instead, I just plugged in my tablet's USB cable and everything **Just Worked!** :santa: 

Now, I want to configure the big buttons on the left-hand side (outside of the tablet drawing surface). :musical_keyboard:

I tried following the instructions in https://github.com/DIGImend/digimend-kernel-drivers/issues/22#issuecomment-166021282 but still `xsetwacom --list` never sees my tablet. :disappointed:

The promising thing is that some portions of the tablet are recognized as "Keyboard" devices, so I'm thinking of creating a custom XKB keyboard model that maps my tablet's scancodes to key presses. :bulb:

Is this the proper way to do things?

Thanks for your consideration.

# /var/log/Xorg.0.log

Here is what I see when I plug in my tablet's USB cable:

```
[  9694.071] (II) config/udev: Adding input device HUION PenTablet Pen (/dev/input/mouse1)
[  9694.071] (II) No input driver specified, ignoring this device.
[  9694.071] (II) This device may have been added with another device file.
[  9694.072] (II) config/udev: Adding input device HUION PenTablet Mouse (/dev/input/mouse2)
[  9694.072] (II) No input driver specified, ignoring this device.
[  9694.072] (II) This device may have been added with another device file.
[  9694.084] (II) config/udev: Adding input device HUION PenTablet Keyboard (/dev/input/event18)
[  9694.084] (**) HUION PenTablet Keyboard: Applying InputClass "evdev keyboard catchall"
[  9694.084] (II) Using input driver 'evdev' for 'HUION PenTablet Keyboard'
[  9694.084] (**) HUION PenTablet Keyboard: always reports core events
[  9694.084] (**) evdev: HUION PenTablet Keyboard: Device: "/dev/input/event18"
[  9694.084] (--) evdev: HUION PenTablet Keyboard: Vendor 0x256c Product 0x6e
[  9694.084] (--) evdev: HUION PenTablet Keyboard: Found keys
[  9694.084] (II) evdev: HUION PenTablet Keyboard: Configuring as keyboard
[  9694.084] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5:1.2/0003:256C:006E.0038/input/input117/event18"
[  9694.084] (II) XINPUT: Adding extended input device "HUION PenTablet Keyboard" (type: KEYBOARD, id 14)
[  9694.084] (**) Option "xkb_rules" "evdev"
[  9694.087] (II) config/udev: Adding input device HUION PenTablet Consumer Control (/dev/input/event19)
[  9694.087] (**) HUION PenTablet Consumer Control: Applying InputClass "evdev keyboard catchall"
[  9694.087] (II) Using input driver 'evdev' for 'HUION PenTablet Consumer Control'
[  9694.087] (**) HUION PenTablet Consumer Control: always reports core events
[  9694.087] (**) evdev: HUION PenTablet Consumer Control: Device: "/dev/input/event19"
[  9694.087] (--) evdev: HUION PenTablet Consumer Control: Vendor 0x256c Product 0x6e
[  9694.087] (--) evdev: HUION PenTablet Consumer Control: Found 1 mouse buttons
[  9694.087] (--) evdev: HUION PenTablet Consumer Control: Found scroll wheel(s)
[  9694.088] (--) evdev: HUION PenTablet Consumer Control: Found relative axes
[  9694.088] (II) evdev: HUION PenTablet Consumer Control: Forcing relative x/y axes to exist.
[  9694.088] (--) evdev: HUION PenTablet Consumer Control: Found absolute axes
[  9694.088] (II) evdev: HUION PenTablet Consumer Control: Forcing absolute x/y axes to exist.
[  9694.088] (--) evdev: HUION PenTablet Consumer Control: Found keys
[  9694.088] (II) evdev: HUION PenTablet Consumer Control: Configuring as mouse
[  9694.088] (II) evdev: HUION PenTablet Consumer Control: Configuring as keyboard
[  9694.088] (II) evdev: HUION PenTablet Consumer Control: Adding scrollwheel support
[  9694.088] (**) evdev: HUION PenTablet Consumer Control: YAxisMapping: buttons 4 and 5
[  9694.088] (**) evdev: HUION PenTablet Consumer Control: EmulateWheelButton: 4, EmulateWheelInertia: 10, EmulateWheelTimeout: 200
[  9694.088] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5:1.2/0003:256C:006E.0038/input/input118/event19"
[  9694.088] (II) XINPUT: Adding extended input device "HUION PenTablet Consumer Control" (type: KEYBOARD, id 15)
[  9694.088] (**) Option "xkb_rules" "evdev"
[  9694.088] (II) evdev: HUION PenTablet Consumer Control: initialized for relative axes.
[  9694.088] (WW) evdev: HUION PenTablet Consumer Control: ignoring absolute axes.
[  9694.088] (**) HUION PenTablet Consumer Control: (accel) keeping acceleration scheme 1
[  9694.088] (**) HUION PenTablet Consumer Control: (accel) acceleration profile 0
[  9694.088] (**) HUION PenTablet Consumer Control: (accel) acceleration factor: 2.000
[  9694.088] (**) HUION PenTablet Consumer Control: (accel) acceleration threshold: 4
[  9694.089] (II) config/udev: Adding input device HUION PenTablet System Control (/dev/input/event20)
[  9694.089] (**) HUION PenTablet System Control: Applying InputClass "evdev keyboard catchall"
[  9694.089] (II) Using input driver 'evdev' for 'HUION PenTablet System Control'
[  9694.089] (**) HUION PenTablet System Control: always reports core events
[  9694.089] (**) evdev: HUION PenTablet System Control: Device: "/dev/input/event20"
[  9694.089] (--) evdev: HUION PenTablet System Control: Vendor 0x256c Product 0x6e
[  9694.089] (--) evdev: HUION PenTablet System Control: Found keys
[... 20 lines cut ...]
[  9694.232] (**) HUION PenTablet Pen: (accel) acceleration profile 0
[  9694.232] (**) HUION PenTablet Pen: (accel) acceleration factor: 2.000
[  9694.232] (**) HUION PenTablet Pen: (accel) acceleration threshold: 4
[  9694.233] (II) config/udev: Adding input device HUION PenTablet Mouse (/dev/input/event17)
[  9694.233] (**) HUION PenTablet Mouse: Applying InputClass "evdev pointer catchall"
[  9694.233] (II) Using input driver 'evdev' for 'HUION PenTablet Mouse'
[  9694.233] (**) HUION PenTablet Mouse: always reports core events
[  9694.233] (**) evdev: HUION PenTablet Mouse: Device: "/dev/input/event17"
[  9694.284] (--) evdev: HUION PenTablet Mouse: Vendor 0x256c Product 0x6e
[  9694.284] (--) evdev: HUION PenTablet Mouse: Found 9 mouse buttons
[  9694.284] (--) evdev: HUION PenTablet Mouse: Found scroll wheel(s)
[  9694.284] (--) evdev: HUION PenTablet Mouse: Found relative axes
[  9694.284] (--) evdev: HUION PenTablet Mouse: Found x and y relative axes
[  9694.284] (II) evdev: HUION PenTablet Mouse: Configuring as mouse
[  9694.284] (II) evdev: HUION PenTablet Mouse: Adding scrollwheel support
[  9694.284] (**) evdev: HUION PenTablet Mouse: YAxisMapping: buttons 4 and 5
[  9694.284] (**) evdev: HUION PenTablet Mouse: EmulateWheelButton: 4, EmulateWheelInertia: 10, EmulateWheelTimeout: 200
[  9694.284] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb2/2-5/2-5:1.1/0003:256C:006E.0037/input/input115/event17"
[  9694.284] (II) XINPUT: Adding extended input device "HUION PenTablet Mouse" (type: MOUSE, id 18)
[  9694.284] (II) evdev: HUION PenTablet Mouse: initialized for relative axes.
[  9694.285] (**) HUION PenTablet Mouse: (accel) keeping acceleration scheme 1
[  9694.285] (**) HUION PenTablet Mouse: (accel) acceleration profile 0
[  9694.285] (**) HUION PenTablet Mouse: (accel) acceleration factor: 2.000
[  9694.285] (**) HUION PenTablet Mouse: (accel) acceleration threshold: 4
```

kriks57, 2016-02-18:
Hi,
I thought I would add the same problem as you, but I could get a little bit further.
It seems evdev is the first module to load for tablet (catchall rule), so I had to comment the tablet part in 
 /usr/share/X11/xorg.conf.d/10-evdev.conf

> Section "InputClass"
>         Identifier "evdev tablet catchall"
>         MatchIsTablet "on"
>         MatchDevicePath "/dev/input/event*"
>         Driver "evdev"
> EndSection

then adding /etc/X11/xorg.conf.d/52-tablet.conf containing

> Section "InputClass"
>     Identifier "Huion on wacom"
>  #    MatchIsTablet "on"
>     MatchProduct "HUION"
>     MatchDevicePath "/dev/input/event*"
>     Driver "wacom"
> EndSection

rebooting.

then `xsetwacom --list` works for me:

> $ xsetwacom --list                                                                                
> HUION PenTablet Pad pad                 id: 12  type: PAD
> HUION PenTablet Pen stylus              id: 13  type: STYLUS

and the button are 1,2,3,8,9,10,11,12 like in the comment you mentioned.

I didn't search for the buttons on the top of the pad.

Maybe it can help you.

Clavos-Studios, 2016-03-05:
## SOLUTION:

- Installing digimend-kernel-drivers-6 from source followings the instructions.
- Creating 52-tablet.conf because it didn't exist:

```
cd /etc/X11
sudo mkdir xorg.conf.d 
cd xorg.conf.d 
sudo gedit 52-tablet.conf 
```

- Copy and paste the following in that file:
```
Section "InputClass"
Identifier "Huion on wacom"
# MatchIsTablet "on"
MatchProduct "HUION"
MatchDevicePath "/dev/input/event*"
Driver "wacom"
EndSection
```

- Saving the file and restart.

- Open the terminal and put the following: `xsetwacom --list`
You should see something like this:
> HUION PenTablet Pad pad         	id: 13	type: PAD
HUION PenTablet Pen stylus      	id: 14	type: STYLUS 

The buttons are: 1,2,3,8,9,10,11,12.

- Put this in the terminal, defining the desired keys for each button (Top-down on the tablet Huion H610 Pro) :

```
xsetwacom --set "HUION PenTablet Pad pad" Button 1 "key +ctrl +z -z -ctrl"
xsetwacom --set "HUION PenTablet Pad pad" Button 2 “key e”
xsetwacom --set "HUION PenTablet Pad pad" Button 3 "key +ctrl"
xsetwacom --set "HUION PenTablet Pad pad" Button 8 "key +shift"
xsetwacom --set "HUION PenTablet Pad pad" Button 9 "key +ctrl +  -ctrl"
xsetwacom --set "HUION PenTablet Pad pad" Button 10 "key +ctrl -  -ctrl"
xsetwacom --set "HUION PenTablet Pad pad" Button 11 "key b"
xsetwacom --set "HUION PenTablet Pad pad" Button 12 "key space"

```

# And ready, the buttons will function. :)

thomanski, 2016-03-06:
I'm trying this for an H610 Pro on Kubuntu 16.04 with 4.4 kernel (tried 4.5 as well via https://wiki.ubuntu.com/Kernel/MainlineBuilds) and it works pretty well without installing the digimend drivers but I'd like to customise the buttons, too, and so have tried the suggestions on here.

`xsetwacom --list` unfortunately shows the stylus only, and when I check the xorg log I see HUION PenTablet Consumer Control|Keyboard|Mouse|Pen|System Control devices only, no HUION PenTablet Pad pad. With `evtest` I can see that the HUION keyboard device does handle the button presses on the tablet.

Note I first tried without installing the digimend drivers, since everything but button reconfig seemed to work fine without them. When that didn't work I installed the .deb package and restarted but then `xinput list` doesn't even list a single HUION device and consequently the tablet doesn't work at all. Any ideas?

kriks57, 2016-03-06:
@thomanski check syslog when you plug the tablet in, to be sure that evdev is not the driver loaded, you may have the same problem I had.

Clavos-Studios, 2016-03-06:
The buttons on the tablet unconfigures whenever the disconnect, Any idea? :/

thomanski, 2016-03-06:
@kriks57 You mean the xorg log, right? Can't find anything in syslog on this. Thanks to /var/log/Xorg.0.log I'm pretty sure the pad device just doesn't show up at all, it doesn't ever get mentioned.

In any case, I've now applied Clavos-Studios' steps to a Ubuntu 14.04 installation and that's all working fine.

The devices listed in Ubuntu 14.04 include "Pad" and "Pad pad", in Kubuntu 16.04 they were missing:
```bash
$fgrep HUION /var/log/Xorg.0.log | sed -e 's/.*HUION PenTablet \([^("'"'"':]*\).*/\1/;s/ $//g' | sort -u | tr "\n" "|"
Consumer Control|Consumer Control pad|Keyboard|Mouse|Pad|Pad pad|Pen|Pen stylus|System Control # <-- Ubuntu 14.04
Consumer Control|Consumer Control pad|Keyboard|Mouse|Pen|Pen stylus|System Control # <-- Kubuntu 16.04
```
@Clavos-Studios Looks like you have to manually re-apply or set up Wdaemon to have the changes automatically applied after a hotplug, see linuxwacom.sourceforge.net/wiki/index.php/Tablet_Configuration#Script_Survival_after_a_Hotplug

spbnick, 2016-06-26:
Hi everyone, the developer here. First of all, thanks to all for cooperating and finding a solution by yourselves!

I would really appreciate someone (everyone actually) to hang around and deal with user issues, letting me use more of my really scarce time to handle lower level issues.

Also, really great would be a HOWTO (or several) for the site. I'll gladly help whoever wants to do that to figure out the markup and the testing of the page. Just ping me.

@thomanski The difference between Ubuntu and Kubuntu is likely in the X.org input driver version.

closed 2016-06-26 by spbnick

momashi, 2016-11-29:
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@ A COMPLETE GUIDE ON HOW TO CONFIGURE THE HUION H610 ON LINUX @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@

by Fadi R

CREDITS:
========================================================
This guide is a meld of my own research and, to a greater degree, the answers of bebop(+David Foester) and pram on Ask Ubuntu (http://askubuntu.com/questions/500141/huion-h610-tablet) AS WELL AS Clavos-Studios post on the Digimend GitHub  (https://github.com/DIGImend/digimend-kernel-drivers/issues/26). Thanks guys, I learned alot from your posts. And of course, Major Kudos to the DigiMend team and Nikolai Kondrashov (project founder). Their tireless work has opened up a wide spectrum of tablets to the Linux community. (http://digimend.github.io)
========================================================

[0] TABLE OF CONTENTS:
======================

[1] DRIVER INSTALLATION
[2] CONFIGURING AND MAPPING TABLET
[3] CONFIGURING APPS TO RUN WITH TABLET
[4] SUGGESTED SOFTWARE

[1] DRIVER INSTALLATION:
========================
(1.1) Ensure that you have the following packages:
	a) xf86-input-wacom
	b) linux-headers (in Arch, Debian, Ubuntu or Mint) or kernel-headers (Red Hat)
(1.2) Download the latest Digimend drivers from Github (https://github.com/DIGImend/digimend-kernel-drivers). For some reason beyond me, I couldn't get the source code from the github.io Digimend site to compile (https://digimend.github.io/drivers/) but it may be different for you...
	a) In terminal, input:
	    sudo rmmod hid-kye
	    sudo rmmod hid-uclogic
	    sudo rmmod hid-huion
	b) extract digimend sourcecode (from github.com, it's: digimend-kernel-drivers-master.zip). In terminal browse to extracted dir and input:
	    make
	    sudo make install		
(1.3) Setting up an 52-tablet.conf entry for the H610

	a) make the 52-tablet.conf file if it doesn't exist:
	sudo mkdir /etc/X11/xorg.conf.d 
	cd /etc/X11/xorg.conf.d 
	sudo gedit 52-tablet.conf 
		
	b) paste this entry for the H610 in 52-tablet.conf:
	
	Section "InputClass"
	Identifier "Huion on wacom"
	# MatchIsTablet "on"
	MatchProduct "HUION"
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
	EndSection 
	
	c) save file and restart

	d) upon restarting, when you open Terminal and input:
	xsetwacom --list
	
	(you should see:)	
	HUION PenTablet Pad pad id: 11 type: PAD
	HUION PenTablet Pen stylus id: 12 type: STYLUS
	(the id will vary from machine to machine)
	
[2] CONFIGURING AND MAPPING TABLET:
===================================

(2.1) Defining Pad Left-Hand Buttons:
	a) As we saw previously, the pad is defined "HUION PenTablet Pad pad" and it's button are: 1,2,3,8,9,10,11,12. These buttons will be undefined every time you log in. To define them, you can input the following command synthax in Terminal

	xsetwacom --set 'DEVICE NAME' Button NUMBER "key KEYSTROKES"

	b) You can automate it by making script file that autoruns every time you open a session. For example, my script is called Huion.Default.sh and I like bebop from AskUbuntu's button scheme:

	#!/bin/sh
[... 26 lines cut ...]
	eg. if you want an eraser on your Stylus, you can sacrifice the middle-click on button 2 by inputing:
	xsetwacom --set 'HUION PenTablet Pen stylus' Button 2 "key e"
	(if you want to revert back to default:)
	xsetwacom --set 'HUION PenTablet Pen stylus' Button 2 "button +2"

	
(2.5) Defining Pressure Curves:
	As far I can tell, the linuxwacom / DigiMend hack doesn't seem to support the PressureCurve parameter. If anyone's figured a way to get it to work, let me know. That being said, alot of application such as Krita and GIMP will have their own pressure curve settings. I have not explored them yet, so I don't know how well they'll work with the H610.

(2.6) Left Hand Support
	--- work in progress
	
		
[3] CONFIGURING APPS TO RUN WITH TABLET:
=========================================
	
DigiMend's Support section has some really awesome guides on how to setup with your tablet with the following applications: Alchemy, Blender, GIMP, Inkscape, Krita, MyPaint, Synfig Studio and apps running out of Wine.
	
Just go to http://digimend.github.io/support/ and skip down to the "Application setup" section.

	
[4] SUGGESTED SOFTWARE:
========================

-- Work in progress --

Aldotablet, 2017-05-07:
I use kde (5.9.5) neon 16.04 - I just updated to kernel 4.11 I noted that /usr/share/libwacom contains huion-h610-pro.tablet I wonder to what extent the h610 works since this kernel update and if the digimend driver is still needed my h610 has been shipped but not yet received.

spbnick, 2017-05-08:
The Wacom driver developers added support for some Huion tablets a while ago and you can try configuring the Wacom driver with your tablet. Even the graphical configuration utility might work, at least there was an effort to make it work.

Zarnaik, 2018-01-14:
Has anyone been able to assign shortcuts to the "soft buttons" (1-16 on-pad markings)? 
Or is there not any support for this feature?

Negirno, 2018-04-24:
I've did the stuff written by momashi to enable the left side buttons on my H610 pro.

It worked great for two days then it basically reverted to the previous state, i.e. pad basically works, there is pressure sensitivity, but the xsetwacom script complains that it cannot find the device. Only the pen is detected with the `xsetwacom --list` command.

I've tried uninstalling and then reinstalling the digimend drivers, but to no avail. What gives?

I'm using kernel 4.4.0-121-generic on Ubuntu 16.04

TheDaviot, 2018-05-16:
No issues with the basic drivers, but after creating the 52-tablet.conf entry and rebooting, "xsetwacom --list" returns nothing.

itreers, 2018-10-03:
Can't use you solution. It's not working for my H640P.

root@UA00507 ~# lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0a2b Intel Corp. 
Bus 001 Device 003: ID 256c:006e  
Bus 001 Device 005: ID 0bda:5650 Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

No Huion here as you can see.

So my config is 
/etc/X11/xorg.conf.d/52-tablet.conf

Section "InputClass"
Identifier "Huion on wacom"
MatchUSBID "256c:006e"
MatchDevicePath "/dev/input/event*"
Driver "wacom"
EndSection

Reboot

After this working as expected

am@UA00507~> xsetwacom --list
HID 256c:006e Pen stylus        	id: 13	type: STYLUS    
HID 256c:006e Pad pad           	id: 14	type: PAD

orbitaldigital, 2019-07-15:
after installing the driver "digimend" and restart pc, the tablet work's but whitout keys,

i have tested different method on this post and others, but nothing...

i have modify the Script witch this :

```
#!/bin/sh
xsetwacom --set 'HID 256c:006e Pad pad' Button 1 "key +ctrl +z -z -ctrl"
xsetwacom --set 'HID 256c:006e Pad pad' Button 2 "key e"
xsetwacom --set 'HID 256c:006e Pad pad' Button 3 "key b"
xsetwacom --set 'HID 256c:006e Pad pad' Button 8 "key +"
xsetwacom --set 'HID 256c:006e Pad pad' Button 9 "key -"
xsetwacom --set 'HID 256c:006e Pad pad' Button 10 "key ]"
xsetwacom --set 'HID 256c:006e Pad pad' Button 11 "key ["
xsetwacom --set 'HID 256c:006e Pad pad' Button 12 "key p"
```

And it's work's for me ! 
I m on linux mint 19.1 x64

AhmadShiddiq, 2019-11-01:
> after installing the driver "digimend" and restart pc, the tablet work's but whitout keys,
>
> i have tested different method on this post and others, but nothing...
>
> i have modify the Script witch [...]

I try that too, but after restart configuration has reset again, my config is gone. How about you?
I want save my configuration key when my pc restarts or shutdown.

Skincanvas, 2019-12-11:
@AhmadShiddiq 
Firstly you want to put your configurations all in one file called a script. To quote @momashi 

```
b) You can automate it by making script file that autoruns every time you open a session. For example, my script is called Huion.Default.sh and I like bebop from AskUbuntu's button scheme:

#!/bin/sh
xsetwacom --set 'HUION PenTablet Pad pad' Button 1 "key +ctrl +z -z -ctrl"
xsetwacom --set 'HUION PenTablet Pad pad' Button 2 "key e"
xsetwacom --set 'HUION PenTablet Pad pad' Button 3 "key b"
xsetwacom --set 'HUION PenTablet Pad pad' Button 8 "key +"
xsetwacom --set 'HUION PenTablet Pad pad' Button 9 "key -"
xsetwacom --set 'HUION PenTablet Pad pad' Button 10 "key ]"
xsetwacom --set 'HUION PenTablet Pad pad' Button 11 "key ["
xsetwacom --set 'HUION PenTablet Pad pad' Button 12 "key p"

c) Don't forget to give the script execution permission:
chmod +x Huion.Default.sh

d) You can make different scripts and button schemes for the different Apps you use with the tablet. When making lauchers, you can make those scripts lauch before you launch the apps themselves. You do this by editing the Command in the laucher or desktop file. For example: Command: ~/./GIMP-tablet-scheme.sh && /usr/bin/gimp 	
```
Then you can add the script file to your desktop environments autostart settings menu as if it were a program. You make a new entry and when it asks for a command you put in /home/user/script.sh or whatever. Alternatively add the script path to your ~/.xinitrc
Either way it will only work if you have it plugged in on boot. I would suggest an alias if that isn't an option for you (i have to unplug the thing when not in use or it clicks on bottom left of my screen)

kortschak, 2020-03-28:
>Either way it will only work if you have it plugged in on boot. **I would suggest an alias** if that isn't an option for you (i have to unplug the thing when not in use or it clicks on bottom left of my screen)

@Skincanvas Can you point to how to set up an alias? (This note saved me huge amounts of frustration when I was trying to debug something that had worked in the past, but then didn't - was not plugged in at boot).

DebsankarBanerjee, 2020-10-18:
Thanks for summing up the set-up process in a very neat way. The tablet works fine but the cursor disappears in inkscape, gimp, Xournal etc and it becomes very difficult to write/draw then. Anyone faced this same issue? Can this be fixed in the X11 config somewhere? Using Ubuntu 16.04.

lf-, 2021-02-11:
In case anyone else is having no hardware events reported,

The suggestion to use evtest was very helpful - it ruled out all the other variables, and I found out something I probably should have read the tablet's manual to discover, but I am a Linux user so I lack the sense to read such manuals ;-)

There's a switch on my H640p that disables the hardware buttons, which is on the opposite side of the tablet as the buttons and has a lock icon. If it's on, no hardware button events will be delivered to the computer. Whoops.

SinTan1729, 2021-11-30:
You can set it up so that the script by @Skincanvas is run whenever the tablet is connected and hence it'll work like plug-and-play. The details can be found [here](https://github.com/SinTan1729/random/blob/main/huion-tablet-setup.sh) with a working script for my H640P (the config works for any other huion tablet, just that the button count may change).