DIGImend/digimend-kernel-drivers #27: Huion H610 Pro with multiple screens

PiranhaGeorge, 2016-02-04:
Hi,

I have a Huion H610 Pro on OpenSuse Leap 42.1. It seems to be working well, but unfortunately I'm having some issues with the drawing area across multiple screens. Currently the the drawing area covers three screens, but I have Inkscape open on my rightmost screen. The result is that I can only use the right third of my tablet to draw. 

I'm not sure if this constitutes a bug, but I couldn't find any way of working around it besides having Inkscape stretched across all three screens, which is far from ideal.

Any help would be much appreciated.

Regards,

George

Clavos-Studios, 2016-03-06:
### Try using this: http://wenhsinjen.github.io/ptxconf/

It looks super comfortable, if it helps please let me know :)

JJSteffen, 2016-05-07:
I just tried Clavos-Studios suggestion and it worked with my H610 pro on CentOS 7. Thank you :)

alt165am, 2016-08-30:
The ptxconf did the trick for me! (Huion H610 Pro under Linux Mint 17.2). However, I would like to change the acceleration of the stylus for one specific screen, since both of my screens have different resolutions and the stylus go waaay too slow on the screen that its bound to... anyone got a work around?

spbnick, 2016-09-03:
I have to regretfully inform you, that I (the project maintainer) am [leaving the project](http://spbnick.github.io/2016/07/31/Wrapping-up-DIGImend-work.html). I'm no longer working on supporting new tablets, nor am I supporting users. I'm still accepting pull requests and work on HOWTOs and syncing with upstream, but only until November 2016. Members of the community may still help other users. Anyone desiring to step in is welcome to contact me.

closed 2016-09-03 by spbnick