DIGImend/digimend-kernel-drivers #28: Yiynova MVP20U+RH buttons

Dracozny, 2016-03-03:
I Recently acquired the above tablet and for the most part it works flawlessly with the .deb version of the digimed-kernel-driver. However the RH controller is not so forgiving. the controller shows up as an extra device 'UC-Logic MVP20U+RH Pad' 
I was able to test each button but have yet to figure out how to assign the buttons to anything else.

Running xinput test "UC-Logic MVP20U+RH Pad" I get the following which doesn't seem so bad but scrolling to the right produces strange results:

button 5
```
button press   9 
button release 9 
```

button 4
```
button press   8 
button release 8 
```

button 3
```
button press   3 
button release 3 
```

button 2 ??
```
button press   2 
button press   1 
button release 1 button release 2 
```

button 1
```
button press   1 
button release 1 
```

press scroll wheel
```
button press   10 
button release 10 
```

scroll wheel left?
```
button press   1 
button release 1 
```

scroll wheel right ????
```
button press   2 
button press   8 
button press   9 
button press   10 
button press   11 
button press   12 
button press   1 
button release 1 button release 2 
button release 8 
button release 9 
button release 10 
button release 11 
button release 12 
```

$ xinput list-props "UC-Logic MVP20U+RH Pad"
```
Device 'UC-Logic MVP20U+RH Pad':
	Device Enabled (149):	1
	Coordinate Transformation Matrix (151):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (276):	0
	Device Accel Constant Deceleration (277):	1.000000
	Device Accel Adaptive Deceleration (278):	1.000000
	Device Accel Velocity Scaling (279):	10.000000
	Device Product ID (265):	21827, 77
	Device Node (266):	"/dev/input/event16"
	Evdev Axis Inversion (280):	0, 0
	Evdev Axis Calibration (281):	<no items>
	Evdev Axes Swap (282):	0
	Axis Labels (283):	"Abs X" (298), "Abs Y" (299), "Abs Misc" (309)
	Button Labels (284):	"Button 0" (297), "Button 1" (301), "Button 2" (302), "Button Wheel Up" (155), "Button Wheel Down" (156), "Button Horiz Wheel Left" (157), "Button Horiz Wheel Right" (158), "Button 3" (303), "Button 4" (304), "Button 5" (305), "Button 6" (306), "Button 7" (307), "Button 8" (308)
	Evdev Middle Button Emulation (285):	1
	Evdev Middle Button Timeout (286):	50
	Evdev Third Button Emulation (287):	0
	Evdev Third Button Emulation Timeout (288):	1000
	Evdev Third Button Emulation Button (289):	3
	Evdev Third Button Emulation Threshold (290):	20
	Evdev Wheel Emulation (291):	0
	Evdev Wheel Emulation Axes (292):	0, 0, 4, 5
	Evdev Wheel Emulation Inertia (293):	10
	Evdev Wheel Emulation Timeout (294):	200
	Evdev Wheel Emulation Button (295):	4
	Evdev Drag Lock Buttons (296):	0
```

Any Suggestions?

Dracozny, 2016-03-11:
Switching to the wacom drivers netted similar results.

xinput test "UC-Logic MVP20U+RH Pad pad"

Button 5
```
button press   9 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 

button release 9 a[3]=0 a[4]=0 a[5]=0 
```

Button 4
```
button press   8 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 

button release 8 a[3]=0 a[4]=0 a[5]=0 
```

Button 3
```
button press   3 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 

button release 3 a[3]=0 a[4]=0 a[5]=0 

```
Button 2
```
button press   2 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
UC-Logic MVP20U+RH Pad
button release 2 a[3]=0 a[4]=0 a[5]=0 
```

Button 1 (note that button press results in nothing on the screen, only button up)
```
button press   1 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 1 a[3]=0 a[4]=0 a[5]=0 
```

Scroll Button
```
button press   10 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 

button release 10 a[3]=0 a[4]=0 a[5]=0 
```

Wheel Left (this is one click)
```
button press   1 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 1 a[3]=0 a[4]=0 a[5]=0 
```

Wheel Right (this is one click)
```
button press   1 a[3]=0 a[4]=0 a[5]=0 
button press   2 a[3]=0 a[4]=0 a[5]=0 
button press   3 a[3]=0 a[4]=0 a[5]=0 
button press   8 a[3]=0 a[4]=0 a[5]=0 
button press   9 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button press   10 a[3]=0 a[4]=0 a[5]=0 
button press   11 a[3]=0 a[4]=0 a[5]=0 
button press   12 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 1 a[3]=0 a[4]=0 a[5]=0 
button release 2 a[3]=0 a[4]=0 a[5]=0 
button release 3 a[3]=0 a[4]=0 a[5]=0 
button release 8 a[3]=0 a[4]=0 a[5]=0 
button release 9 a[3]=0 a[4]=0 a[5]=0 
motion a[3]=0 a[4]=0 a[5]=0 
button release 10 a[3]=0 a[4]=0 a[5]=0 
button release 11 a[3]=0 a[4]=0 a[5]=0 
button release 12 a[3]=0 a[4]=0 a[5]=0 
```

xinput list-props "UC-Logic MVP20U+RH Pad pad"
```
Device 'UC-Logic MVP20U+RH Pad pad':
	Device Enabled (149):	1
	Coordinate Transformation Matrix (151):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (276):	0
	Device Accel Constant Deceleration (277):	1.000000
	Device Accel Adaptive Deceleration (278):	1.000000
	Device Accel Velocity Scaling (279):	10.000000
	Device Node (266):	"/dev/input/event4"
	Wacom Serial IDs (308):	77, 0, 15, 17, 15
	Wacom Serial ID binding (309):	0
	Wacom Pressure Threshold (310):	0
	Wacom Sample and Suppress (311):	2, 4
	Wacom Enable Touch (312):	0
	Wacom Enable Touch Gesture (314):	0
	Wacom Touch Gesture Parameters (315):	0, 0, 250
	Wacom Tool Type (316):	"PAD" (322)
	Wacom Button Actions (317):	"Wacom button action 0" (318), "Wacom button action 1" (319), "Wacom button action 2" (320), "None" (0), "None" (0), "None" (0), "None" (0), "Wacom button action 3" (323), "Wacom button action 4" (324), "Wacom button action 5" (325), "Wacom button action 6" (326), "Wacom button action 7" (327), "Wacom button action 8" (328)
	Wacom button action 0 (318):	1572865
	Wacom button action 1 (319):	1572866
	Wacom button action 2 (320):	1572867
	Wacom button action 3 (323):	1572872
	Wacom button action 4 (324):	1572873
	Wacom button action 5 (325):	1572874
	Wacom button action 6 (326):	1572875
	Wacom button action 7 (327):	1572876
	Wacom button action 8 (328):	1572877
	Wacom Strip Buttons (329):	"Wacom strip action 0" (330), "Wacom strip action 1" (331), "Wacom strip action 2" (332), "Wacom strip action 3" (333)
	Wacom strip action 0 (330):	1572868
	Wacom strip action 1 (331):	1572869
	Wacom strip action 2 (332):	1572868
	Wacom strip action 3 (333):	1572869
	Wacom Wheel Buttons (334):	"Wacom wheel action 0" (335), "Wacom wheel action 1" (336), "Wacom wheel action 2" (337), "Wacom wheel action 3" (338), "Wacom wheel action 4" (339), "Wacom wheel action 5" (340)
	Wacom wheel action 0 (335):	1572869
	Wacom wheel action 1 (336):	1572868
	Wacom wheel action 2 (337):	1572868
	Wacom wheel action 3 (338):	1572869
	Wacom wheel action 4 (339):	1572868
	Wacom wheel action 5 (340):	1572869
	Device Product ID (265):	21827, 77
	Wacom Debug Levels (321):	0, 0
```

Dracozny, 2016-03-13:
extra info if needed:

Linux Mint 17.3 amd64
uname -r
`3.19.0-32-generic`

uclogic-probe 001 002
```
M 55 00 43 00 2D 00 4C 00 6F 00 67 00 69 00 63 00
P 4D 00 56 00 50 00 32 00 30 00 55 00 2B 00 52 00 48 00
S 64 0C 03 C2 85 6A 49 03 00 FF 07 D0 07
S 65 04 03 09 A0
S 6E 04 03 31 00
S 79 18 03 46 00 34 00 30 00 31 00 20 00 53 00 65 00 72 00 69 00 65 00 73 00
S 7A 08 03 01 06 01 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
```
uclogic-probe 001 002 | uclogic-decode 
  ```
Manufacturer: UC-Logic
       Product: MVP20U+RH
         Max X: 34242
         Max Y: 18794
  Max pressure: 2047
    Resolution: 2000
Internal model: F401 Series
Buttons status: HK On

```

Dracozny, 2016-03-13:
Using usbhid-dump, things seem more logical
```

001:002:000:DESCRIPTOR         1457848135.715586
 05 0D 09 02 A1 01 85 07 09 20 A1 00 09 42 09 44
 09 45 09 3C 09 43 09 44 15 00 25 01 75 01 95 06
 81 02 09 32 75 01 95 01 81 02 81 03 05 01 09 30
 75 10 95 01 A4 55 0D 65 13 35 00 46 E1 42 27 C2
 85 00 00 81 02 09 31 46 B5 24 26 6A 49 81 02 B4
 05 0D 09 30 26 FF 07 46 FF 07 55 00 66 11 E1 75
 10 81 02 C0 C0 05 01 09 02 A1 01 85 09 09 01 A1
 00 05 09 19 01 29 03 15 00 25 01 95 03 75 01 81
 02 95 05 81 01 05 01 09 30 75 10 95 01 A4 35 00
 46 00 08 26 00 08 81 02 09 31 46 00 08 26 00 08
 81 02 B4 05 0D 09 30 26 FF 07 46 FF 07 75 10 81
 02 C0 C0

```

scroll right
```
001:002:000:STREAM             1457848679.254278
 07 E0 01 10 FF 00 00 00

001:002:000:STREAM             1457848679.264217
 07 E0 01 10 00 00 00 00
```

scroll left
```
001:002:000:STREAM             1457848687.933859
 07 E0 01 10 01 00 00 00

001:002:000:STREAM             1457848687.945845
 07 E0 01 10 00 00 00 00

```
scroll press down
```
001:002:000:STREAM             1457848700.043328
 07 E0 01 01 20 00 00 00

```
scroll press up
```
001:002:000:STREAM             1457848700.197311
 07 E0 01 01 00 00 00 00

```
button 1 down
```
001:002:000:STREAM             1457848708.762899
 07 E0 01 01 01 00 00 00
```

button 1 up
```
001:002:000:STREAM             1457848723.436249
 07 E0 01 01 00 00 00 00

```
button 2 down
```
001:002:000:STREAM             1457848734.339738
 07 E0 01 01 02 00 00 00

```
button 2 up
```
001:002:000:STREAM             1457848744.121342
 07 E0 01 01 00 00 00 00
```

button 3 down
```
001:002:000:STREAM             1457848753.098950
 07 E0 01 01 04 00 00 00
```

button 3 up
```
001:002:000:STREAM             1457848762.418533
 07 E0 01 01 00 00 00 00

```
button 4 down
```
001:002:000:STREAM             1457848772.724098
 07 E0 01 01 08 00 00 00

```
button 4 up
```
001:002:000:STREAM             1457848779.911724
 07 E0 01 01 00 00 00 00
```

button 5 down
```
001:002:000:STREAM             1457848789.053357
 07 E0 01 01 10 00 00 00
```

button 5 up
```
001:002:000:STREAM             1457848796.682963
 07 E0 01 01 00 00 00 00
```

Dracozny, 2016-03-13:
Full diagnostics emailed to DIGImend-devel "Tablet diagnostics Yiynova MVP20u+RH"

spbnick, 2016-06-18:
Thank you for the diagnostics, @Dracozny, I'm looking at them right now. Expect an answer within a week.

spbnick, 2016-06-25:
Alright, the protocol seems simple and I can the wheel support to the driver. However, I would like to make it compatible with the Wacom drivers, and for that I'm waiting for an answer from one of the developers, recommending the best way to do it.

spbnick, 2016-06-27:
I got a reply from the Wacom developer and can try implementing the wheel support, although I won't have much free time until the weekend. Stay tuned.

Dracozny, 2016-07-31:
Sounds good, thank you.

spbnick, 2016-09-03:
I have to regretfully inform you, that I (the project maintainer) am [leaving the project](http://spbnick.github.io/2016/07/31/Wrapping-up-DIGImend-work.html). I'm no longer working on supporting new tablets, nor am I supporting users. I'm still accepting pull requests and work on HOWTOs and syncing with upstream, but only until November 2016. Members of the community may still help other users. Anyone desiring to step in is welcome to contact me.