DIGImend/digimend-kernel-drivers #29: Bosto Kingtee

antonvdh, 2016-03-16:
Will this driver work for the Bosto Kingtee 22U Mini with pressure sensitivity?

spbnick, 2016-03-16:
Hi Antonvdh,

Not likely. However, there appears to be a driver for at least some Bosto models (which appear to be Hanwang-based): https://github.com/aidyw/bosto-2g-linux-kernel-module - try asking there.

antonvdh, 2016-03-16:
Yes I know but  bosto-2g linux kernel driver is made for kernel 3.1 so it
is outdated.
We are already at kernel 4.2 !

spbnick, 2016-03-16:
Hmm, the driver isn't that big and it shouldn't be a problem to update it to work on 4.2. There's also the matter of compatibility with that specific tablet. Also, I won't have the time for the tablet work for at least two more months (still studying while also working).

What kind of issues do you get using it with newer kernels?

antonvdh, 2016-03-16:
Pencil cursor is stick to the right edge from my bosto kingtee 22U  tablet.
Also no pressure sensitivity.

antonvdh, 2016-03-17:
Hello Nikolai,

Hope you can find some time.
I have some info about the bosto kingtee 22U mini digitizer
The 14WA, 22HD and 22HDX seem to use digitizers based on Hanvon technology .
None of these drivers work for the 22U mini.
The 22U mini seems to use a digitizer based on Waltop technology.

spbnick, 2016-09-03:
I have to regretfully inform you, that I (the project maintainer) am [leaving the project](http://spbnick.github.io/2016/07/31/Wrapping-up-DIGImend-work.html). I'm no longer working on supporting new tablets, nor am I supporting users. I'm still accepting pull requests and work on HOWTOs and syncing with upstream, but only until November 2016. Members of the community may still help other users. Anyone desiring to step in is welcome to contact me.