DIGImend/digimend-kernel-drivers #31: Compatibility issues with kernel 4.4+

SketchStick, 2016-04-20:
I have a Yiynova MVP22U (V4) and I'm running Arch with all the latest packages.

With 4.5.1, I was finding my tablet not listed in xinput and not responding at all. In some circumstances I had it listed, but the identifiers and a whole bunch of features were missing and the lack of sub-pixel precision was making any strokes jittery.

I tried a wide range of kernel versions and found that the 4.4 release was when it broke, 4.3.3 still working perfectly and being my fall-back for now. To make sure this wasn't an Arch issue I also tested this on Debian in a VM and had the same result...

spbnick, 2016-04-20:
Thanks for the report! I'll try to look at this this week.

spbnick, 2016-04-24:
First of all, I see that it simply doesn't build on newer kernels, because the HID driver interface has changed slightly in v4.4 (to be fixed). So I suspect that you simply didn't have the driver installed for the newer version. How did you verify it was there?

spbnick, 2016-04-24:
I mean I'll be fixing my driver to use the updated HID driver interface in the kernel.

SketchStick, 2016-04-24:
Well actually over here it appears to build, at least make didn't throw any errors and the module appears to load.
http://pastebin.com/raw/t3GWxwAS

SketchStick, 2016-04-24:
Actually the issues I'm having with the in-kernel version could just be due to the lack of Yiynova support... Here's the xinput: http://pastebin.com/raw/A3vdz8DX

spbnick, 2016-04-24:
Yes, the difference is due to the in-kernel driver being used. Do you have dmesg output for when you plug in the tablet on the broken kernel?

SketchStick, 2016-04-24:
Here's the dmesg when plugging it in, 4.4.8 with module loaded: http://pastebin.com/raw/4VHad8fQ

spbnick, 2016-04-24:
Still, since the module builds (even with a warning), it should install and load. I'm not sure what would exactly happen when that module is used, but I suspect it should either work fine, or completely fail to initialize. I'm not sure if another (generic) driver will be retried on failure (doesn't look like it).

spbnick, 2016-04-24:
The dmesg output you provide shows the generic HID driver loading (which can't handle the tablet properly). So it either doesn't install, or is not rebound by hid-rebind (which might be a separate issue).

SketchStick, 2016-04-24:
Well here's the dmesg output for the others, just for the hell of it: http://pastebin.com/raw/C99UVLDk

spbnick, 2016-04-24:
Yeah, the older kernel output shows the tablet being rebound to the out-of-tree driver. So, if the driver indeed installed with the newer kernel, then rebinding of the module is going wrong.

spbnick, 2016-04-24:
I'm still suspicious why the module hasn't loaded on the newer kernel, but please try this patch which should fix the compatibility:

https://gist.github.com/spbnick/a7ec56164bc830ff6f6bc56281c60e18

Meanwhile I'll work on making the driver build properly on both the older and the newer kernels.

SketchStick, 2016-04-24:
Had to manually apply the patch but it's all working! 4.4.8 with the module loaded, shows up in xinput and draws nice clean strokes.

spbnick, 2016-04-24:
Nice, thanks for testing!

SketchStick, 2016-04-24:
Thank you for the wonderful driver!

I was also wondering if you had any plans to push a new version to the kernel? It seems like you've added quite a few devices and it'd be nice to have those working out of the box.

spbnick, 2016-04-24:
Yes, I have a plan to contribute the recent changes to the kernel, but not until I finish my Finnish courses which take almost all of my free time. That can happen in about a month.

closed 2016-04-24 by spbnick

SketchStick, 2016-04-24:
Just a final confirmation that the latest commit works perfectly under 4.5.1! All the best with your courses and thanks again for everything.

spbnick, 2016-04-24:
You're welcome :)

commit b2ae2551 (2016-04-24): Support HID interface in v4.4.0 and newer kernels

Return int from input_configured with v4.4.0 and older kernels to
reflect the change in the HID driver interface.

Resolves #31