DIGImend/digimend-kernel-drivers #33: Huion H610  PRO wheel emulation

RomuloPBenedetti, 2016-05-23:
Hi, I've tried :

    xinput set-int-prop $deviceNumb "Evdev Wheel Emulation" 8 1 
    xinput set-int-prop $deviceNumb "Evdev Wheel Emulation Button" 8 2

Both in the HUION mouse and pen ID, but both appear not to have any effect, the values are set as i've checked it with `-list-props`, did it on my Logitech mouse and there it works. Any idea? :cry:

spbnick, 2016-06-18:
Hi @RomuloPBenedetti,

Please post the output of "xinput list-props $deviceNumb".

RomuloPBenedetti, 2016-06-18:
Sure @spbnick;

    Device 'HUION PenTablet Pen':
        Device Enabled (138):   1
        Coordinate Transformation Matrix (140): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (271):     0
        Device Accel Constant Deceleration (272):       1.000000
        Device Accel Adaptive Deceleration (273):       1.000000
        Device Accel Velocity Scaling (274):    10.000000
        Device Product ID (258):        9580, 110
        Device Node (259):      "/dev/input/event16"
        Evdev Axis Inversion (275):     0, 0
        Evdev Axis Calibration (276):   <no items>
        Evdev Axes Swap (277):  0
        Axis Labels (278):      "Abs X" (448), "Abs Y" (449), "Abs Pressure" (450)
        Button Labels (279):    "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Wheel Up" (144), "Button Wheel Down" (145), "Button Horiz Wheel Left" (146), "Button Horiz Wheel Right" (147)
        Evdev Scrolling Distance (280): 0, 0, 0
        Evdev Middle Button Emulation (281):    0
        Evdev Middle Button Timeout (282):      50
        Evdev Third Button Emulation (283):     0
        Evdev Third Button Emulation Timeout (284):     1000
        Evdev Third Button Emulation Button (285):      3
        Evdev Third Button Emulation Threshold (286):   20
        Evdev Wheel Emulation (287):    0
        Evdev Wheel Emulation Axes (288):       0, 0, 4, 5
        Evdev Wheel Emulation Inertia (289):    10
        Evdev Wheel Emulation Timeout (290):    200
        Evdev Wheel Emulation Button (291):     4
        Evdev Drag Lock Buttons (292):  0

    Device 'HUION PenTablet Mouse':
        Device Enabled (138):   1
        Coordinate Transformation Matrix (140): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (271):     0
        Device Accel Constant Deceleration (272):       1.000000
        Device Accel Adaptive Deceleration (273):       1.000000
        Device Accel Velocity Scaling (274):    10.000000
        Device Product ID (258):        9580, 110
        Device Node (259):      "/dev/input/event17"
        Evdev Axis Inversion (275):     0, 0
        Evdev Axes Swap (277):  0
        Axis Labels (278):      "Rel X" (148), "Rel Y" (149), "Rel Horiz Wheel" (268), "Rel Vert Wheel" (270)
        Button Labels (279):    "Button Left" (141), "Button Middle" (142), "Button Right" (143), "Button Wheel Up" (144), "Button Wheel Down" (145), "Button Horiz Wheel Left" (146), "Button Horiz Wheel Right" (147), "Button Side" (263), "Button Extra" (264), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261)
        Evdev Scrolling Distance (280): 1, 1, 1
        Evdev Middle Button Emulation (281):    0
        Evdev Middle Button Timeout (282):      50
        Evdev Third Button Emulation (283):     0
        Evdev Third Button Emulation Timeout (284):     1000
        Evdev Third Button Emulation Button (285):      3
        Evdev Third Button Emulation Threshold (286):   20
        Evdev Wheel Emulation (287):    0
        Evdev Wheel Emulation Axes (288):       0, 0, 4, 5
        Evdev Wheel Emulation Inertia (289):    10
        Evdev Wheel Emulation Timeout (290):    200
        Evdev Wheel Emulation Button (291):     4
        Evdev Drag Lock Buttons (292):  0

    Device 'HUION PenTablet Consumer Control':
        Device Enabled (138):   1
        Coordinate Transformation Matrix (140): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (271):     0
        Device Accel Constant Deceleration (272):       1.000000
        Device Accel Adaptive Deceleration (273):       1.000000
        Device Accel Velocity Scaling (274):    10.000000
        Device Product ID (258):        9580, 110
        Device Node (259):      "/dev/input/event19"
        Evdev Axis Inversion (275):     0, 0
        Evdev Axes Swap (277):  0
        Axis Labels (278):      "Rel X" (148), "Rel Y" (149), "Rel Horiz Wheel" (268)
        Button Labels (279):    "Button 0" (262), "Button Unknown" (261), "Button Unknown" (261), "Button Wheel Up" (144), "Button Wheel Down" (145), "Button Horiz Wheel Left" (146), "Button Horiz Wheel Right" (147)
        Evdev Scrolling Distance (280): 1, 1, 1
        Evdev Middle Button Emulation (281):    0
        Evdev Middle Button Timeout (282):      50
        Evdev Third Button Emulation (283):     0
        Evdev Third Button Emulation Timeout (284):     1000
        Evdev Third Button Emulation Button (285):      3
        Evdev Third Button Emulation Threshold (286):   20
        Evdev Wheel Emulation (287):    0
        Evdev Wheel Emulation Axes (288):       0, 0, 4, 5
        Evdev Wheel Emulation Inertia (289):    10
        Evdev Wheel Emulation Timeout (290):    200
        Evdev Wheel Emulation Button (291):     4
        Evdev Drag Lock Buttons (292):  0

spbnick, 2016-06-18:
Thanks! I see that your "Evdev Wheel Emulation Button" is just "4", whereas IIRC it should be at least two numbers. My IBM TrackPoint has four, for example: "6, 7, 4, 5".

See ftp://www.x.org/pub/X11R7.5/doc/man/man4/evdev.4.html

RomuloPBenedetti, 2016-06-18:
but how I put 4 numbers there if as stated:

    Evdev Wheel Emulation Button
    1 8-bit value, allowed range 0-32, 0 disables the button.

If I do as I pointed on my first post, to a mouse, it works fine. But even setting it to "2" in huion, it did not behaves as the mouse... it simply did not scroll.

RomuloPBenedetti, 2016-06-18:
for example, this mouse scroll normally :

    Device 'Logitech USB Receiver':
        Device Enabled (138):   1
        Coordinate Transformation Matrix (140): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (271):     0
        Device Accel Constant Deceleration (272):       1.000000
        Device Accel Adaptive Deceleration (273):       1.000000
        Device Accel Velocity Scaling (274):    10.000000
        Device Product ID (258):        1133, 50484
        Device Node (259):      "/dev/input/event8"
        Evdev Axis Inversion (275):     0, 0
        Evdev Axes Swap (277):  0
        Axis Labels (278):      "Rel X" (148), "Rel Y" (149), "Rel Horiz Wheel" (268), "Rel Dial" (269), "Rel Vert Wheel" (270)
        Button Labels (279):    "Button Left" (141), "Button Middle" (142), "Button Right" (143), "Button Wheel Up" (144), "Button Wheel Down" (145), "Button Horiz Wheel Left" (146), "Button Horiz Wheel Right" (147), "Button Side" (263), "Button Extra" (264), "Button Forward" (265), "Button Back" (266), "Button Task" (267), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261)
        Evdev Scrolling Distance (280): 1, 1, 1
        Evdev Middle Button Emulation (281):    0
        Evdev Middle Button Timeout (282):      50
        Evdev Third Button Emulation (283):     0
        Evdev Third Button Emulation Timeout (284):     1000
        Evdev Third Button Emulation Button (285):      3
        Evdev Third Button Emulation Threshold (286):   20
        Evdev Wheel Emulation (287):    1
        Evdev Wheel Emulation Axes (288):       0, 0, 4, 5
        Evdev Wheel Emulation Inertia (289):    10
        Evdev Wheel Emulation Timeout (290):    200
        Evdev Wheel Emulation Button (291):     2
        Evdev Drag Lock Buttons (292):  0

The Huion pen just stop pasting, but did not scroll with this setting:

    Device 'HUION PenTablet Pen':
        Device Enabled (138):   1
        Coordinate Transformation Matrix (140): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (271):     0
        Device Accel Constant Deceleration (272):       1.000000
        Device Accel Adaptive Deceleration (273):       1.000000
        Device Accel Velocity Scaling (274):    10.000000
        Device Product ID (258):        9580, 110
        Device Node (259):      "/dev/input/event16"
        Evdev Axis Inversion (275):     0, 0
        Evdev Axis Calibration (276):   <no items>
        Evdev Axes Swap (277):  0
        Axis Labels (278):      "Abs X" (448), "Abs Y" (449), "Abs Pressure" (450)
        Button Labels (279):    "Button Unknown" (261), "Button Unknown" (261), "Button Unknown" (261), "Button Wheel Up" (144), "Button Wheel Down" (145), "Button Horiz Wheel Left" (146), "Button Horiz Wheel Right" (147)
        Evdev Scrolling Distance (280): 0, 0, 0
        Evdev Middle Button Emulation (281):    0
        Evdev Middle Button Timeout (282):      50
        Evdev Third Button Emulation (283):     0
        Evdev Third Button Emulation Timeout (284):     1000
        Evdev Third Button Emulation Button (285):      3
        Evdev Third Button Emulation Threshold (286):   20
        Evdev Wheel Emulation (287):    1
        Evdev Wheel Emulation Axes (288):       0, 0, 4, 5
        Evdev Wheel Emulation Inertia (289):    10
        Evdev Wheel Emulation Timeout (290):    200
        Evdev Wheel Emulation Button (291):     2
        Evdev Drag Lock Buttons (292):  0

spbnick, 2016-06-26:
I just tried these commands and they worked on my H610, both for vertical and
horizontal scrolling using the pen's bottom side button:

    xinput set-prop "10594 Pen" "Evdev Wheel Emulation" 1
    xinput set-prop "10594 Pen" "Evdev Wheel Emulation Axes" 6 7 4 5
    xinput set-prop "10594 Pen" "Evdev Wheel Emulation Button" 2

"10594" is my tablet's name. For your tablet these commands should be:

    xinput set-prop "HUION PenTablet Pen" "Evdev Wheel Emulation" 1
    xinput set-prop "HUION PenTablet Pen" "Evdev Wheel Emulation Button" 2
    xinput set-prop "HUION PenTablet Pen" "Evdev Wheel Emulation Axes" 6 7 4 5

If you need just the vertical scrolling use this command instead:

    xinput set-prop "HUION PenTablet Pen" "Evdev Wheel Emulation Axes" 0 0 4 5

Hope this helps.

closed 2016-07-17 by spbnick