DIGImend/digimend-kernel-drivers #34: Not fully working with recent kernel (4.5.4-1) on Arch Linux

davethecipo, 2016-05-29:
Hello,

I have a WP8060U tablet

    Bus 003 Device 008: ID 5543:0005 UC-Logic Technology Corp. Tablet WP8060U

Kernel:

    [davide@fisso ~]$ uname -r
    4.5.4-1-ARCH

I run the following

    cd ~/Downloads
    git clone https://github.com/DIGImend/digimend-kernel-drivers.git
    cd digimend-kernel-drivers
    make
    sudo make install

A reboot follows. Expected result: fully working pen on painting programs.

Actual result:  the cursor moves in GIMP but no painting is done. In Krita 2.9/3.0 RC, the pen draws but no pressure is detected.

Do you need any other information? I might add that the deb package works on Kubuntu which uses an older kernel. I directly run `git clone` because of the message of commit b2ae255155d48b3da7be2302468bc94bd8a7ae3d.

spbnick, 2016-05-29:
Hi Dave,

You don't need digimend-kernel-drivers to use your tablet. It's been supported in the upstream kernel for a long time. However, you may need to configure your applications. See this:

http://digimend.github.io/support/

davethecipo, 2016-05-29:
Hi,

I removed digimend-kernel-driver from kubuntu and as you said the evdev driver does its job there. I can't say the same for Arch Linux, I'll probably investigate further.

gaharris, 2016-05-30:
Hi,

Upgrading my Arch system this morning has caused a problem that looks the same.
I'm using a Huion K58, which was working fine this morning, before the upgrade.
I now get zero pressure, but everything else looks normal.

Here's the output from `xinput` while pressing the stylus tip down firmly:
~~~
HUION Pentablet Pen                             id=20   [slave  pointer  (2)]
        Reporting 4 classes:
                Class originated from: 20. Type: XIButtonClass
                Buttons supported: 7
                Button labels: "Button Unknown" "Button Unknown" "Button Unknown" "Button Wheel Up" "Button Wheel Down" "Button Horiz Wheel Left" "Button Horiz Wheel Right"
                Button state: 1
                Class originated from: 20. Type: XIValuatorClass
                Detail for Valuator 0:
                  Label: Abs X
                  Range: 0.000000 - 32000.000000
                  Resolution: 157000 units/m
                  Mode: absolute
                  Current value: 20892.000000
                Class originated from: 20. Type: XIValuatorClass
                Detail for Valuator 1:
                  Label: Abs Y
                  Range: 0.000000 - 20000.000000
                  Resolution: 157000 units/m
                  Mode: absolute
                  Current value: 269.000000
                Class originated from: 20. Type: XIValuatorClass
                Detail for Valuator 2:
                  Label: Abs Pressure
                  Range: 0.000000 - 2047.000000
                  Resolution: 0 units/m
                  Mode: absolute
                  Current value: 0.000000
~~~
Note the pressure being registered as "Button state: 1" but not as "Abs Pressure" (so not a hardware issue) and sensible X and Y coordinates.

I think I've traced the problem to package xf86-input-evdev version 2.10.2-1
**My workaround: downgrade to xf86-input-evdev version 2.10.1-3**

I came to this conclusion after:
Downgrading to kernel versions 4.5.1-1, 4.4.3-1 and 4.3.3-3 (the version that worked this morning) with no effect (so not the kernel).
Downgrading xf86-input-evdev to version 2.10.1-3 and libevdev to version 1.4.6-1 which fixed the problem.
Upgrading libevdev back to version 1.5.1-1; the problem stayed fixed.
Upgrading the kernel back to version 4.5.4-1 and rebooting; the problem stayed fixed.

I hope this is useful to anyone else looking for a workaround.

spbnick, 2016-05-30:
@gaharris Thank you. Please report the issue to your distribution as well, if you can.

gaharris, 2016-05-30:
I think the [latest commit](https://cgit.freedesktop.org/xorg/driver/xf86-input-evdev/commit/?id=9a6952dafe7241d4987eedd4637e5a9785d7ec23) on xf86-input-evdev is a fix for this issue, so hopefully this will be fixed in the next update.

davethecipo, 2016-06-05:
Solved with xf86-input-evdev 2.10.3-1.

closed 2016-06-18 by spbnick