DIGImend/digimend-kernel-drivers #35: Huion W58 - New Digimend Kernel Drivers made it stop working?

lordamit, 2016-06-04:
Hi all,
I bought a Huion W58 today, after checking that Digimend supports it in Linux. With great hope, I plugged it in, and it worked as mouse, it worked great for drawing and everything - without me doing anything.

Then I decided to change the button mappings, and since  I could not find any way - I thought installing the Digimend drivers may help me.

I installed the latest debian driver from here: https://github.com/DIGImend/digimend-kernel-drivers/releases 
I made sure that I had linux-headers preinstalled beforehand.

After that, I relogged in - and the tablet pen stopped working. I could not drag the mouse using the tablet pen.

Here are some more details:
```
OS: Ubuntu 16.04 X64
uname -a 
Linux amit-laptop 4.4.0-22-generic #40-Ubuntu SMP Thu May 12 22:03:46 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ SIGMACHIP USB Keyboard                  	id=11	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=14	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ SIGMACHIP USB Keyboard                  	id=10	[slave  keyboard (3)]
    ↳ HP Truevision HD                        	id=12	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=13	[slave  keyboard (3)]
    ↳ HP Wireless hotkeys                     	id=15	[slave  keyboard (3)]
    ↳ HP WMI hotkeys                          	id=16	[slave  keyboard (3)]
```

Please note that before installing the drivers, xinput listed the PenTablet and some more under it. 

```
lsusb
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 04f2:b2f8 Chicony Electronics Co., Ltd 
Bus 001 Device 003: ID 8087:07da Intel Corp. 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 1c4f:0026 SiGma Micro Keyboard
Bus 003 Device 006: ID 256c:006e  
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Please tell me what I should do at this stage :smile: 

Looking forward to your help.

lordamit, 2016-06-04:
BTW,uninstalling seems to enable the pentablet again. 

```
sudo dpkg -r digimend-dkms
```
And then, xinput gives this: 

```
xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ PenTablet Pen                           	id=10	[slave  pointer  (2)]
⎜   ↳ PenTablet Mouse                         	id=11	[slave  pointer  (2)]
⎜   ↳ PenTablet Consumer Control              	id=13	[slave  pointer  (2)]
⎜   ↳ SIGMACHIP USB Keyboard                  	id=16	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=19	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ PenTablet Keyboard                      	id=12	[slave  keyboard (3)]
    ↳ PenTablet System Control                	id=14	[slave  keyboard (3)]
    ↳ SIGMACHIP USB Keyboard                  	id=15	[slave  keyboard (3)]
    ↳ HP Truevision HD                        	id=17	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=18	[slave  keyboard (3)]
    ↳ HP Wireless hotkeys                     	id=20	[slave  keyboard (3)]
    ↳ HP WMI hotkeys                          	id=21	[slave  keyboard (3)]
```

lordamit, 2016-06-04:
Should I try the steps described here? 

https://github.com/DIGImend/digimend-kernel-drivers/issues/26#issuecomment-192761796

lordamit, 2016-06-04:
ps: I am happy with the tablet pen performance without using the digimend drivers btw. It can pick up the pressure, 3 buttons and everything by default in Ubuntu 16.04. 

Just wanted to make sure that you guys know about the part after installing Digimend drivers.

spbnick, 2016-06-05:
Hi Amit,

Thanks for your report. I see that at least uninstallation worked fine :)
I see no reason for you to use digimend-kernel-drivers as your kernel has the support you need.
I'm not sure what you mean by "button remapping", but if you simply want to move the meaning of mouse buttons around, then you should be able to do it with "xinput set-button-map".

If you still would like to help with figuring out what's wrong with digimend-kernel-drivers, for the start, please install them again and post the output of "dmesg" after plugging your tablet in.

Thank you!

lordamit, 2016-06-05:
Haha, yes - lucky for me :) 

I will be using the xinput set-button-map and see what I can do. 

Since my current kernel provides out of the box support for the PenTablet - I don't think I will pursue the digimend part anymore, since it already correctly detects pretty much everything.  

Since it is my first ever experience of using a Tablet, I won't even know if it is not working properly.. except the most obvious ones!

Thank you for your response. If you require any more information, and I will definitely try to to provide.

Closing this issue for now. Take care, and thanks! :smile:

closed 2016-06-05 by lordamit