DIGImend/digimend-kernel-drivers #36: Ugee M708

Ascdeath, 2016-06-25:
I don't understand how this works I couldn't even get my Ugee M708 to work in linux the pen would at least kinda move around before I installed this but now it does nothing at all how do I  get it to work do I need to do something special I installed headers and the experimental drivers but still nothing I've tried installing and reinstalling can someone help I'm new to linux in general

Ascdeath, 2016-06-25:
Edit: it seems to be working fine in Mypaint no pressure in gimp or Krita but when I hover the pen the cursor disappears so I can't see what i'm doing when I try to draw

alglez, 2016-06-26:
Actually I'm experiencing issues with this version (kernel 4.6.2) with ugee g3 tablet. The tablet motion work fine, but no pressure sensitivity in mypaint or krita, but I should test older version before to know where is the problem and fill a bug report. 

But you should provide more information as kernel (uname -r) evdev version, etc, 

Did you install it from source or are you using evdev in new kernels?

Ascdeath, 2016-06-29:
I guess I installed it from source Idk what evdev is  and how do I provide more info?

alglez, 2016-07-01:
Digimend it is supported in recent kernels, so is good to know if your kernel support already this table, In recent kernels evdev  package is just the way to install this drivers. 

just provide the output from
`uname -a`

spbnick, 2016-09-03:
I have to regretfully inform you, that I (the project maintainer) am [leaving the project](http://spbnick.github.io/2016/07/31/Wrapping-up-DIGImend-work.html). I'm no longer working on supporting new tablets, nor am I supporting users. I'm still accepting pull requests and work on HOWTOs and syncing with upstream, but only until November 2016. Members of the community may still help other users. Anyone desiring to step in is welcome to contact me.