DIGImend/digimend-kernel-drivers #37: Cannot get Huion H610PRO Pad device

snowkeep, 2016-07-03:
This seems to be a very similar problem to Issue #26 opened by Sunaku.  Pressure sensitivy works.  Buttons do not.  I cannot get the "Huion PenTablet Pad pad" device.  Any other suggestion?  Here is my setup and results.

Arch with kernel 4.6.3.  Using in-kernel drivers, not the digimend ones.  The hid-uclogic module is being used.

/etc/X11/xorg.conf.d/52-tablet.conf:
```
Section "InputClass"
  Identifier "Huion on wacom"
  # MatchIsTablet "on"
  MatchProduct "HUION"
  MatchDevicePath "/dev/input/event*"
  Driver "wacom"
EndSection
```

I commented-out the MatchIsTablet section in usr/share/X11/xorg.conf.d/10-evdev.conf, as suggested by kriks57.
Both the rule above and "evdev keyboard catchall" were being triggered when I plugged the tablet in, so I commented out the keyboard catchall and added a MatchProduct "USB Keyboard" entry instead.  Other than a few lines in Xorg.0.log, no difference.

I also tried adding this from https://sourceforge.net/p/digimend/mailman/message/33604572/.
/etc/udev/rules.d/65-digimend.rules:
```
# udev rules for digimend devices

ACTION!="add|change", GOTO="digimend_end"
KERNEL!="event[0-9]*", GOTO="digimend_end"

ENV{ID_BUS}=="usb", ENV{ID_VENDOR_ID}=="256c", ENV{ID_MODEL_ID}=="006e", ENV{ID_INPUT}="1", ENV{ID_INPUT_JOYSTICK}="", ENV{ID_INPUT_TABLET}="1"
ATTRS{name}=="* Pad", ENV{ID_BUS}=="usb", ENV{ID_VENDOR_ID}=="256c", ENV{ID_MODEL_ID}=="006e",  ENV{ID_INPUT_TABLET_PAD}="1"

LABEL="digimend_end"
```

dmesg on plug-in:
```
[   56.951886] usb 5-1.3: new full-speed USB device number 5 using xhci_hcd
[   57.073871] input: HUION PenTablet Pen as /devices/pci0000:00/0000:00:10.0/usb5/5-1/5-1.3/5-1.3:1.0/0003:256C:006E.000C/input/input24
[   57.074049] uclogic 0003:256C:006E.000C: input,hidraw9: USB HID v1.11 Device [HUION PenTablet] on usb-0000:00:10.0-1.3/input0
[   57.074505] input: HUION PenTablet Mouse as /devices/pci0000:00/0000:00:10.0/usb5/5-1/5-1.3/5-1.3:1.1/0003:256C:006E.000D/input/input25
[   57.074757] uclogic 0003:256C:006E.000D: input,hiddev0,hidraw10: USB HID v1.11 Mouse [HUION PenTablet] on usb-0000:00:10.0-1.3/input1
[   57.077266] input: HUION PenTablet Keyboard as /devices/pci0000:00/0000:00:10.0/usb5/5-1/5-1.3/5-1.3:1.2/0003:256C:006E.000E/input/input27
[   57.128621] input: HUION PenTablet Consumer Control as /devices/pci0000:00/0000:00:10.0/usb5/5-1/5-1.3/5-1.3:1.2/0003:256C:006E.000E/input/input28
[   57.128732] input: HUION PenTablet System Control as /devices/pci0000:00/0000:00:10.0/usb5/5-1/5-1.3/5-1.3:1.2/0003:256C:006E.000E/input/input29
[   57.128827] uclogic 0003:256C:006E.000E: input,hidraw11: USB HID v1.11 Keyboard [HUION PenTablet] on usb-0000:00:10.0-1.3/input2
'''

Xorg.0.log on plug-in:
```
[    57.032] (II) config/udev: Adding input device HUION PenTablet Pen (/dev/input/mouse2)
[    57.032] (II) No input driver specified, ignoring this device.
[    57.032] (II) This device may have been added with another device file.
[    57.033] (II) config/udev: Adding input device HUION PenTablet Mouse (/dev/input/mouse3)
[    57.033] (II) No input driver specified, ignoring this device.
[    57.033] (II) This device may have been added with another device file.
[    57.063] (II) config/udev: Adding input device HUION PenTablet Consumer Control (/dev/input/event24)
[    57.063] (**) HUION PenTablet Consumer Control: Applying InputClass "Huion on wacom"
[    57.063] (II) LoadModule: "wacom"
[    57.064] (II) Loading /usr/lib/xorg/modules/input/wacom_drv.so
[    57.065] (II) Module wacom: vendor="X.Org Foundation"
[    57.065] 	compiled for 1.18.3, module version = 0.33.0
[    57.065] 	Module class: X.Org XInput Driver
[    57.065] 	ABI class: X.Org XInput driver, version 22.1
[    57.065] (II) wacom: Driver for Wacom graphics tablets: PenPartner, Graphire,
	Graphire2 4x5, Graphire2 5x7, Graphire3 4x5, Graphire3 6x8,
	Graphire4 4x5, Graphire4 6x8, BambooFun 4x5, BambooFun 6x8,
	Bamboo1 Medium, Graphire4 6x8 BlueTooth, CTL-460, CTH-461, CTL-660,
	CTL-461/S, Bamboo Touch, CTH-460/K, CTH-461/S, CTH-661/S1, CTH-461/L,
	CTH-661/L, Intuos 4x5, Intuos 6x8, Intuos 9x12, Intuos 12x12,
	Intuos 12x18, PTU600, PL400, PL500, PL600, PL600SX, PL550, PL800,
	PL700, PL510, PL710, DTI520, DTF720, DTF720a, DTF521, DTU1931,
	DTU2231, DTU1631, Intuos2 4x5, Intuos2 6x8, Intuos2 9x12,
[... 89 lines cut ...]
[    57.303] (II) HUION PenTablet Pen stylus: hotplugging completed.
[    57.336] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:10.0/usb5/5-1/5-1.3/5-1.3:1.0/0003:256C:006E.000C/input/input24/event21"
[    57.336] (II) XINPUT: Adding extended input device "HUION PenTablet Pen stylus" (type: STYLUS, id 10)
[    57.337] (**) HUION PenTablet Pen stylus: (accel) keeping acceleration scheme 1
[    57.337] (**) HUION PenTablet Pen stylus: (accel) acceleration profile 0
[    57.337] (**) HUION PenTablet Pen stylus: (accel) acceleration factor: 2.000
[    57.337] (**) HUION PenTablet Pen stylus: (accel) acceleration threshold: 4
```

xinput --list
```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Logitech Anywhere MX                    	id=9	[slave  pointer  (2)]
⎜   ↳ HUION PenTablet Pen stylus              	id=10	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ USB Keyboard                            	id=7	[slave  keyboard (3)]
    ↳ USB Keyboard                            	id=8	[slave  keyboard (3)]
```

xsetwacom --list 
```
HUION PenTablet Pen stylus      	id: 10	type: STYLUS    
```

alt165am, 2016-07-06:
Hello! I'm having the same problem on Linux Mint 18 MATE (kernel 4.4.0-21) with a Huion H610pro.
I was using Linux Mint 17.2 XFCE before upgrading and it did recognize my tablet without a problem. I have tried #26 solution but it didn't work. (sorry if i'm not helpful but i'm a n00b on the subject :+1: ).

reanimastudios, 2016-07-13:
Compiling from Source on Debian Sid against GCC 6.1.

make -C /lib/modules/4.6.0-1-amd64/build SUBDIRS=/home/mdriftmeyer/Downloads/Digimend/digimend-kernel-drivers-6 modules
make[1]: Entering directory '/usr/src/linux-headers-4.6.0-1-amd64'
  CC [M]  /home/mdriftmeyer/Downloads/Digimend/digimend-kernel-drivers-6/hid-kye.o
  CC [M]  /home/mdriftmeyer/Downloads/Digimend/digimend-kernel-drivers-6/hid-uclogic.o
  CC [M]  /home/mdriftmeyer/Downloads/Digimend/digimend-kernel-drivers-6/hid-polostar.o
/home/mdriftmeyer/Downloads/Digimend/digimend-kernel-drivers-6/hid-uclogic.c:1101:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .input_configured = uclogic_input_configured,
                      ^
/home/mdriftmeyer/Downloads/Digimend/digimend-kernel-drivers-6/hid-uclogic.c:1101:22: note: (near initialization for ‘uclogic_driver.input_configured’)
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.6.0-1-common/scripts/Makefile.build:302: recipe for target '/home/mdriftmeyer/Downloads/Digimend/digimend-kernel-drivers-6/hid-uclogic.o' failed
make[4]: *** [/home/mdriftmeyer/Downloads/Digimend/digimend-kernel-drivers-6/hid-uclogic.o] Error 1
/usr/src/linux-headers-4.6.0-1-common/Makefile:1446: recipe for target '_module_/home/mdriftmeyer/Downloads/Digimend/digimend-kernel-drivers-6' failed
make[3]: *** [_module_/home/mdriftmeyer/Downloads/Digimend/digimend-kernel-drivers-6] Error 2
Makefile:146: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.6.0-1-amd64'
Makefile:9: recipe for target 'modules' failed
make: *** [modules] Error 2
mdriftmeyer@horus:~/Downloads/Digimend/digimend-kernel-drivers-6$

spbnick, 2016-07-14:
@reanimastudios Please use the latest development version which has your compilation issue fixed: https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip

reanimastudios, 2016-07-16:
Thank you!

Built cleanly and installed. Shouldn't a deb be built against newer 
kernels with these changes? Just a suggestion.

Now to test if a beta/trunk from MyPaint works correctly for freehand brush.

Have a great weekend!

- Marc

spbnick, 2016-07-16:
@reanimastudios You're welcome! There doesn't seem to be a way to have a .deb published for each development commit on GitHub. I'm doing them for releases only. That said, I'll probably do a release soon.

spbnick, 2016-07-17:
@snowkeep The stock kernel doesn't create a pad device yet. Please try using the latest development version of digimend-kernel-drivers.

snowkeep, 2016-07-17:
@spbnick  Thank you.  a git clone of the development version worked.  Closing.

closed 2016-07-17 by snowkeep