DIGImend/digimend-kernel-drivers #38: Deb package didn't work on LinuxMint 17.3

Robert42, 2016-07-11:
I've never installed any previous versions of the driver, this is my first attempt

##### What happened:

- Before driver installation via the deb file: moving the pen over the tablet has no effect
- installing the deb file went without any error messages
- After driver installation via the deb file: moving the pen over the tablet has no effect

##### System information:

- Huion 1060 plus
- LinuxMint 17.3
- Linux Kernel 3.19

spbnick, 2016-07-13:
Hi Robert,

Could you please post the output of `/sbin/modinfo hid-uclogic` and also the output of `dmesg` after you plug the tablet in?

Thank you.

Robert42, 2016-07-13:
Both commands have the same output, before and after I plug in the Tablet, so I don't know, whether it is usefull.

(BTW, I've updated now to Linux Mint 18 (based on Ubuntu 16.04, Kernel 4.4.0-28), still exactly same behavior as with LinuxMint 17.3)

Output of `/sbin/modinfo hid-uclogic`
```
filename:       /lib/modules/4.4.0-28-generic/extra/hid-uclogic.ko
version:        6
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     34A15E7DFFC53AA6F4BEACF
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid
vermagic:       4.4.0-28-generic SMP mod_unload modversions 
```

Do you want the whole output of dmesg, of just the last few lines?

The next think I would try is to compile the driver from source instead using the deb file

spbnick, 2016-07-13:
The modinfo output looks good. Yes, about 50 lines from dmesg should be enough.

Robert42, 2016-07-13:
After trying to install the huion tablet on windows, without windows/the driver detecting any device, I assume it's a hardware problem with the usb connection.

I am contacting Huion and closing this ticket for now.

closed 2016-07-13 by Robert42

spbnick, 2016-07-13:
Alright, good luck!

spbnick, 2016-07-13:
Alright, good luck!