DIGImend/digimend-kernel-drivers #39: UGEE G5 on Archlinux no pressure in MyPaint/Krita

ilovepumpkin, 2016-07-15:
Hi, I am using Archlinux and installed the digimend-kernel-drivers-6 using the source code. My UGEE G5 tablet has no pressure in MyPaint 1.2.0 and Krita 3.0, but works well in GIMP. 

Any other information should I give to help me solve this issue? Thanks in advance!

spbnick, 2016-07-17:
Since it works in GIMP, I would suspect it's likely a configuration problem. However, please send the output of "xinput test" for your tablet pen device where you do a few strokes with the pen.

ilovepumpkin, 2016-07-18:
@spbnick , here is the result when drawing two circles in MyPaint.
===========================================================
[ilovepumpkin@ruige ~]$ xinput test "UGTABLET TABLET G5 9x6"
motion a[0]=19187 a[1]=10958 a[2]=0 
motion a[0]=20467 a[1]=11480 a[2]=0 
motion a[0]=20502 a[1]=11511 a[2]=0 
motion a[0]=20533 a[1]=11547 a[2]=0 
motion a[0]=20560 a[1]=11586 a[2]=0 
motion a[0]=20583 a[1]=11629 a[2]=0 
motion a[0]=20603 a[1]=11673 a[2]=0 
motion a[0]=20621 a[1]=11720 a[2]=0 
motion a[0]=20637 a[1]=11768 a[2]=0 
motion a[0]=20652 a[1]=11819 a[2]=0 
motion a[0]=20663 a[1]=11867 a[2]=0 
motion a[0]=20673 a[1]=11916 a[2]=0 
motion a[0]=20685 a[1]=11964 a[2]=0 
motion a[0]=20697 a[1]=12009 a[2]=0 
motion a[0]=20706 a[1]=12059 a[2]=0 
motion a[0]=20706 a[1]=12101 a[2]=0 
motion a[0]=20715 a[1]=12144 a[2]=0 
motion a[0]=20725 a[1]=12185 a[2]=0 
motion a[0]=20736 a[1]=12228 a[2]=0 
motion a[0]=20743 a[1]=12269 a[2]=0 
motion a[0]=20743 a[1]=12308 a[2]=0 
motion a[0]=20743 a[1]=12351 a[2]=0 
motion a[0]=20725 a[1]=12389 a[2]=0 
motion a[0]=20715 a[1]=12429 a[2]=0 
motion a[0]=20707 a[1]=12462 a[2]=0 
motion a[0]=20700 a[1]=12496 a[2]=0 
motion a[0]=20694 a[1]=12526 a[2]=0 
motion a[0]=20592 a[1]=12776 a[2]=0 
motion a[0]=20592 a[1]=12790 a[2]=0 
motion a[0]=20592 a[1]=12806 a[2]=0 
motion a[0]=20592 a[1]=12828 a[2]=0 
motion a[0]=20584 a[1]=12852 a[2]=0 
motion a[0]=20574 a[1]=12875 a[2]=0 
motion a[0]=20562 a[1]=12897 a[2]=0 
motion a[0]=20550 a[1]=12926 a[2]=0 
motion a[0]=20536 a[1]=12958 a[2]=0 
motion a[0]=20529 a[1]=12990 a[2]=0 
motion a[0]=20529 a[1]=13019 a[2]=0 
motion a[0]=20514 a[1]=13046 a[2]=0 
motion a[0]=20500 a[1]=13073 a[2]=0 
motion a[0]=20491 a[1]=13097 a[2]=0 
motion a[0]=20485 a[1]=13121 a[2]=0 
motion a[0]=20482 a[1]=13141 a[2]=0 
motion a[0]=20477 a[1]=13164 a[2]=0 
motion a[0]=20469 a[1]=13186 a[2]=0 
motion a[0]=20462 a[1]=13207 a[2]=0 
motion a[0]=20457 a[1]=13226 a[2]=0 
motion a[0]=20457 a[1]=13242 a[2]=0 
motion a[0]=20457 a[1]=13259 a[2]=0 
motion a[0]=20457 a[1]=13272 a[2]=0 
motion a[0]=20457 a[1]=13289 a[2]=0 
motion a[0]=20466 a[1]=13301 a[2]=0 
motion a[0]=20474 a[1]=13301 a[2]=0 
motion a[0]=20482 a[1]=13301 a[2]=0 
motion a[0]=20492 a[1]=13314 a[2]=0 
motion a[0]=20503 a[1]=13314 a[2]=0 
motion a[0]=20513 a[1]=13325 a[2]=0 
motion a[0]=20523 a[1]=13325 a[2]=0 
motion a[0]=20535 a[1]=13341 a[2]=0 
motion a[0]=20550 a[1]=13341 a[2]=0 
motion a[0]=20563 a[1]=13356 a[2]=0 
motion a[0]=20576 a[1]=13371 a[2]=0 
motion a[0]=20588 a[1]=13388 a[2]=0 
motion a[0]=20601 a[1]=13402 a[2]=0 
motion a[0]=20614 a[1]=13418 a[2]=0 
motion a[0]=20628 a[1]=13434 a[2]=0 
motion a[0]=20639 a[1]=13451 a[2]=0 
[... 547 lines cut ...]
motion a[0]=12256 a[1]=13690 a[2]=0 
motion a[0]=12256 a[1]=13680 a[2]=0 
motion a[0]=12266 a[1]=13674 a[2]=0 
motion a[0]=12274 a[1]=13674 a[2]=0 
motion a[0]=12274 a[1]=13668 a[2]=0 
motion a[0]=12278 a[1]=13668 a[2]=0 
motion a[0]=12284 a[1]=13668 a[2]=0 
motion a[0]=12284 a[1]=13660 a[2]=0 
motion a[0]=12288 a[1]=13660 a[2]=0 
motion a[0]=12291 a[1]=13660 a[2]=0 
motion a[0]=12291 a[1]=13656 a[2]=0 
motion a[0]=12291 a[1]=13644 a[2]=0 
motion a[0]=12291 a[1]=13632 a[2]=0 
motion a[0]=12299 a[1]=13622 a[2]=0 
motion a[0]=12299 a[1]=13612 a[2]=0 
motion a[0]=12309 a[1]=13612 a[2]=0 
motion a[0]=12317 a[1]=13612 a[2]=0 
motion a[0]=12323 a[1]=13612 a[2]=0 
motion a[0]=12329 a[1]=13602 a[2]=0 
motion a[0]=12336 a[1]=13602 a[2]=0 
motion a[0]=12336 a[1]=13590 a[2]=0 
motion a[0]=12327 a[1]=13590 a[2]=0 
motion a[0]=12319 a[1]=13590 a[2]=0 
^C
[ilovepumpkin@ruige ~]$

spbnick, 2016-09-03:
I have to regretfully inform you, that I (the project maintainer) am [leaving the project](http://spbnick.github.io/2016/07/31/Wrapping-up-DIGImend-work.html). I'm no longer working on supporting new tablets, nor am I supporting users. I'm still accepting pull requests and work on HOWTOs and syncing with upstream, but only until November 2016. Members of the community may still help other users. Anyone desiring to step in is welcome to contact me.