DIGImend/digimend-kernel-drivers #40: Huion 1060 Plus not working on USB3

gadic, 2016-07-22:
Hi,

I have received a Huion 1060 Plus tablet. It works well when connected to USB2 but NOT on USB3 (I've tested 3 different ports).
When playing with the manual installation of the driver, I succeeded twice to make it work on USB3 but without understanding why. After a reboot, it is always not working. Going back to a USB2 port is always fine.

My PC: Shuttle DS437T

$ uname -a
Linux silence 4.4.0-31-generic #50-Ubuntu SMP Wed Jul 13 00:07:12 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial

gadic, 2016-07-22:
More information when the tablet is connected to USB3:

$ lspci
00:00.0 Host bridge: Intel Corporation 3rd Gen Core processor DRAM Controller (rev 09)
00:02.0 VGA compatible controller: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
00:16.0 Communication controller: Intel Corporation 7 Series/C210 Series Chipset Family MEI Controller #1 (rev 04)
00:1a.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #2 (rev 04)
00:1b.0 Audio device: Intel Corporation 7 Series/C210 Series Chipset Family High Definition Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 1 (rev c4)
00:1c.1 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 2 (rev c4)
00:1c.2 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 3 (rev c4)
00:1c.3 PCI bridge: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 4 (rev c4)
00:1d.0 USB controller: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation 7 Series Chipset Family LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 7 Series Chipset Family 6-port SATA Controller [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 7 Series/C210 Series Chipset Family SMBus Controller (rev 04)
01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8188CE 802.11b/g/n WiFi Adapter (rev 01)
02:00.0 USB controller: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller
03:00.0 USB controller: ASMedia Technology Inc. ASM1042A USB 3.0 Host Controller
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 0c)

$ lsusb
Bus 002 Device 003: ID 05e3:0723 Genesys Logic, Inc. GL827L SD/MMC/MS Flash Card Reader
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 007: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 003 Device 006: ID 256c:006e  
Bus 003 Device 005: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 046d:c318 Logitech, Inc. Illuminated Keyboard
Bus 001 Device 003: ID 046d:c016 Logitech, Inc. Optical Wheel Mouse
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ dmesg
[ 1037.952231] usb 3-1: new high-speed USB device number 5 using xhci_hcd
[ 1038.136557] usb 3-1: New USB device found, idVendor=058f, idProduct=6254
[ 1038.136562] usb 3-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 1038.136564] usb 3-1: Product: USB2.0Hub
[ 1038.137892] hub 3-1:1.0: USB hub found
[ 1038.138038] hub 3-1:1.0: 4 ports detected
[ 1038.424161] usb 3-1.1: new full-speed USB device number 6 using xhci_hcd
[ 1038.714053] usb 3-1.1: New USB device found, idVendor=256c, idProduct=006e
[ 1038.714058] usb 3-1.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[ 1038.714060] usb 3-1.1: Product: PenTablet 
[ 1038.726543] input: PenTablet  Pen as /devices/pci0000:00/0000:00:1c.1/0000:02:00.0/usb3/3-1/3-1.1/3-1.1:1.0/0003:256C:006E.000A/input/input22
[ 1038.780710] uclogic 0003:256C:006E.000A: input,hidraw3: USB HID v1.11 Device [PenTablet ] on usb-0000:02:00.0-1.1/input0
[ 1038.786583] input: PenTablet  Mouse as /devices/pci0000:00/0000:00:1c.1/0000:02:00.0/usb3/3-1/3-1.1/3-1.1:1.1/0003:256C:006E.000B/input/input23
[ 1038.787440] uclogic 0003:256C:006E.000B: input,hiddev0,hidraw4: USB HID v1.11 Mouse [PenTablet ] on usb-0000:02:00.0-1.1/input1
[ 1038.800737] input: PenTablet  Keyboard as /devices/pci0000:00/0000:00:1c.1/0000:02:00.0/usb3/3-1/3-1.1/3-1.1:1.2/0003:256C:006E.000C/input/input25
[ 1038.856557] input: PenTablet  Consumer Control as /devices/pci0000:00/0000:00:1c.1/0000:02:00.0/usb3/3-1/3-1.1/3-1.1:1.2/0003:256C:006E.000C/input/input26
[ 1038.912503] input: PenTablet  System Control as /devices/pci0000:00/0000:00:1c.1/0000:02:00.0/usb3/3-1/3-1.1/3-1.1:1.2/0003:256C:006E.000C/input/input27
[ 1038.913009] uclogic 0003:256C:006E.000C: input,hidraw5: USB HID v1.0b Keyboard [PenTablet ] on usb-0000:02:00.0-1.1/input2
[ 1039.000226] usb 3-1.4: new high-speed USB device number 7 using xhci_hcd
[ 1039.106804] usb 3-1.4: New USB device found, idVendor=058f, idProduct=6366
[ 1039.106809] usb 3-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1039.106812] usb 3-1.4: Product: Mass Storage Device
[ 1039.106814] usb 3-1.4: Manufacturer: Generic
[ 1039.106816] usb 3-1.4: SerialNumber: 058F0O1111B1
[ 1039.109313] usb-storage 3-1.4:1.0: USB Mass Storage device detected
[ 1039.109671] scsi host7: usb-storage 3-1.4:1.0
[ 1040.109265] scsi 7:0:0:0: Direct-Access     Multi    Flash Reader     1.00 PQ: 0 ANSI: 0
[ 1040.109889] sd 7:0:0:0: Attached scsi generic sg2 type 0
[ 1040.636162] sd 7:0:0:0: [sdc] 15597568 512-byte logical blocks: (7.99 GB/7.44 GiB)
[ 1040.636714] sd 7:0:0:0: [sdc] Write Protect is off
[ 1040.636719] sd 7:0:0:0: [sdc] Mode Sense: 03 00 00 00
[ 1040.637327] sd 7:0:0:0: [sdc] No Caching mode page found
[... cut ...]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Logitech Logitech Illuminated Keyboard  	id=10	[slave  keyboard (3)]
    ↳ PenTablet  Keyboard                     	id=12	[slave  keyboard (3)]
    ↳ PenTablet  System Control               	id=14	[slave  keyboard (3)]

mabbamOG, 2016-08-21:
Is the rest fully working on usb2? All of the functionality? Can the buttons be easily configured?

gadic, 2016-08-29:
The tablet area matches the screen size and the pen pressure works well.
Regarding the buttons, I have not tested them. Can you please provide a link for configuring and testing them ?

mabbamOG, 2016-09-01:
i'm sorry i cannot help

spbnick, 2016-09-03:
I have to regretfully inform you, that I (the project maintainer) am [leaving the project](http://spbnick.github.io/2016/07/31/Wrapping-up-DIGImend-work.html). I'm no longer working on supporting new tablets, nor am I supporting users. I'm still accepting pull requests and work on HOWTOs and syncing with upstream, but only until November 2016. Members of the community may still help other users. Anyone desiring to step in is welcome to contact me.

ArcFutahito, 2017-09-08:
Have the same problem, I wonder if it's possible to force it go with EHCI through USB hub or something. Other than that, linux does see the device but it seems like it just doesn't want to accept any input.

kress95, 2018-09-08:
I have the same pen (H1060+ 8192) and I'm able to reproduce this issue.

I'm using archlinux with linux-zen 4.18.6 kernel.

### Tested scenarios:

## reboot connected to usb3 port

Not working, reconnecting doesn't help, works fine after `modprobe -r[...]`

## reboot connected to usb2 port

Working fine, reconnecting works, even when moving it to usb3.

## reboot unplugged

Connecting to both usb3 and usb2 works right away.