DIGImend/digimend-kernel-drivers #41: Huion 610pro effects a click when 'nearby'

sleekweasel, 2016-08-01:
My tablet's been working fine for several months, but now when it's close enough to move the pointer, it's behaving as if touching the tablet (and goes crazy if I'm only juuuust in range). I suspect the pen is reporting pressures of 1-1000 instead of 0-1000 (or something like that), so it always reports 'touching' the tablet.

Assuming this is what's happening, can I tweak the driver to ignore very very light pressure?

Is there a load-time parameter I can set, or would I have to recompile?

spbnick, 2016-08-01:
Could you please post the output of "usbhid-dump -es" for your tablet, while you reproduce that? Thank you.

sleekweasel, 2016-08-01:
I get home in a few hours. I'll do that then.

I presume I'm using this driver - I forget what I had to do to get the Huion working back then - but I haven't upgraded any software since last week, so this is going to be a hardware issue.

sleekweasel, 2016-08-01:
Hi.

Hopefully this will be useful - it's a fairly slow approach, a touch, then a retreat, repeated. And then some button clicks.

[approach-slowly-press-retreat-faster.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/394419/approach-slowly-press-retreat-faster.txt)

spbnick, 2016-08-01:
Yes, this looks like a hardware issue. It shows pressure of 1 or 2, which is practically impossible to achieve manually, given the pressure value range. The driver doesn't support filtering this out, but you can add your own code to uclogic_raw_event to clear pressure and tip bit if pressure is less than, say 5. The tip button bit is 0 in second byte and pressure is bytes 7 (lower) and 8 (higher).

I didn't test it, but I suspect that you can add the following to inside the top-most if statement:

    if (((uint16_t)data[7] << 8) | (uint16_t)data[6]) <= 5) {
        data[1] &= ~1;
        data[6] = 0;
        data[7] = 0;
    }

However, the best course of action would be to reach Huion with a warranty claim.

sleekweasel, 2016-08-02:
Thanks, that's very helpful indeed. I've just written to them, but in the meantime I'll be able to continue drawing lots of badgers. :)

spbnick, 2016-08-02:
Actually that code needs to be in the "else", along with the `if (drvdata->invert_pen_inrange)` condition. Otherwise frame buttons might malfunction.

Good luck with the support and the badgers!

sleekweasel, 2016-08-02:
Worked like a charm. Many thanks!

closed 2016-08-03 by spbnick