DIGImend/digimend-kernel-drivers #42: Configuring frame buttons on Monoprice 10594

blm768, 2016-08-11:
I've been playing around with a Monoprice 10594 graphics tablet, which as far as I can tell is basically just a rebranded Huion H610. I've gotten the pen to work perfectly, but the frame buttons are kind of flaky right now. When I use the `evdev` driver, the buttons produce keypress events, but the default keycodes I'm getting aren't terribly useful. When I switch to the Wacom driver, the button events no longer get to Xorg (although they show up in the `evtest` output) and `xsetwacom` doesn't seem to allow configuration of the buttons.

Output of `xsetwacom --list`:
```
10594 Pen stylus                        id: 16  type: STYLUS
```

I've noticed that the output on my machine only includes a "stylus" line, not a "pad" line (like the output shown in Issue #10), which could be related to the issue. It seems that only the pen properties are accessible:

```
xsetwacom set "16" "Area" "0 0 40000 25000"
xsetwacom set "16" "Button" "1" "button +1 "
xsetwacom set "16" "Button" "2" "key +e -e "
xsetwacom set "16" "Button" "3" "button +3 "
xsetwacom set "16" "ToolDebugLevel" "0"
xsetwacom set "16" "TabletDebugLevel" "0"
xsetwacom set "16" "Suppress" "2"
xsetwacom set "16" "RawSample" "4"
xsetwacom set "16" "PressureCurve" "0 0 100 100"
xsetwacom set "16" "Mode" "Absolute"
xsetwacom set "16" "TabletPCButton" "off"
xsetwacom set "16" "Touch" "off"
Property 'Wacom Hardware Touch Switch' does not exist on device.
xsetwacom set "16" "Gesture" "off"
xsetwacom set "16" "ZoomDistance" "0"
xsetwacom set "16" "ScrollDistance" "0"
xsetwacom set "16" "TapTime" "250"
Property 'Wacom Proximity Threshold' does not exist on device.
xsetwacom set "16" "Rotate" "none"
Property 'Wacom Wheel Buttons' does not exist on device.
[repeated lines]
Property 'Wacom Strip Buttons' does not exist on device.
[repeated lines]
xsetwacom set "16" "Threshold" "27"
xsetwacom set "16" "BindToSerial" "0"
xsetwacom set "16" "PressureRecalibration" "on"
```
I'm using `/etc/X11/xorg.conf.d/9-tablet.conf` to make Xorg use the Wacom driver. Since my system doesn't show a USB device name (possibly related to #40; I'm using USB 3.0 like in that issue), I have to match by device class:

```
Section "InputClass"
   Identifier "Monoprice tablet pen"
   MatchIsTablet "on"
   matchUSBID "256c:006e"
   MatchDevicePath "/dev/input/event*"
   Driver "wacom"
EndSection

Section "InputClass"
   Identifier "Monoprice tablet buttons"
   MatchIsKeyboard "on"
   matchUSBID "256c:006e"
   MatchDevicePath "/dev/input/event*"
   Driver "wacom"
EndSection
```

Is there any more debugging info I could dig up to make troubleshooting the issue easier?

I'm using Arch Linux:
*  Kernel version 4.6.4
* Xorg 1.18.4 with evdev and Wacom input drivers

spbnick, 2016-09-03:
I have to regretfully inform you, that I (the project maintainer) am [leaving the project](http://spbnick.github.io/2016/07/31/Wrapping-up-DIGImend-work.html). I'm no longer working on supporting new tablets, nor am I supporting users. I'm still accepting pull requests and work on HOWTOs and syncing with upstream, but only until November 2016. Members of the community may still help other users, and a potential replacement maintainer is being sought and welcome.

blm768, 2016-09-03:
Well, I'm sorry to hear that, but thanks for all the effort you've put into this project! Unfortunately, I don''t have nearly enough experience to help as a maintainer, but maybe I'll get into USB reverse engineering at some point.

RubeRad, 2017-12-26:
Hey there, I managed to get Monoprice 10594 buttons to work on Ubuntu 16.04 (like you, the stylus part worked out of the box for me).

I'm also seeing in lsusb no name
   `    Bus 002 Device 007: ID 256c:006e`
And originally when I did  
   `    xsetwacom --list`
I got nothing

I followed through [sbpnick's diagnostic HOWTO](http://digimend.github.io/support/howto/trbl/locating_failure/), and information seems to be getting through in every case, I can evtest device 14 and see events from BTN_0 -- BTN_7. I can also 
`    xinput test 16`
and see that my buttons create press and release events numbered 1,2,3,8,9,10,11,12. 

It seems good you are putting your xorg config into /etc/X11, but maybe you should choose a higher number than 9? spbnick recommends /etc/X11/xorg.conf.d/52-tablet.conf, because larger numbers go later and overrule. (However, /etc should also overrule /usr/share, so I dunno there).

Here's the conf that works for me in /etc/X11/xorg.conf.d/52-tablet.conf:
```
Section "InputClass"
   Identifier "Monoprice tablet pen"
   MatchIsTablet "on"
   matchUSBID "256c:006e"
   MatchDevicePath "/dev/input/event*"
   Driver "wacom"
EndSection

Section "InputClass"
   Identifier "Monoprice tablet buttons"
   MatchIsKeyboard "on"
   matchUSBID "256c:006e"
   MatchDevicePath "/dev/input/event*"
   Driver "wacom"
EndSection
```

I think it's the same as yours but in a different filename.

After placing that file and rebooting, I was able to 
```
$ xsetwacom --list
10594 Pen stylus                        id: 9   type: STYLUS    
10594 Pad pad                           id: 10  type: PAD       
```
Maybe your root problem is that your xsetwacom --list is only showing stylus, not pad. If that's the case, then probably I can't help you :-(

And that label I was able to use in xsetwacom commands to set the buttons. For instance
`xsetwacom --set '10594 Pad pad' Button 1 "key +ctrl +z -z -ctrl"`
was able to make the Undo button work. I'm setting this up for my kid to make animations with Tupi, so the keyboard shortcuts are apparently different than standard, but see for instance [here](https://github.com/DIGImend/digimend-kernel-drivers/issues/26) where user momashi drops an extensive comment on configuring the Huion 610 with a standard button configuration, which probably works for gimp etc. Just replace the Huion device label with '10594 Pad pad', and then gather the commands into a script that gets automatically run either at login, or different scripts that get run pre-launch of various apps.

Hopefully that gets you there! (and others too)

blm768, 2017-12-26:
If I dig my tablet up again, I guess I'll have to give it another shot. Part of the issue might be distribution-specific; I was trying to use the tablet on Arch Linux. In any case, it looks like some change in the drivers has probably fixed the issues I was seeing.

closed 2017-12-26 by blm768