DIGImend/digimend-kernel-drivers #43: Can't download the drivers on linux mint 18

juliablabla, 2016-08-12:
this comes up when i try to hit "make" in the terminal:

julia@julia-To-be-filled-by-O-E-M ~/Downloads/drivers tablet/digimend-kernel-drivers-6 $ make
make -C /lib/modules/4.4.0-21-generic/build SUBDIRS=/home/julia/Downloads/drivers tablet/digimend-kernel-drivers-6 modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-21-generic'
arch/x86/Makefile:148: CONFIG_X86_X32 enabled but no binutils support
Makefile:670: Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
make[1]: *** No rule to make target 'tablet/digimend-kernel-drivers-6'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-21-generic'
Makefile:9: recipe for target 'modules' failed
make: *** [modules] Error 2

any help here please:/

spbnick, 2016-08-12:
I suspect that you need the "binutils" package installed.

juliablabla, 2016-08-12:
what is that? :P

juliablabla, 2016-08-12:
Im a newcomer to linux so it's a fresh install

blm768, 2016-08-13:
The binutils package includes utilities that are used when compiling and linking programs. Without those tools, many programs won't compile properly. You should be able to install it using the Synaptic Package Manager.

juliablabla, 2016-08-13:
I searched binutils in the software manager and it said i already had it but i could not click it or anything.. but u mean i need synaptic package manager too? And if that works will i need to uninstall the digmend kernel drivers 6 and then install them again just like last time? I tried to uninstall in but i couldn't do that cuz i deleted the files. Im super greatful for your help!! :)

blm768, 2016-08-15:
Any package management utility should work; Synaptic is just one of the most common package managers for Debian-based distributions. What distribution are you using?

juliablabla, 2016-08-15:
Im using linux mint 18 ( fresh install).
And the way i get the digmend kernel drivers is through the terminal following a tutorial and when i try to run the :sudo make and install it just says the thing i wrote  above :/

blm768, 2016-08-18:
Hmm. What happens after you run `sudo apt-get update && sudo apt-get install binutils`?

juliablabla, 2016-08-21:
julia@julia-To-be-filled-by-O-E-M ~/Downloads/drivers tablet/digimend-kernel-drivers-6 $ make
make -C /lib/modules/4.4.0-21-generic/build SUBDIRS=/home/julia/Downloads/drivers tablet/digimend-kernel-drivers-6 modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-21-generic'
arch/x86/Makefile:148: CONFIG_X86_X32 enabled but no binutils support
Makefile:670: Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
make[1]: *** No rule to make target 'tablet/digimend-kernel-drivers-6'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-21-generic'
Makefile:9: recipe for target 'modules' failed
make: *** [modules] Error 2
julia@julia-To-be-filled-by-O-E-M ~/Downloads/drivers tablet/digimend-kernel-drivers-6 $ 

thats the first "make"

this comes next julia@julia-To-be-filled-by-O-E-M ~/Downloads/drivers tablet/digimend-kernel-drivers-6 $ sudo apt-get update
[sudo] password for julia: 
Err:1 http://ftp.df.lth.se/ubuntu xenial InRelease
  Could not resolve 'ftp.df.lth.se'
Err:2 http://ftp.df.lth.se/ubuntu xenial-updates InRelease                     
  Could not resolve 'ftp.df.lth.se'
Err:3 http://ftp.df.lth.se/ubuntu xenial-backports InRelease                   
  Could not resolve 'ftp.df.lth.se'
Hit:4 http://archive.canonical.com/ubuntu xenial InRelease                     
Hit:5 http://security.ubuntu.com/ubuntu xenial-security InRelease          
Err:6 https://mirrors.c0urier.net/?dir=linux/linuxmint/packages sarah InRelease
  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
Fetched 2 451 B in 0s (6 788 B/s)
Reading package lists... Done
E: Failed to fetch https://mirrors.c0urier.net/?dir=linux/linuxmint/packages/dists/sarah/InRelease  Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?)
W: Failed to fetch http://ftp.df.lth.se/ubuntu/dists/xenial/InRelease  Could not resolve 'ftp.df.lth.se'
W: Failed to fetch http://ftp.df.lth.se/ubuntu/dists/xenial-updates/InRelease  Could not resolve 'ftp.df.lth.se'
W: Failed to fetch http://ftp.df.lth.se/ubuntu/dists/xenial-backports/InRelease  Could not resolve 'ftp.df.lth.se'
W: Some index files failed to download. They have been ignored, or old ones used instead.
****

next is this a@julia-To-be-filled-by-O-E-M ~/Downloads/drivers tablet/digimend-kernel-drivers-6 $ sudo apt-get install binutils
Reading package lists... Done
Building dependency tree       
Reading state information... Done
binutils is already the newest version (2.26.1-1ubuntu1~16.04.3).
The following packages were automatically installed and are no longer required:
  breeze-icon-theme calligra-data calligra-libs kate-data katepart kde-runtime
  kde-runtime-data kde-style-breeze kde-style-breeze-qt4 kdelibs-bin
  kdelibs5-data kdelibs5-plugins kdoctools libakonadi-kde4
  libakonadiprotocolinternals1 libattica0.4 libcauchy0.0v5 libdlrestrictions1
  libfam0 libgsl2 libkabc4 libkactivities6 libkatepartinterfaces4 libkcalcore4
  libkcmutils4 libkdcraw-data libkdcraw23 libkde3support4 libkdeclarative5
  libkdecore5 libkdesu5 libkdeui5 libkdewebkit5 libkdnssd4 libkemoticons4
  libkf5auth-data libkf5auth5 libkf5codecs-data libkf5codecs5
  libkf5config-data libkf5configcore5 libkf5configgui5
  libkf5configwidgets-data libkf5configwidgets5 libkf5coreaddons-data
  libkf5coreaddons5 libkf5guiaddons5 libkf5i18n-data libkf5i18n5
  libkf5iconthemes-data libkf5iconthemes5 libkf5itemviews-data
  libkf5itemviews5 libkf5style5 libkf5widgetsaddons-data libkf5widgetsaddons5
  libkf5windowsystem-data libkf5windowsystem5 libkfile4 libkhtml5 libkio5
  libkjsapi4 libkjsembed4 libkldap4 libkmediaplayer4 libknewstuff3-4
  libknotifyconfig4 libkntlm4 libkparts4 libkpty4 libkresources4 libkrosscore4
  libkrossui4 libktexteditor4 libkxmlrpcclient4 libm2mml0.0v5
  libnl-route-3-200 libntrack-qt4-1 libntrack0 libopencolorio1v5 libphonon4
  libphononexperimental4 libplasma3 libpolkit-qt-1-1 libpolkit-qt5-1-1
  libpoppler-qt4-4 libqca2 libqjson0 libqt4-qt3support libqt5script5 libsolid4
  libsoprano4 libspnav0 libstreamanalyzer0v5 libstreams0v5 libthreadweaver4
  libtinyxml2.6.2v5 libutempter0 libyaml-cpp0.3v5 ntrack-module-libnl-0
  oxygen-icon-theme oxygen5-icon-theme phonon phonon-backend-gstreamer
  phonon-backend-gstreamer-common plasma-scriptengine-javascript
  soprano-daemon
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

juliablabla, 2016-08-21:
i tried downloading the v5 verision instead but it's the same problem there.

spbnick, 2016-08-21:
Apparently this happens if you try to build it under a path containing spaces. @juliablabla, check if the directory you put it in, or one of the directories above have space in their names. If they do, remove the spaces. E.g. replace them with underscores. Or move the driver directory into other place, without spaces in the path.

Dracozny, 2016-08-23:
I Installed from Git on mint 18 without issue.

spbnick, 2016-09-03:
I have to regretfully inform you, that I (the project maintainer) am [leaving the project](http://spbnick.github.io/2016/07/31/Wrapping-up-DIGImend-work.html). I'm no longer working on supporting new tablets, nor am I supporting users. I'm still accepting pull requests and work on HOWTOs and syncing with upstream, but only until November 2016.

dullroar, 2016-11-24:
There's a space in the path to the project. That causes it. I was tearing my hair out until I saw the answer about spaces in the path to the project causing issues [here](http://stackoverflow.com/questions/12501124/errors-in-compiling-linux-kernel-module-config-x86-x32-enabled-but-no-binutils/27916315#27916315). Sounds stupid, I know. FWIW, this was on Linux Mint 18, 4.4.0.47 X86_64 kernel, gcc v5.4.0.