DIGImend/digimend-kernel-drivers #44: Huion H610 not working in ubuntu 16.04

Ryan1729, 2016-08-18:
My Huion H610, (it actually says Turcom on it, but it matches the H610 picture on https://digimend.github.io/tablets/) hasn't worked (drawing with the pen does nothing) since I updated to Ubuntu 16.04.
It worked fine on 14.04 and I've tried it on a windows PC and it worked fine.

After updating to 16.04, I downloaded the deb and installed it by running 
```bash
sudo dpkg -i digimend-dkms_6_all.deb
```

If I run `lsusb` with the tablet plugged in and not, the only difference in the ouput is the line

```
Bus 007 Device 005: ID 256c:006e 
```

When I plug in the tablet the following is added to the output of `dmesg` except that the device number keeps going up if I plug and unplug it repeatedly

```
[299851.776070] usb 7-1: new full-speed USB device number 7 using uhci_hcd
[299851.939106] usb 7-1: New USB device found, idVendor=256c, idProduct=006e
[299851.939114] usb 7-1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[299851.939119] usb 7-1: Product: PenTablet
[299851.939124] usb 7-1: Manufacturer: HUION
[299851.949053] uclogic 0003:256C:006E.0017: hidraw5: USB HID v1.11 Keypad [HUION PenTablet] on usb-0000:00:1d.1-1/input0
[299851.952645] uclogic 0003:256C:006E.0018: hiddev0,hidraw6: USB HID v1.11 Mouse [HUION PenTablet] on usb-0000:00:1d.1-1/input1
[299851.957255] uclogic 0003:256C:006E.0019: hidraw7: USB HID v1.11 Keyboard [HUION PenTablet] on usb-0000:00:1d.1-1/input2
```

Dracozny, 2016-08-23:
My Yiynova (UC-Logic) 22+H has the same issue under mint 18 (ubuntu 16.04)

Dracozny, 2016-08-23:
I'm thinking it may be related to this bug. https://github.com/DIGImend/digimend-kernel-drivers/issues/31

Dracozny, 2016-08-23:
I Compiled the most recent code and installed and now my Tablet is detected.

Ryan1729, 2016-08-26:
I just compiled and installed the most recent code too and my tablet works just like it did before. Thanks @Dracozny !

To the maintainer(s):
It might be a good idea to suggest compiling the most recent code if the packaged stuff doesn't work in the readme or something.

closed 2016-08-26 by Ryan1729