DIGImend/digimend-kernel-drivers #47: Huion H610 Pro - xwacom --list - No Pad Device, just Stylus

0BobTheJanitor, 2016-09-20:
#26 
So pretty much the same problem that @thomanski had in that issue, except I don't know where to go. Everything works fine, just can't get pad device to be discovered. Buttons used to work until I followed the instruction to create a 52-tablet.conf, now they don't work anymore. xinput --list, only "HUION PenTablet Pen stylus" is there, used to have a couple more, such as "HUION PenTablet Mouse", "HUION PenTablet Keyboard", but those are gone now. A quick check of Xorg.0.log gives:
`[    93.995] (**) Option "xkb_rules" "evdev"
[    93.996] (II) config/udev: Adding input device HUION PenTablet Pen (/dev/input/event2)
[    93.996] (**) HUION PenTablet Pen: Applying InputClass "evdev tablet catchall"
[    93.996] (**) HUION PenTablet Pen: Applying InputClass "Huion on wacom"
[    93.996] (II) LoadModule: "wacom"
[    93.996] (II) Loading /usr/lib/xorg/modules/input/wacom_drv.so
[    93.997] (II) Module wacom: vendor="X.Org Foundation"
[    93.997] 	compiled for 1.18.3, module version = 0.33.0
[    93.997] 	Module class: X.Org XInput Driver
[    93.997] 	ABI class: X.Org XInput driver, version 22.1
[    93.997] (II) wacom: Driver for Wacom graphics tablets: PenPartner, Graphire,
	Graphire2 4x5, Graphire2 5x7, Graphire3 4x5, Graphire3 6x8,
	Graphire4 4x5, Graphire4 6x8, BambooFun 4x5, BambooFun 6x8,
	Bamboo1 Medium, Graphire4 6x8 BlueTooth, CTL-460, CTH-461, CTL-660,
	CTL-461/S, Bamboo Touch, CTH-460/K, CTH-461/S, CTH-661/S1, CTH-461/L,
	CTH-661/L, Intuos 4x5, Intuos 6x8, Intuos 9x12, Intuos 12x12,
	Intuos 12x18, PTU600, PL400, PL500, PL600, PL600SX, PL550, PL800,
	PL700, PL510, PL710, DTI520, DTF720, DTF720a, DTF521, DTU1931,
	DTU2231, DTU1631, Intuos2 4x5, Intuos2 6x8, Intuos2 9x12,
	Intuos2 12x12, Intuos2 12x18, Intuos2 6x8 , Volito, PenStation,
	Volito2 4x5, Volito2 2x3, PenPartner2, Bamboo, Bamboo1, Bamboo1 4x6,
	Bamboo1 5x8, Intuos3 4x5, Intuos3 6x8, Intuos3 9x12, Intuos3 12x12,
	Intuos3 12x19, Intuos3 6x11, Intuos3 4x6, Intuos4 4x6, Intuos4 6x9,
	Intuos4 8x13, Intuos4 12x19, Intuos4 WL USB Endpoint,
	Intuos4 WL Bluetooth Endpoint, Intuos5 touch S, Intuos5 touch M,
	Intuos5 touch L, Intuos5 S, Intuos5 M, Intuos Pro S, Intuos Pro M,
	Intuos Pro L, Cintiq 21UX, Cintiq 20WSX, Cintiq 12WX, Cintiq 21UX2,
	Cintiq 24HD, Cintiq 22HD, Cintiq 24HD touch (EMR digitizer),
	Cintiq 13HD, DTK2241, DTH2242, Cintiq 22HDT, TabletPC 0x90,
	TabletPC 0x93, TabletPC 0x97, TabletPC 0x9A, CapPlus  0x9F,
	TabletPC 0xE2, TabletPC 0xE3, TabletPC 0xE5, TabletPC 0xE6,
	TabletPC 0xEC, TabletPC 0xED, TabletPC 0xEF, TabletPC 0x100,
	TabletPC 0x101, TabletPC 0x10D, TabletPC 0x116, TabletPC 0x12C,
	TabletPC 0x4001, TabletPC 0x4004, TabletPC 0x5000, TabletPC 0x5002,
	usb:172f:0024, usb:172f:0025, usb:172f:0026, usb:172f:0027,
	usb:172f:0028, usb:172f:0030, usb:172f:0031, usb:172f:0032,
	usb:172f:0033, usb:172f:0034, usb:172f:0035, usb:172f:0036,
	usb:172f:0037, usb:172f:0038, usb:172f:0039, usb:172f:0051,
	usb:172f:0052, usb:172f:0053, usb:172f:0054, usb:172f:0055,
	usb:172f:0056, usb:172f:0057, usb:172f:0058, usb:172f:0500,
	usb:172f:0501, usb:172f:0502, usb:172f:0503, usb:1b96:0001,
	usb:17ef:6004
[    93.997] (II) Using input driver 'wacom' for 'HUION PenTablet Pen'
[    93.997] (**) HUION PenTablet Pen: always reports core events
[    93.997] (**) Option "Device" "/dev/input/event2"
[    94.050] (II) HUION PenTablet Pen: type not specified, assuming 'stylus'.
[    94.050] (II) HUION PenTablet Pen: other types will be automatically added.
[    94.050] (--) HUION PenTablet Pen stylus: using pressure threshold of 27 for button 1
[    94.050] (--) HUION PenTablet Pen stylus: maxX=40000 maxY=25000 maxZ=2047 resX=157000 resY=157000  tilt=enabled
[    94.050] (II) HUION PenTablet Pen stylus: hotplugging dependent devices.
[    94.050] (EE) HUION PenTablet Pen stylus: Invalid type 'eraser' for this device.
[    94.050] (EE) HUION PenTablet Pen stylus: Invalid type 'cursor' for this device.
[    94.050] (EE) HUION PenTablet Pen stylus: Invalid type 'touch' for this device.
[    94.050] (EE) HUION PenTablet Pen stylus: Invalid type 'pad' for this device.
[    94.050] (II) HUION PenTablet Pen stylus: hotplugging completed.
[    94.096] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:256C:006E.0003/input/input4/event2"
[    94.096] (II) XINPUT: Adding extended input device "HUION PenTablet Pen stylus" (type: STYLUS, id 10)
[    94.097] (**) HUION PenTablet Pen stylus: (accel) keeping acceleration scheme 1
[    94.097] (**) HUION PenTablet Pen stylus: (accel) acceleration profile 0
[    94.097] (**) HUION PenTablet Pen stylus: (accel) acceleration factor: 2.000
[    94.097] (**) HUION PenTablet Pen stylus: (accel) acceleration threshold: 4
[    94.150] (II) config/udev: Adding input device HUION PenTablet Pen (/dev/input/mouse1)
[    94.150] (II) No input driver specified, ignoring this device.
[    94.150] (II) This device may have been added with another device file.
[    94.151] (II) config/udev: Adding input device HUION PenTablet Mouse (/dev/input/event3)
[    94.151] (**) HUION PenTablet Mouse: Applying InputClass "evdev pointer catchall"
[    94.151] (**) HUION PenTablet Mouse: Applying InputClass "evdev tablet catchall"
[    94.151] (**) HUION PenTablet Mouse: Applying InputClass "Huion on wacom"
[... 42 lines cut ...]
[    94.350] (EE) PreInit returned 8 for "HUION PenTablet Consumer Control pad"
[    94.350] (II) UnloadModule: "wacom"
[    94.350] (II) config/udev: Adding input device HUION PenTablet System Control (/dev/input/event6)
[    94.350] (**) HUION PenTablet System Control: Applying InputClass "evdev keyboard catchall"
[    94.350] (**) HUION PenTablet System Control: Applying InputClass "evdev tablet catchall"
[    94.350] (**) HUION PenTablet System Control: Applying InputClass "Huion on wacom"
[    94.350] (II) Using input driver 'wacom' for 'HUION PenTablet System Control'
[    94.350] (**) HUION PenTablet System Control: always reports core events
[    94.350] (**) Option "Device" "/dev/input/event6"
[    94.350] (EE) HUION PenTablet System Control: Invalid type 'stylus' for this device.
[    94.350] (EE) HUION PenTablet System Control: Invalid type 'eraser' for this device.
[    94.350] (EE) HUION PenTablet System Control: Invalid type 'cursor' for this device.
[    94.350] (EE) HUION PenTablet System Control: Invalid type 'touch' for this device.
[    94.350] (EE) HUION PenTablet System Control: Invalid type 'pad' for this`

What stands out is

`[    94.050] (EE) HUION PenTablet Pen stylus: Invalid type 'eraser' for this device.
[    94.050] (EE) HUION PenTablet Pen stylus: Invalid type 'cursor' for this device.
[    94.050] (EE) HUION PenTablet Pen stylus: Invalid type 'touch' for this device.
[    94.050] (EE) HUION PenTablet Pen stylus: Invalid type 'pad' for this device.`

So what gives? Is it a configuration problem, or should I install through DKMS, which I thought it was already a part of the main branch? Thanks

Also good luck @spbnick with whatever you decide to do next

spbnick, 2016-09-20:
Thanks @0BobTheJanitor, I fixed your formatting a bit (check out the source if you'd like to see how).

I guess these "Invalid type ..." messages is just the Wacom driver trying to apply different types to each device node.

Can you send the output of `sudo evtest`, which lists all the devices you have? And also the output of `sudo modinfo hid-uclogic`.

0BobTheJanitor, 2016-09-20:
```
/dev/input/event2:      HUION PenTablet Pen
/dev/input/event3:      HUION PenTablet Mouse
/dev/input/event4:      HUION PenTablet Keyboard
/dev/input/event5:      HUION PenTablet Consumer Control
/dev/input/event6:      HUION PenTablet System Control
```
```
filename:       /lib/modules/4.7.4-1-ARCH/extra/hid-uclogic.ko.gz
version:        7
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     BB8839BDC437A3C74CFA5D2
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid,usbcore
vermagic:       4.7.4-1-ARCH SMP preempt mod_unload modversions
```

spbnick, 2016-09-20:
This seems right with the exception of missing pad device. Could you please send the output of `dmesg` after plugging your tablet in? The last 100 lines should be enough. Thank you.

0BobTheJanitor, 2016-09-20:
```
[    0.926438] NetLabel: Initializing
[    0.926439] NetLabel:  domain hash size = 128
[    0.926440] NetLabel:  protocols = UNLABELED CIPSOv4
[    0.926448] NetLabel:  unlabeled traffic allowed by default
[    0.926478] hpet0: at MMIO 0xfed00000, IRQs 2, 8, 0, 0, 0, 0, 0, 0
[    0.926482] hpet0: 8 comparators, 64-bit 14.318180 MHz counter
[    0.928491] amd_nb: Cannot enumerate AMD northbridges
[    0.928500] clocksource: Switched to clocksource hpet
[    0.932090] VFS: Disk quotas dquot_6.6.0
[    0.932101] VFS: Dquot-cache hash table entries: 512 (order 0, 4096 bytes)
[    0.932145] pnp: PnP ACPI init
[    0.932199] system 00:00: [mem 0xfed40000-0xfed44fff] has been reserved
[    0.932202] system 00:00: Plug and Play ACPI device, IDs PNP0c01 (active)
[    0.932253] system 00:01: [io  0x0680-0x069f] has been reserved
[    0.932254] system 00:01: [io  0x1000-0x100f] has been reserved
[    0.932255] system 00:01: [io  0xffff] has been reserved
[    0.932256] system 00:01: [io  0xffff] has been reserved
[    0.932257] system 00:01: [io  0x0400-0x0453] has been reserved
[    0.932258] system 00:01: [io  0x0458-0x047f] has been reserved
[    0.932260] system 00:01: [io  0x0500-0x057f] has been reserved
[    0.932261] system 00:01: [io  0x164e-0x164f] has been reserved
[    0.932262] system 00:01: Plug and Play ACPI device, IDs PNP0c02 (active)
[    0.932277] pnp 00:02: Plug and Play ACPI device, IDs PNP0b00 (active)
[    0.932311] system 00:03: [io  0x0454-0x0457] has been reserved
[    0.932312] system 00:03: Plug and Play ACPI device, IDs INT3f0d PNP0c02 (active)
[    0.932364] system 00:04: [io  0x0290-0x029f] has been reserved
[    0.932365] system 00:04: Plug and Play ACPI device, IDs PNP0c02 (active)
[    0.932501] system 00:05: [io  0x04d0-0x04d1] has been reserved
[    0.932503] system 00:05: Plug and Play ACPI device, IDs PNP0c02 (active)
[    0.932598] pnp 00:06: [dma 0 disabled]
[    0.932629] pnp 00:06: Plug and Play ACPI device, IDs PNP0501 (active)
[    0.932764] system 00:07: [mem 0xfed1c000-0xfed1ffff] has been reserved
[    0.932766] system 00:07: [mem 0xfed10000-0xfed17fff] has been reserved
[    0.932767] system 00:07: [mem 0xfed18000-0xfed18fff] has been reserved
[    0.932768] system 00:07: [mem 0xfed19000-0xfed19fff] has been reserved
[    0.932769] system 00:07: [mem 0xf8000000-0xfbffffff] has been reserved
[    0.932770] system 00:07: [mem 0xfed20000-0xfed3ffff] has been reserved
[    0.932772] system 00:07: [mem 0xfed90000-0xfed93fff] has been reserved
[    0.932773] system 00:07: [mem 0xfed45000-0xfed8ffff] has been reserved
[    0.932774] system 00:07: [mem 0xff000000-0xffffffff] has been reserved
[    0.932775] system 00:07: [mem 0xfee00000-0xfeefffff] could not be reserved
[    0.932776] system 00:07: [mem 0xcfa00000-0xcfa00fff] has been reserved
[    0.932778] system 00:07: Plug and Play ACPI device, IDs PNP0c02 (active)
[    0.932879] system 00:08: [mem 0x20000000-0x201fffff] has been reserved
[    0.932880] system 00:08: [mem 0x40004000-0x40004fff] has been reserved
[    0.932882] system 00:08: Plug and Play ACPI device, IDs PNP0c01 (active)
[    0.932898] pnp: PnP ACPI: found 9 devices
[    0.938781] clocksource: acpi_pm: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 2085701024 ns
[    0.938831] pci 0000:00:01.0: PCI bridge to [bus 01]
[    0.938833] pci 0000:00:01.0:   bridge window [io  0xe000-0xefff]
[    0.938835] pci 0000:00:01.0:   bridge window [mem 0xf6000000-0xf70fffff]
[    0.938837] pci 0000:00:01.0:   bridge window [mem 0xe0000000-0xe9ffffff 64bit pref]
[    0.938839] pci 0000:00:1c.0: PCI bridge to [bus 02]
[    0.938849] pci 0000:00:1c.3: PCI bridge to [bus 03]
[    0.938851] pci 0000:00:1c.3:   bridge window [io  0xd000-0xdfff]
[    0.938854] pci 0000:00:1c.3:   bridge window [mem 0xf7a00000-0xf7afffff]
[    0.938861] pci 0000:00:1c.4: PCI bridge to [bus 04]
[    0.938865] pci 0000:00:1c.4:   bridge window [mem 0xf7900000-0xf79fffff]
[    0.938868] pci 0000:00:1c.4:   bridge window [mem 0xea100000-0xea1fffff 64bit pref]
[    0.938873] pci 0000:05:00.0: PCI bridge to [bus 06]
[    0.938891] pci 0000:00:1c.5: PCI bridge to [bus 05-06]
[    0.938901] pci 0000:00:1c.7: PCI bridge to [bus 07]
[    0.938904] pci 0000:00:1c.7:   bridge window [mem 0xf7800000-0xf78fffff]
[    0.938912] pci_bus 0000:00: resource 4 [io  0x0000-0x0cf7 window]
[    0.938913] pci_bus 0000:00: resource 5 [io  0x0d00-0xffff window]
[    0.938914] pci_bus 0000:00: resource 6 [mem 0x000a0000-0x000bffff window]
[    0.938915] pci_bus 0000:00: resource 7 [mem 0x000d0000-0x000d3fff window]
[    0.938916] pci_bus 0000:00: resource 8 [mem 0x000d4000-0x000d7fff window]
[    0.938917] pci_bus 0000:00: resource 9 [mem 0x000d8000-0x000dbfff window]
[... 546 lines cut ...]
[ 9099.257542] IPv6: ADDRCONF(NETDEV_UP): enp4s0: link is not ready
[ 9099.412305] IPv6: ADDRCONF(NETDEV_UP): enp4s0: link is not ready
[ 9101.675682] ata4: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[ 9101.781213] ata4.00: ACPI cmd ef/10:06:00:00:00:00 (SET FEATURES) succeeded
[ 9101.781219] ata4.00: ACPI cmd f5/00:00:00:00:00:00 (SECURITY FREEZE LOCK) filtered out
[ 9101.781222] ata4.00: ACPI cmd b1/c1:00:00:00:00:00 (DEVICE CONFIGURATION OVERLAY) filtered out
[ 9101.782494] ata4.00: ACPI cmd ef/10:06:00:00:00:00 (SET FEATURES) succeeded
[ 9101.782509] ata4.00: ACPI cmd f5/00:00:00:00:00:00 (SECURITY FREEZE LOCK) filtered out
[... cut ...]
[11201.389649] input: HUION PenTablet Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:256C:006E.000D/input/input41
[11201.441344] input: HUION PenTablet Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:256C:006E.000D/input/input42
[11201.441521] input: HUION PenTablet System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:256C:006E.000D/input/input43
[11201.441771] uclogic 0003:256C:006E.000D: input,hidraw4: USB HID v1.11 Keyboard [HUION PenTablet] on usb-0000:00:14.0-3/input2
```

spbnick, 2016-09-20:
Alright, I suspect there is some issue with enabling abstract keyboard mode. Please send the output of [`uclogic-probe`](https://github.com/DIGImend/uclogic-tools) for your tablet.

0BobTheJanitor, 2016-09-20:
```
M 48 00 55 00 49 00 4F 00 4E 00 00 00 00 00 00 00
P 50 00 65 00 6E 00 54 00 61 00 62 00 6C 00 65 00 74 00 00 00
S 64 0E 03 40 9C A8 61 03 00 FF 07 A0 0F 08 00
S 65 04 03 20 A0
S 6E 04 03 00 30
S 79 14 03 48 00 41 00 36 00 30 00 00 00 00 00 00 00 00 00 00 00
S 7A 08 03 01 08 00 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
```

spbnick, 2016-09-20:
Hmm, this looks correct. I'll dig at the driver source tomorrow.

0BobTheJanitor, 2016-09-20:
Okay, thanks

guyvdb, 2016-09-21:
Hi, 

I had the pad buttons recognised and working under kernel 4.4.0-36-lowlatency. Initially I had the the same issue as #26 (fixed by creating 52-tablet.conf). Yesterday my kernel updated to 4.4.0-38 and xsetwacom --list no longer sees "HUION PenTablet Pad pad id: 13 type: PAD". Reverting back to 4.4.0-36 does not seem to fix the issue.

Have checked config in /usr/share/X11/xorg.conf.d -- both 10-evdev.conf & 52-tablet.conf but they are unchanged.

Think this is the same issue. Here is output from my machine to assist. 

 * evtest - https://www.pastiebin.com/57e28b64d3f01
 * modinfo hid-uclogic - https://www.pastiebin.com/57e28be2aa2d9 
 * dmesg - https://www.pastiebin.com/57e28c8fc5f18
 * uclogic-probe - https://www.pastiebin.com/57e28db0e319a

Hope this helps.

spbnick, 2016-09-21:
Thanks, @guyvdb. Your comment reinforces then need to fix this - it is not a simple misconfiguration, apparently.

However, if you can help it, @guyvdb and @0BobTheJanitor please post the output directly into the issue comments, or at least as [GitHub Gist](https://gist.github.com/). Pastebin and similar pastes tend to disappear quickly.

guyvdb, 2016-09-21:
Will do on gists/ticket for output.

I am a developer. I have done some minor kernel module stuff before. I am looking at the code now to see if I can help but and still trying to find my way. If you can point me into looking at a particular direction I can see if I can help.

guyvdb, 2016-09-21:
I noticed that there is an SSL signing error on make install on my machine.

before modules would be installed at: 

  *  /lib/modules/xxx/kernel/drivers/hid/hid-uclogic.ko 
  * /lib/modules/xxx/extra/hid-uclogic.ko

Now only at:

 * /lib/modules/xxx/extra/hid-uclogic.ko

Output
==========
make -C /lib/modules/4.4.0-36-lowlatency/build SUBDIRS=/home/guy/Downloads/digimend/digimend-kernel-drivers modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-36-lowlatency'
  INSTALL /home/guy/Downloads/digimend/digimend-kernel-drivers/hid-kye.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/guy/Downloads/digimend/digimend-kernel-drivers/hid-polostar.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/guy/Downloads/digimend/digimend-kernel-drivers/hid-uclogic.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-36-lowlatency
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-36-lowlatency'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules
udevadm control --reload

guyvdb, 2016-09-21:
Ok I have got xsetwacom to see the "HUION PenTablet Pad pad         	id: 22	type: PAD"

What I did is as follows:

* unplug tablet
* rmmod hid-uclogic
* updatedb, locate hid-uclogic.ko, delete all
* make, make install (creates /lib/modules/xxx/extra/hid-uclogic.ko)
* insmod /lib/modules/xxx/extra/hid-uclogic.ko
* plug in tablet

spbnick, 2016-09-21:
@guyvdb I don't think the SSL errors affect anything apart from module signatures. The hid-uclogic module (and any other) shouldn't be installed by the out-of-tree driver under /lib/modules/xxx/kernel/drivers/hid/, only under .../extra/... The module under /lib/modules/xxx/kernel/drivers/hid/ should be the driver coming with the kernel package. You can verify that with `modinfo <module_path>`. The stock kernel driver shouldn't have a version.

spbnick, 2016-09-21:
Otherwise I'm not sure what's happening here.

spbnick, 2016-09-21:
Perhaps there is some problem with module rebinding. Could you please restore the module under /lib/modules/xxx/kernel/drivers/hid/ (e.g. by "reinstalling" the kernel package), and then reconnect your tablet and post the last 100 lines from syslog (wherever it is kept on your distro).

guyvdb, 2016-09-21:
I reinstalled the kernel package and confirmed it installed driver at /lib/modules/xxx/kernel/drivers/hid. 

modinfo for kernel package driver:
filename:       /lib/modules/4.4.0-38-lowlatency/kernel/drivers/hid/hid-uclogic.ko
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     44EA734BBBDA6BD0C1155F1
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid
intree:         Y
vermagic:       4.4.0-38-lowlatency SMP preempt mod_unload modversions 

I confirmed that out of tree driver is still installed at /lib/modules/xxx/extra from when I built it.
modinfo for the out of tree driver:
filename:       /lib/modules/4.4.0-38-lowlatency/extra/hid-uclogic.ko
version:        7
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     BB8839BDC437A3C74CFA5D2
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid
vermagic:       4.4.0-38-lowlatency SMP preempt mod_unload modversions

I then rebooted and after boot complete tail -f syslog before plugging in the tablet. Output below.
I cannot figure out what is going on, as after this I rebooted and simply plugged in the tablet,  xsetwacom --list reported correctly.

HUION PenTablet Pen stylus      	id: 22	type: STYLUS    
HUION PenTablet Pad pad         	id: 24	type: PAD 

Output from tail of syslog:
------------
Sep 21 20:32:04 guy-pc kernel: [  136.444587] ieee80211 phy0: rt2x00queue_flush_queue: Warning - Queue 2 failed to flush
Sep 21 20:32:17 guy-pc kernel: [  149.569053] usb 2-4: new full-speed USB device number 6 using xhci_hcd
Sep 21 20:32:17 guy-pc kernel: [  149.734210] usb 2-4: New USB device found, idVendor=256c, idProduct=006e
Sep 21 20:32:17 guy-pc kernel: [  149.734214] usb 2-4: New USB device strings: Mfr=5, Product=6, SerialNumber=0
Sep 21 20:32:17 guy-pc kernel: [  149.734215] usb 2-4: Product: PenTablet
Sep 21 20:32:17 guy-pc kernel: [  149.734217] usb 2-4: Manufacturer: HUION
Sep 21 20:32:17 guy-pc mtp-probe: checking bus 2, device 6: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-4"
Sep 21 20:32:17 guy-pc mtp-probe: bus: 2, device: 6 was not an MTP device
Sep 21 20:32:18 guy-pc kernel: [  150.258021] input: HUION PenTablet Pen as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/0003:256C:006E.0005/input/input21
Sep 21 20:32:18 guy-pc kernel: [  150.309304] input: HUION PenTablet Pad as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/0003:256C:006E.0005/input/input22
Sep 21 20:32:18 guy-pc kernel: [  150.309484] uclogic 0003:256C:006E.0005: input,hidraw4: USB HID v1.11 Keypad [HUION PenTablet] on usb-0000:00:14.0-4/input0
Sep 21 20:32:18 guy-pc kernel: [  150.309715] input: HUION PenTablet Mouse as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.1/0003:256C:006E.0006/input/input23
Sep 21 20:32:18 guy-pc kernel: [  150.309885] uclogic 0003:256C:006E.0006: input,hiddev0,hidraw5: USB HID v1.11 Mouse [HUION PenTablet] on usb-0000:00:14.0-4/input1
Sep 21 20:32:18 guy-pc kernel: [  150.310353] input: HUION PenTablet Keyboard as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.2/0003:256C:006E.0007/input/input25
Sep 21 20:32:18 guy-pc kernel: [  150.361365] input: HUION PenTablet Consumer Control as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.2/0003:256C:006E.0007/input/input26
Sep 21 20:32:18 guy-pc kernel: [  150.361490] input: HUION PenTablet System Control as /devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.2/0003:256C:006E.0007/input/input27
Sep 21 20:32:18 guy-pc kernel: [  150.361653] uclogic 0003:256C:006E.0007: input,hidraw6: USB HID v1.11 Keyboard [HUION PenTablet] on usb-0000:00:14.0-4/input2

spbnick, 2016-09-21:
Hmm. I looked at hid-rebind code and it shouldn't and can't do anything here. Because both the in-tree and out-of-tree drivers have the same name. I.e. it cannot rebind it. The only thing that could have happened, it seems, is that the in-kernel module remained loaded and was serving the tablet. Have you unloaded hid-uclogic after installing the driver, before reporting the issue?

guyvdb, 2016-09-21:
I do not think so (not sure). It could of had the kernel driver loaded.

What causes the kernel to pick the out-of-tree driver over the in-tree driver? Is that hid-rebind?

spbnick, 2016-09-21:
If it is a driver that already exists in the kernel, then it is the
modprobe configuration. The driver supplies a piece of it asking the driver
from the "extra" directory to take precedence.

spbnick, 2016-09-22:
Still, I doubt that the issue was driver not being unloaded. @0BobTheJanitor, I assume you tried rebooting after installing the driver and it didn't work, there was no Pad interface?

0BobTheJanitor, 2016-09-23:
Sorry for the lack of response these last few days. Yes, I installed the driver then rebooted, did not work. However, I did not insmod as @guyvdb and now there is a PAD device. Thank you.

Still new to Linux, so I'm figuring my way around and learning, one new command at a time.

spbnick, 2016-09-23:
Thank you. I'll do some more investigation, though. It should just work.

Fidyyuan, 2016-12-30:
Hello! I have the same tablet and the same issue. I'm really new to Linux but I'm more than happy to get you whatever information you need. I am running Linux 16.04 at the moment with 4.7.2-galliumos kernel. What steps should I take to get my buttons working?

0BobTheJanitor, 2016-12-30:
My solution was to compile the source code manually and then insert the module. Get the source code and do the following:

> Unplug tablet
> Traverse to directory where source code is found
> make, make install
> insmod /lib/modules/xxx/extra/hid-uclogic.ko (xxx is the folder of the current version)
> Do whatever you need to do

Or course, I had to rebind my buttons, which I just copy pasted this:

> xsetwacom --set "HUION PenTablet Pad pad" Button 1 "key +ctrl +z -z -ctrl"
> xsetwacom --set "HUION PenTablet Pad pad" Button 2 “key e”
> xsetwacom --set "HUION PenTablet Pad pad" Button 3 "key +ctrl"
> xsetwacom --set "HUION PenTablet Pad pad" Button 8 "key +shift"
> xsetwacom --set "HUION PenTablet Pad pad" Button 9 "key +ctrl +  -ctrl"
> xsetwacom --set "HUION PenTablet Pad pad" Button 10 "key +ctrl -  -ctrl"
> xsetwacom --set "HUION PenTablet Pad pad" Button 11 "key b"
> xsetwacom --set "HUION PenTablet Pad pad" Button 12 "key space"

I'm not sure if this will solve your problem, but it did for me.

There's also not good UI for doing key rebinds and other xsetwacom commands either, so I'm probably going to make that sometime this or next week

Fidyyuan, 2016-12-30:
Are there any good guides on how to make and find the source? This is literally my first week with Linux ><