DIGImend/digimend-kernel-drivers #48: Mint 18 / Linux 4.4.0 / Ugee M708

wefhy, 2016-10-17:
I can't get my Ugee tablet working un new linux mint. It worked fine on Mint 17-17.3, but doesn't after installing Mint 18 with kernel 4.4.0 
-Without drivers tablet works as a mouse (no pressure)
-With digimend it's totally dead. No response.
I think I have similar problem @SketchStick had. 
https://github.com/DIGImend/digimend-kernel-drivers/issues/31

I tried debian .deb package and also installing it from source. 
I applied patch manually (everything was moved by 3 lines) and still nothing.
this patch: https://gist.github.com/spbnick/a7ec56164bc830ff6f6bc56281c60e18
I'm quite new to linux. I searched web, but I can't find solution.
That's how my installation went(are the SSL errors important?)
```
wefhy@Mint18-wefhy ~/Downloads/digimend-kernel-drivers-6 $ sudo make install
make -C /lib/modules/4.4.0-21-generic/build SUBDIRS=/home/wefhy/Downloads/digimend-kernel-drivers-6 modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-21-generic'
  INSTALL /home/wefhy/Downloads/digimend-kernel-drivers-6/hid-kye.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/wefhy/Downloads/digimend-kernel-drivers-6/hid-polostar.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/wefhy/Downloads/digimend-kernel-drivers-6/hid-uclogic.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-21-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-21-generic'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebin
```
I heard that main developer leaves project, but I hope there's someone to help me.

SketchStick, 2016-10-17:
By the looks of it, you seem to be using the files provided on the releases page? Those are quite old and from before the fixes in #31 were committed. If that's actually the issue you just need to grab the latest code and use that instead.

You can grab the latest code by going to the main page: https://github.com/DIGImend/digimend-kernel-drivers
From there you can just click the green "Clone or download" button and then "Download ZIP".

Alternatively you can also use the git command to stay up to date:
To install Git: `aptitude install git`
To grab the latest code: `git clone https://github.com/DIGImend/digimend-kernel-drivers.git`

From then on you can use `git pull` inside the folder to grab the latest changes and just follow the same steps you would use to install from source.

SketchStick, 2016-10-17:
Those SSL errors look nasty but I'm not sure what would cause that?

wefhy, 2016-10-17:
Thanks :)
So this version doesn't need to be patched?
I just uninstalled previous one, downloaded, unpacked, make and make install this and that's what I got:
```
Mint18-wefhy digimend-kernel-drivers-master # sudo make install
make -C /lib/modules/4.4.0-21-generic/build SUBDIRS=/home/wefhy/Downloads/digimend-kernel-drivers-master modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-21-generic'
  INSTALL /home/wefhy/Downloads/digimend-kernel-drivers-master/hid-kye.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/wefhy/Downloads/digimend-kernel-drivers-master/hid-polostar.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/wefhy/Downloads/digimend-kernel-drivers-master/hid-uclogic.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-21-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-21-generic'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules
udevadm control --reload
```

Now the tablet works but in my apps (blender and gimp) I still can't get pressure working.
I have no idea what are these ssl errors from... It's quite clean system after installing it yesterday I installed only simple programs like blender, geany, chromium, callibre... Nothing special.

SketchStick, 2016-10-17:
Correct, the latest code includes the patch already.

In GIMP I believe for pressure sensitivity you have to go to "Edit" -> "Input Devices" and find the Tablet devices, set it to "Mode: Screen". If you are doing digital painting and not photo manipulation, I'd highly recommend trying Krita instead of GIMP.

wefhy, 2016-10-17:
Ok, sorry, I was lying 
After rebooting it works fine. So now rmmod and reconnecting wasn't enouch.
Thanks for everything :)
Gimp now see about 10 devices (wow! I don't remember this fact from previous systems) but I found the correct one.

I was to install Krita :) I know it's better for me but gimp is just preinstalled on Mint ;)
I hope ssl errors aren't important.
I think my problem is solved, thanks :)

SketchStick, 2016-10-17:
Krita has a global pressure curve in the tablet settings that you can adjust to your liking, out of the box my Yiynova required far too much pressure and I ended up using that curve to max it out with a much softer touch.

spbnick, 2016-10-17:
Glad to hear you figured it out! The SSL errors come from attempts of the kernel build system to sign the installed driver. You can ignore that.

wefhy, 2016-11-12:
Hi, I installed drivers on two more machines and I have deducted something.
I think, there is something missing in newest version in comparison to older ones. Thats not critical, but I think I may help in future improvements.

The newest drivers work when and only when there was previous version installed (and removed).
If I try to install it on clean system(for me Linux Mint), I don't get pressure. If I install and remove .deb package, then I can install newest from github and it works fine.

I'll be installing it on one more device this week, so if you could tell me how to log difference, I think, it could help in future development(if any)... I mean how to diagnose why pressure does not work when it's installed on clean system and why it works when there was .deb installed and removed.
For me it's not a problem, just few more clicks, but maybe it'll help someone else.

spbnick, 2016-11-19:
Hi @wefhy, thanks for your offer. Yes, I think the way to find out what's happening would be to try installing the latest driver on a fresh system, have it fail and then see where it fails.

You can us the [diagnosing HOWTO](http://digimend.github.io/support/howto/trbl/locating_failure/) to help you with that. Look at the "Kernel" and "Userspace" chapters. Write here if you have a problem with that.