DIGImend/digimend-kernel-drivers #49: hid-rebind[28188]: /sbin/hid-rebind: line 65: modprobe: command not found

philuser, 2016-10-29:
I go back here a problem of issue # 8
"HID-rebind [28187]: / sbin / hid-rebind: line 65: modprobe: command not found
"
I am under Opensuse Tumbleweed
1060PRO HUION  tablet
I installed digimend-kernel-drivers-master

after rebolot and connecting my tablet, I read in the journalctl:

-----------------
 new high-speed USB device number 68 using xhci_hcd
 New USB device found, idVendor=058f, idProduct=6254
 New USB device strings: Mfr=0, Product=1, SerialNumber=0
 Product: USB2.0Hub
 hub 1-3:1.0: USB hub found
 hub 1-3:1.0: 4 ports detected
 usb 1-3.1: new full-speed USB device number 69 using xhci_hcd
 usb 1-3.1: New USB device found, idVendor=256c, idProduct=006e
 usb 1-3.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
 usb 1-3.1: Product: PenTablet 
 input: PenTablet  Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.0/0003:256C:006E.002D/input/input117
 input: PenTablet  Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.0/0003:256C:006E.002D/input/input118
 uclogic 0003:256C:006E.002D: input,hidraw3: USB HID v1.11 Keypad [PenTablet ] on usb-0000:00:14.0-3.1/input0
 input: PenTablet  Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.1/0003:256C:006E.002E/input/input119
 uclogic 0003:256C:006E.002E: input,hiddev0,hidraw4: USB HID v1.11 Mouse [PenTablet ] on usb-0000:00:14.0-3.1/input1
 input: PenTablet  Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.2/0003:256C:006E.002F/input/input121
 input: PenTablet  Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.2/0003:256C:006E.002F/input/input122
 input: PenTablet  System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.2/0003:256C:006E.002F/input/input123
 uclogic 0003:256C:006E.002F: input,hidraw5: USB HID v1.0b Keyboard [PenTablet ] on usb-0000:00:14.0-3.1/input2
 usb 1-3.4: new high-speed USB device number 70 using xhci_hcd
 usb 1-3.4: New USB device found, idVendor=058f, idProduct=6366
 New USB device strings: Mfr=1, Product=2, SerialNumber=3
 Product: Mass Storage Device
 usb 1-3.4: Manufacturer: Generic
 usb 1-3.4: SerialNumber: 058F0O1111B1
 usb-storage 1-3.4:1.0: USB Mass Storage device detected
 scsi host4: usb-storage 1-3.4:1.0
 mtp-probe[28157]: checking bus 1, device 70: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.4"
 mtp-probe[28157]: bus: 1, device: 70 was not an MTP device
 checking bus 1, device 69: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1"
 mtp-probe[28156]: bus: 1, device: 69 was not an MTP device
 hid-rebind[28188]: /sbin/hid-rebind: line 65: modprobe: command not found
 hid-rebind[28190]: /sbin/hid-rebind: line 65: modprobe: command not found
 hid-rebind[28187]: /sbin/hid-rebind: line 65: modprobe: command not found
 kernel: scsi 4:0:0:0: Direct-Access     Multi    Flash Reader     1.00 PQ: 0 ANSI: 0
 kernel: sd 4:0:0:0: Attached scsi generic sg2 type 0
 sd 4:0:0:0: [sdc] Attached SCSI removable disk
---------------------------------------------------

The tablet does not work

thank you for helping me

spbnick, 2016-10-31:
Thank you for your report, @philuser. Could you please post the output of "sudo which modprobe"?

philuser, 2016-10-31:
thank you for treating my problem.
Here are the items requested and more.

philuser@philasus:~➤ sudo which modprob
/usr/sbin/modprobe

philuser@philasus:~➤ locate modprobe
/etc/modprobe.d
/etc/modprobe.d/00-system.conf
/etc/modprobe.d/10-unsupported-modules.conf
/etc/modprobe.d/50-alsa.conf
/etc/modprobe.d/50-blacklist.conf
/etc/modprobe.d/50-bluetooth.conf
/etc/modprobe.d/50-ipw2200.conf
/etc/modprobe.d/50-iwl3945.conf
/etc/modprobe.d/50-prism54.conf
/etc/modprobe.d/99-local.conf
/sbin/modprobe
/usr/lib/kmod/modprobe
/usr/sbin/modprobe
/usr/share/YaST2/scrconf/modprobe_blacklist.scr
/usr/share/YaST2/scrconf/modprobe_newid.scr
/usr/share/YaST2/scrconf/modprobe_sound.scr
/usr/share/augeas/lenses/dist/modprobe.aug
/usr/share/bash-completion/completions/modprobe
/usr/share/man/man5/modprobe.d.5.gz
/usr/share/man/man8/modprobe.8.gz

philuser@philasus:~➤ ll /usr/sbin/modprobe
lrwxrwxrwx 1 root root 11 16 oct.  14:32 /usr/sbin/modprobe -> ../bin/kmod

philuser@philasus:~➤ ll /sbin/modprobe
lrwxrwxrwx 1 root root 13 16 oct.  14:32 /sbin/modprobe -> /usr/bin/kmod

philuser@philasus:~➤ ll /usr/lib/kmod/modprobe
lrwxrwxrwx 1 root root 14 16 oct.  14:32 /usr/lib/kmod/modprobe ->
../../bin/kmod

spbnick, 2016-11-01:
Thank you for a comprehensive response, @philuser. Please try the latest [master](https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip), it should fix your problem.

closed 2016-11-01 by spbnick

philuser, 2016-11-01:
Thank you for this effort, since this correction with this new line ''

export PATH = "/ sbin: / usr / sbin: $ PATH"

in the file 'HID-Rebin' I do not have the error on startup.

Despite reboot my computer my tablet "huion 1060PLUS" is totally possessed
by the devil, Halloween may be the cause, but I do not think.

The LED HUION flashes  high frequency and the mouse pointer travels in all
directions causing random clicks mouse. In short my tablet is totally
crazy, how give it reason?

spbnick, 2016-11-01:
Please post the output of `sudo journalctl -n100` right after plugging in your tablet.

philuser, 2016-11-01:
sudo journalctl -f 
when I connect my tablet

`
>nov. 01 11:04:48 philasus.philuser kernel: usb 1-3: new high-speed USB device number 13 using xhci_hcd
nov. 01 11:04:48 philasus.philuser kernel: usb 1-3: New USB device found, idVendor=058f, idProduct=6254
nov. 01 11:04:48 philasus.philuser kernel: usb 1-3: New USB device strings: Mfr=0, Product=1, SerialNumber=0
nov. 01 11:04:48 philasus.philuser kernel: usb 1-3: Product: USB2.0Hub
nov. 01 11:04:48 philasus.philuser kernel: hub 1-3:1.0: USB hub found
nov. 01 11:04:48 philasus.philuser kernel: hub 1-3:1.0: 4 ports detected
nov. 01 11:04:48 philasus.philuser kernel: usb 1-3.1: new full-speed USB device number 14 using xhci_hcd
nov. 01 11:04:49 philasus.philuser kernel: usb 1-3.1: New USB device found, idVendor=256c, idProduct=006e
nov. 01 11:04:49 philasus.philuser kernel: usb 1-3.1: New USB device strings: Mfr=5, Product=6, SerialNumber=0
nov. 01 11:04:49 philasus.philuser kernel: usb 1-3.1: Product: PenTablet 
nov. 01 11:04:49 philasus.philuser kernel: input: PenTablet  Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.0/0003:256C:006E.000A/input/input39
nov. 01 11:04:49 philasus.philuser kernel: input: PenTablet  Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.0/0003:256C:006E.000A/input/input40
nov. 01 11:04:49 philasus.philuser kernel: uclogic 0003:256C:006E.000A: input,hidraw3: USB HID v1.11 Keypad [PenTablet ] on usb-0000:00:14.0-3.1/input0
nov. 01 11:04:49 philasus.philuser kernel: input: PenTablet  Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.1/0003:256C:006E.000B/input/input41
nov. 01 11:04:49 philasus.philuser kernel: uclogic 0003:256C:006E.000B: input,hiddev0,hidraw4: USB HID v1.11 Mouse [PenTablet ] on usb-0000:00:14.0-3.1/input1
nov. 01 11:04:49 philasus.philuser kernel: input: PenTablet  Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.2/0003:256C:006E.000C/input/input43
nov. 01 11:04:49 philasus.philuser kernel: input: PenTablet  Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.2/0003:256C:006E.000C/input/input44
nov. 01 11:04:49 philasus.philuser kernel: input: PenTablet  System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1/1-3.1:1.2/0003:256C:006E.000C/input/input45
nov. 01 11:04:49 philasus.philuser kernel: uclogic 0003:256C:006E.000C: input,hidraw5: USB HID v1.0b Keyboard [PenTablet ] on usb-0000:00:14.0-3.1/input2
nov. 01 11:04:49 philasus.philuser kernel: usb 1-3.4: new high-speed USB device number 15 using xhci_hcd
nov. 01 11:04:49 philasus.philuser kernel: usb 1-3.4: New USB device found, idVendor=058f, idProduct=6366
nov. 01 11:04:49 philasus.philuser kernel: usb 1-3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
nov. 01 11:04:49 philasus.philuser kernel: usb 1-3.4: Product: Mass Storage Device
nov. 01 11:04:49 philasus.philuser kernel: usb 1-3.4: Manufacturer: Generic
nov. 01 11:04:49 philasus.philuser kernel: usb 1-3.4: SerialNumber: 058F0O1111B1
nov. 01 11:04:49 philasus.philuser kernel: usb-storage 1-3.4:1.0: USB Mass Storage device detected
nov. 01 11:04:49 philasus.philuser kernel: scsi host4: usb-storage 1-3.4:1.0
nov. 01 11:04:49 philasus.philuser mtp-probe[5756]: checking bus 1, device 15: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.4"
nov. 01 11:04:49 philasus.philuser mtp-probe[5756]: bus: 1, device: 15 was not an MTP device
nov. 01 11:04:49 philasus.philuser mtp-probe[5755]: checking bus 1, device 14: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.1"
nov. 01 11:04:49 philasus.philuser mtp-probe[5755]: bus: 1, device: 14 was not an MTP device
nov. 01 11:04:50 philasus.philuser kernel: scsi 4:0:0:0: Direct-Access     Multi    Flash Reader     1.00 PQ: 0 ANSI: 0
nov. 01 11:04:50 philasus.philuser kernel: sd 4:0:0:0: Attached scsi generic sg2 type 0
nov. 01 11:04:50 philasus.philuser kernel: sd 4:0:0:0: [sdc] 15564800 512-byte logical blocks: (7.97 GB/7.42 GiB)
nov. 01 11:04:50 philasus.philuser kernel: sd 4:0:0:0: [sdc] Write Protect is off
nov. 01 11:04:50 philasus.philuser kernel: sd 4:0:0:0: [sdc] Mode Sense: 03 00 00 00
nov. 01 11:04:50 philasus.philuser kernel: sd 4:0:0:0: [sdc] No Caching mode page found
nov. 01 11:04:50 philasus.philuser kernel: sd 4:0:0:0: [sdc] Assuming drive cache: write through
nov. 01 11:04:50 philasus.philuser kernel:  sdc: sdc1
nov. 01 11:04:50 philasus.philuser kernel: sd 4:0:0:0: [sdc] Attached SCSI removable disk
nov. 01 11:04:51 philasus.philuser systemd[1]: run-media-philuser-197A\x2d8A5B.mount: Failed to run 'mount' task: No such file or directory
nov. 01 11:04:51 philasus.philuser systemd[1]: Failed to mount /run/media/philuser/197A-8A5B.
nov. 01 11:04:52 philasus.philuser kernel: FAT-fs (sdc1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
nov. 01 11:04:52 philasus.philuser udisksd[1798]: Mounted /dev/sdc1 at /run/media/philuser/197A-8A5B1 on behalf of uid 1000
`

 ----------------------
then when I disconnected
------------------------
`
>nov. 01 11:05:18 philasus.philuser kernel: usb 1-3: USB disconnect, device number 13
nov. 01 11:05:18 philasus.philuser kernel: usb 1-3.1: USB disconnect, device number 14
nov. 01 11:05:19 philasus.philuser kernel: usb 1-3.4: USB disconnect, device number 15
nov. 01 11:05:19 philasus.philuser udisksd[1798]: Cleaning up mount point /run/media/philuser/197A-8A5B1 (device 8:33 no longer exist)
nov. 01 11:05:19 philasus.philuser systemd[1]: Unmounting /run/media/philuser/197A-8A5B1...
nov. 01 11:05:19 philasus.philuser umount[5947]: umount: /run/media/philuser/197A-8A5B1: not mounted
nov. 01 11:05:19 philasus.philuser systemd[1]: run-media-philuser-197A\x2d8A5B1.mount: Mount process exited, code=exited status=32
nov. 01 11:05:19 philasus.philuser systemd[1]: Unmounted /run/media/philuser/197A-8A5B1.
nov. 01 11:05:19 philasus.philuser systemd[1]: run-media-philuser-197A\x2d8A5B1.mount: Unit entered failed state.
`

philuser, 2016-11-01:
removing the flash card that seems to have a mounting problem. the reaction of the tablet remains the same

spbnick, 2016-11-01:
Thank you. Please post the output of command `cat /sys/module/hid_uclogic/version`.

philuser, 2016-11-01:
cat /sys/module/hid_uclogic/version
7

spbnick, 2016-11-01:
Thank you. It seems you're using the right driver, indeed. Please try to use the [diagnosing HOWTO](http://localhost:4000/support/howto/trbl/locating_failure/) I finished recently to find where the input data from your tablet is mangled. Look for irregularities in reported coordinates.

When you find it, or if you have a problem finding it, please open a new issue, as this one (problem with modprobe) will be unrelated.

philuser, 2016-11-01:
sorry but your link connects you to a website in localhost

spbnick, 2016-11-02:
Aargh, same mistake again, sorry. The correct link is http://digimend.github.io/support/howto/trbl/locating_failure/

commit acdd0702 (2016-11-01): Ensure /sbin and /usr/sbin are in hid-rebind PATH

Explicitly add /sbin and /usr/sbin to PATH in hid-rebind as apparently
in some setups they aren't there.

Fixes #49