DIGImend/digimend-kernel-drivers #50: Driver stop working on Ubuntu 16.10 (kernel 4.8)

ericho, 2016-10-29:
Hi

I've using your driver (I have a huion h610) for a few months (I'm really thankful of this driver :smile: ) on Ubuntu 16.04 which has kernel 4.4. I have a new computer and I tried Ubuntu 16.10 there, but I was unable to get it to work. 
Nothing appears on `dmesg`, this is the last that I get.

```
[ 1525.803907] usb 1-3: new full-speed USB device number 13 using xhci_hcd
[ 1525.945718] usb 1-3: New USB device found, idVendor=256c, idProduct=006e
[ 1525.945723] usb 1-3: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[ 1525.945726] usb 1-3: Product: PenTablet
[ 1525.945729] usb 1-3: Manufacturer: HUION
[ 1525.947930] input: HUION PenTablet Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:256C:006E.0016/input/input63
[ 1525.948360] input: HUION PenTablet Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:256C:006E.0016/input/input64
[ 1525.948692] uclogic 0003:256C:006E.0016: input,hidraw3: USB HID v1.11 Keypad [HUION PenTablet] on usb-0000:00:14.0-3/input0
[ 1525.950286] input: HUION PenTablet Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:256C:006E.0017/input/input65
[ 1525.950865] uclogic 0003:256C:006E.0017: input,hiddev0,hidraw4: USB HID v1.11 Mouse [HUION PenTablet] on usb-0000:00:14.0-3/input1
[ 1525.952260] input: HUION PenTablet Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:256C:006E.0018/input/input67
[ 1526.068679] input: HUION PenTablet Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:256C:006E.0018/input/input68
[ 1526.069006] input: HUION PenTablet System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:256C:006E.0018/input/input69
[ 1526.069337] uclogic 0003:256C:006E.0018: input,hidraw5: USB HID v1.11 Keyboard [HUION PenTablet] on usb-0000:00:14.0-3/input2
```

Without any other message that can show an error. 
I tried to search about changes in the kernel 4.8 regarding the HID interface, but I didn't found any significant change. 

lsusb shows: 

```
Bus 001 Device 013: ID 256c:006e  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x256c 
  idProduct          0x006e 
  bcdDevice            0.00
  iManufacturer           5 HUION
  iProduct                6 PenTablet
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     179
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     244
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      92
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
Device Status:     0x0000
  (Bus Powered)
```

I know you just about to not continue supporting this project, but I'll appreciate if you can give some hint on how to debug the issue to find the root cause. 

Thank very much.

spbnick, 2016-10-31:
This can be an issue with the libinput driver for X.org. Could you please follow the [diagnosing HOWTO](http://digimend.github.io/support/howto/trbl/locating_failure/) I finished yesterday to try to find the point of the failure? You can start from the higher level and move down to save time, in particular from the [Userspace drivers part](http://digimend.github.io/support/howto/trbl/locating_failure/#userspace-drivers). Comments on the HOWTO itself are welcome!

cbothamy, 2017-05-10:
I encountered a similar issue with a WH1409 tablet and ubuntu 16.10, that ships a buggy xserver-xorg-input-evdev package (version 2.10.2). XY axes work as expected, but no application get any pressure readings from the tablet.
This issue as been fixed in version 2.10.3.

The work-around is to downgrade to a previous package version as explained on this page https://askubuntu.com/questions/840262/problem-with-old-genius-digital-drawing-tablet-ubuntu-16-10

thwaller, 2017-07-12:
I am on 17.04, xserver-xorg-input-evdev version 2.10.5. Does not work for me.