DIGImend/digimend-kernel-drivers #51: Adding Support tho Huion WH1409 - How to?

jounathaen, 2016-11-10:
Hello,
I'm aware of the fact, that the maintainer is no longer maintaining this project, but maybe someone can help me anyways.
I'd like to get the Huion WH1409 working. I think it shouldn be too hard, as i think it is simmilar to other Huion Devices, which are supported. The Pen and the Pressure Works, simply the 12 Buttons don't.
I have some c and hardware knowledge, but I do not know where to start. I run the instructions here [https://digimend.github.io/support/howto/trbl/diagnostics/](url) and can easily identify, which Button is which code.
I "just" don't know, how to bring this information into the driver.
Greetings Jonathan
BTW: Great Work so far

spbnick, 2016-11-11:
Hello @jounathaen,

Yes, I tried to get some info on how Huion tablets with more than 8 buttons work for a while, but now I don't have much time to work on them. If you'd like to do it, then use [uclogic-dump](https://github.com/DIGImend/uclogic-tools) on your tablet, then, without reconnecting it or rebooting, run usbhid-dump to see the native button reporting protocol.

Then see how that matches the uclogic_buttonpad_rdesc descriptor, uclogic_raw_event function and consult uclogic_button_enable function in [hid-uclogic.c](https://github.com/DIGImend/digimend-kernel-drivers/blob/master/hid-uclogic.c).

Come back with more questions.

SoulEst, 2017-02-08:
I am working on getting the 12 buttons supported by the driver. Have a look at my diagnostic work here: #60 for more info. So far, nine of the 12 buttons on my Huion New 1060PLUS are working using the instructions I found here: https://www.reddit.com/r/krita/comments/3w6ohb/krita_huion_h610_tablet_on_dual_displays_issue/cxu5qqt/.

Deevad, 2017-06-22:
Hi,
I was able to setup my WH1409: I reported an article to get the tablet setup compatible with xsetwacom command line : http://www.davidrevoy.com/article331/setup-huion-giano-wh1409-tablet-on-linux-mint-18-1-ubuntu-16-04  

[image]

Thank you for the Digimend driver all your years on it @spbnick : it looks like this tablet is able to run thanks to other Huion's driver already merged in Digimend. It's probably same technology/data stream than the one produced by the (supported) Huion H610.

My observation are pretty similar to the one reported by @SoulEst :  9 button on 12 available here. Also the tablet use a generic name in ```xinput --list``` and ```xsetwacom``` ; named ```TABLET``` . But in overall it's working well.

spbnick, 2017-06-23:
Hi David! Glad to see a celebrity artist finding our work useful ;) Big thanks for the article, I'm sure it will help many people, and thanks for the kind words regarding the DIGImend project. I still have very little time for it, but am not abandoning it completely, yet.

ArcFutahito, 2017-09-08:
@SoulEst Having huge difficulties with new1060+ on both manjaro and ubuntu, already doing a setup for sniffing usb. Do you actually have any pen recognition?

SoulEst, 2017-09-08:
Full pen recognition. Do you have the driver installed along with libwacom and xf86-input-wacom? Do you have a configuration file in the /etc/X11/xorg.conf.d/ folder with this as the content?

```
Section "InputClass"
    Identifier          "HUION HV800 class"
    MatchProduct        "PenTablet "
    MatchIsTablet       "on"
    MatchDevicePath     "/dev/input/event*"
    Driver              "wacom"
EndSection

Section "InputClass"
    Identifier          "HUION HV800 buttons"
    MatchProduct        "PenTablet "
    MatchIsKeyboard     "on"
    MatchDevicePath     "/dev/input/event*"
    Driver              "evdev"
EndSection

Section "InputClass"
    Identifier          "HUION HV800 scroll"
    MatchProduct        "PenTablet "
    MatchIsPointer      "off"
    MatchIsKeyboard     "off"
    MatchIsTouchpad     "off"
    MatchIsTablet       "off"
    MatchIsTouchscreen  "off"
    MatchDevicePath     "/dev/input/event*"
    Driver              "evdev"
EndSection
```

That plus a script for configuring parameters of the tablet and pen using xsetwacom are all you need.

ChiSamurai, 2018-11-03:
Hi all, 
first of all many thanks for all the work to get these tablets working under linux! I'm totally new on this tablet-thing and got one of these Huion WH1409 tablet as a birthday present. I followed the guide from  @Deevad (thanks for sharing this!) on Linux Mint 18.3 (Kernel 4.10.0-38-generic) but unfotunately I got no luck on get the device running.
Out of the box I can use it as a mouse but when I installed the digimend-drivers, the pen movement does nothing at all.
I tried the installation of 0.6 up to 0.8 using the dkms method as well as the .deb installation - no luck. 

Syslog gives the following entry when I connect the wireless stick:
```
 kernel: [  305.277658] usb 1-1.2: new full-speed USB device number 4 using ehci-pci
 kernel: [  305.468240] usb 1-1.2: New USB device found, idVendor=256c, idProduct=006e
 kernel: [  305.468243] usb 1-1.2: New USB device strings: Mfr=5, Product=6, SerialNumber=0
 kernel: [  305.468245] usb 1-1.2: Product: PenTablet 
 kernel: [  305.498322] input: PenTablet  Pen as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:256C:006E.000B/input/input29
 kernel: [  305.557985] input: PenTablet  Pad as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:256C:006E.000B/input/input30
 kernel: [  305.558280] uclogic 0003:256C:006E.000B: input,hidraw4: USB HID v1.11 Keypad [PenTablet ] on usb-0000:00:1a.0-1.2/input0
 kernel: [  305.575968] input: PenTablet  Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.1/0003:256C:006E.000C/input/input31
 kernel: [  305.576375] uclogic 0003:256C:006E.000C: input,hiddev0,hidraw5: USB HID v1.11 Mouse [PenTablet ] on usb-0000:00:1a.0-1.2/input1
 kernel: [  305.611226] input: PenTablet  Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.2/0003:256C:006E.000D/input/input33
 kernel: [  305.721994] input: PenTablet  Consumer Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.2/0003:256C:006E.000D/input/input34
 kernel: [  305.724186] input: PenTablet  System Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.2/0003:256C:006E.000D/input/input35
 kernel: [  305.724421] uclogic 0003:256C:006E.000D: input,hidraw6: USB HID v1.0b Keyboard [PenTablet ] on usb-0000:00:1a.0-1.2/input2
 mtp-probe: checking bus 1, device 4: "/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2"
 mtp-probe: bus: 1, device: 4 was not an MTP device
```
xinput --list returns
```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Cherry Wireless Keyboard and Mouse      	id=15	[slave  pointer  (2)]
⎜   ↳ Logitech Trackball                      	id=17	[slave  pointer  (2)]
⎜   ↳ PenTablet  Consumer Control             	id=9	[slave  pointer  (2)]
⎜   ↳ PenTablet  Mouse                        	id=11	[slave  pointer  (2)]
⎜   ↳ PenTablet  Pad                          	id=12	[slave  pointer  (2)]
⎜   ↳ PenTablet  Pen                          	id=18	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ UVC Camera (046d:0825)                  	id=13	[slave  keyboard (3)]
    ↳ Cherry Wireless Keyboard and Mouse      	id=14	[slave  keyboard (3)]
    ↳ Burr-Brown from TI               USB Audio CODEC 	id=16	[slave  keyboard (3)]
    ↳ PenTablet  Keyboard                     	id=8	[slave  keyboard (3)]
    ↳ PenTablet  System Control               	id=10	[slave  keyboard (3)]
```
and xsetwacom --list these lines:
```
TABLET Pen Tablet Pen stylus    	id: 11	type: STYLUS    
TABLET Pen Tablet Pad pad       	id: 12	type: PAD   
```
My huion xorg.conf looks like that:

```
# Huion tablets
Section "InputClass"
    Identifier "Huion class"
    MatchProduct "TABLET"
    MatchIsTablet "on"
    MatchDevicePath "/dev/input/event*"
    Driver "wacom"
EndSection

Section "InputClass"
    Identifier "Huion buttons"
    MatchProduct "TABLET"
    MatchIsKeyboard "on"
    MatchDevicePath "/dev/input/event*"
    Driver "evdev"
EndSection

Section "InputClass"
    Identifier "Huion scroll"
    MatchProduct "TABLET"
    MatchIsPointer "off"
    MatchIsKeyboard "off"
    MatchIsTouchpad "off"
    MatchIsTablet "off"
    MatchIsTouchscreen "off"
    MatchDevicePath "/dev/input/event*"
    Driver "evdev"
EndSection
```

Does anybody have some hints? Or can I support you somehow?

Best wishes,
Matthias

ChiSamurai, 2018-11-04:
Hi @Deevad and thank you for your fast answer! Unfortunately changing the MatchProduct matcher did not help. In fact xsetwacom --list returns no entry at all :-( According to xinput --list the productName is like "TABLET Pen Tablet*" so the "TABLET" matcher in your article seems ok to me (but I must confess, I'm not very familiar with xorg configurations).

I wonder if `xinput test $xinput-id` should output something when I move the pen around or use the keys... no output at any of the "PenTablet"-Devices :-(