DIGImend/digimend-kernel-drivers #52: Huion H610 Pro not working after suspend

Wahffle, 2016-11-14:
Hi there, I got my both my stylus and pad working great under Ubuntu 16.04 by installing the most updated drivers from the github source and creating a file under /etc/X11/xorg.conf.d/52-tablet.conf (source: [https://github.com/DIGImend/digimend-kernel-drivers/issues/26](url))

The problem is, everytime I suspend my computer and wake it up, I end up with a non-functional tablet and having to manually disconnect and reconnect the usb ports.

I also tried resetting the usb ports though the terminal to avoid wearing them out (source: [http://askubuntu.com/a/661](url)), but it doesn't seen to work either.

spbnick, 2016-11-19:
Hi @Wahffle, I reproduced this issue, and this needs a driver code change to re-initialize the tablet on resume. I'll see if I have the time to do that.

spbnick, 2016-11-19:
Alright, I pushed a fix to master, please try it:
https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip

Wahffle, 2016-11-19:
I got my tablet working after a suspend, thank you so much for fixing this issue. I still have to run my script to configure it, but that's expected since no Desktop Environment supports another way to configure non-wacom tablets.

closed 2016-11-19 by Wahffle