DIGImend/digimend-kernel-drivers #56: Huion H 610 pro doesn't work (either via .deb or by compiling the source)

freeasinbear, 2016-12-19:
It "worked" (pressure wasn't fully recognized) on a previous install, but when I tried to install it on Debian 64 bit (kernel 4.8) it gets recognized, but doesn't work.

I tried installing it via .deb package and compiling the source, but neither of these solutions worked.

Any idea?

spbnick, 2016-12-20:
Please provide the output of `dmesg` after plugging in the tablet. Also the output of `/sbin/modinfo hid-uclogic`. Thank you.

freeasinbear, 2016-12-20:
for dmesg I realize that most of the output is useless to you, except for the one about the tablet, so here it is (if you need all the rest let me know, because it's very long):

92.462847] usb 2-1.2: new full-speed USB device number 3 using ehci-pci
[   92.572775] usb 2-1.2: New USB device found, idVendor=256c, idProduct=006e
[   92.572782] usb 2-1.2: New USB device strings: Mfr=5, Product=6, SerialNumber=0
[   92.572786] usb 2-1.2: Product: PenTablet
[   92.572789] usb 2-1.2: Manufacturer: HUION
[   92.605084] hidraw: raw HID events driver (C) Jiri Kosina
[   92.613788] usbcore: registered new interface driver usbhid
[   92.613791] usbhid: USB HID core driver
[   92.619698] hid_uclogic: loading out-of-tree module taints kernel.
[   92.619765] hid_uclogic: module verification failed: signature and/or required key missing - tainting kernel
[   92.620783] input: HUION PenTablet Pen as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/0003:256C:006E.0001/input/input19
[   92.621048] input: HUION PenTablet Pad as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/0003:256C:006E.0001/input/input20
[   92.621255] uclogic 0003:256C:006E.0001: input,hidraw0: USB HID v1.11 Keypad [HUION PenTablet] on usb-0000:00:1d.0-1.2/input0
[   92.621742] input: HUION PenTablet Mouse as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.1/0003:256C:006E.0002/input/input21
[   92.622017] uclogic 0003:256C:006E.0002: input,hiddev0,hidraw1: USB HID v1.11 Mouse [HUION PenTablet] on usb-0000:00:1d.0-1.2/input1
[   92.623178] input: HUION PenTablet Keyboard as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.2/0003:256C:006E.0003/input/input23
[   92.683303] input: HUION PenTablet Consumer Control as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.2/0003:256C:006E.0003/input/input24
[   92.683657] input: HUION PenTablet System Control as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.2/0003:256C:006E.0003/input/input25
[   92.683926] uclogic 0003:256C:006E.0003: input,hidraw2: USB HID v1.11 Keyboard [HUION PenTablet] on usb-0000:00:1d.0-1.2/input2

And this is the output of /sbin/modinfo hid-uclogic:

filename:       /lib/modules/4.8.0-2-amd64/extra/hid-uclogic.ko
version:        7
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     0D830731A0C738343D0A756
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid,usbcore
vermagic:       4.8.0-2-amd64 SMP mod_unload modversions

freeasinbear, 2016-12-24:
Hi, any idea?

spbnick, 2016-12-27:
It can be an issue with libinput. Could you please try installing the `xserver-xorg-input-evdev` package?

If that doesn't help please send the output of `evtest` (interrupt it without selecting any device to dump), and also `xinput list`.

freeasinbear, 2016-12-27:
Yes, installing `xserver-xorg-input-evdev` worked! Thanks a lot!

spbnick, 2016-12-27:
Right. I'll have to figure out what's wrong with libinput, then.

closed 2016-12-27 by spbnick