DIGImend/digimend-kernel-drivers #60: Supporting tablets with more than 12 buttons

SoulEst, 2017-02-07:
I own a Huion New 1060PLUS with 12 Express Keys. I tested the tablet with the DIGImend drivers and linuxwacom driver, and now have nine of the 12 buttons on the tablet recognizable. I tested the buttons using evtest and found that the last three buttons show up as axes ([HUION_NEW_1060PLUS.evdev.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/759007/HUION_NEW_1060PLUS.evdev.txt)). I then tested the buttons using usbhid-dump and found something interesting ([HUION_NEW_1060PLUS.usbhid.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/759011/HUION_NEW_1060PLUS.usbhid.txt)). The codes for all the buttons are actually 16-bit, with the first eight buttons using the lower eight bits and the last four using the higher eight bits.

RJQuiralta, 2017-02-08:
It looks like a least you got an extra button with that model, the 1060pro+ only has 8 out of the 12, I have no knowledge on all this, but it seems to me that the other buttons are only plausible trough their software and are not real hardware buttons.

At one point, someone mentioned was able to build the driver on linux from the Huion code, unfortunately the huion forum with that information is no longer available, thus I don't know if that driver is capable of anything else than pressure, as the person never mentioned anything else. I wish I knew how to build, patch, etc. I would probably give it a go myself.

I'm keeping an eye on this project to see if other users come with new information.

SoulEst, 2017-02-09:
Thank you for your comment about Huion's code and for your support. The last three buttons are an anomaly since they are mapped to ABS_X, ABS_Y, and ABS_MISC. As you can probably imagine, this presents a unique problem.

RJQuiralta, 2017-02-09:
Ok, never mind the Huion driver, managed yesterday to build it using the digimend makefile as a the base, installed, nothing that is not provided by the digimend drivers already. 8 buttons still, on my model.

Obviously their code is dated from 2010, way before the 1060 was up for sale, was likely aim at the h610 which has in fact only 8 keys.

Here is the file I found http://bbs.huiontablet.com/thread-203-1-1.html I used the second link, but I think they are exactly the same files.

Also remember reading that xsetwacom makes "phantom" key numbers 4-7 and that the other buttons "may" be beyond the number 12. I haven't found any more on my tablet though.

Not sure if all of this is relevant or of any use to you, but I'm dropping in this lines for future self reference : )

spbnick, 2019-05-05:
At the moment the driver supports up to 12 buttons. If we see any tablets with more than 12 buttons, we can support them as well. Closing as there's no evidence that there are any such tablets so far.

closed 2019-05-05 by spbnick