DIGImend/digimend-kernel-drivers #61: Any plans to sync this repo with the mainline kernel?

tgharib, 2017-02-22:
Hi,

I have two questions:

1. Does this repo get periodically sync'd with mainline or is this something I should do?

2. The XP-Pen G430 that I have seems to have been accidentally supported with this [commit](https://github.com/DIGImend/digimend-kernel-drivers/commit/3335f96d9c1c3ed690fda583fcf381f9be0f7908) for a similar model.

In fact, it seems XP-Pen G430 has the same VID:PID pair of 28bd:0075 as the G540. In that case, should `USB_DEVICE_ID_XPPEN_TABLET_G540` be renamed to `USB_DEVICE_ID_XPPEN_TABLET_G540_G430`?

spbnick, 2017-03-07:
Hi @tgharib, I do sync with upstream, but not regularly. It would help if you did it :)

johanson, 2018-02-13:
G430 has indeed the same ID 28bd:0075.

I tried to get the device working with digimend, but I'm stuck. Here is a relevant [freedesktop bug](https://bugs.freedesktop.org/show_bug.cgi?id=99898). @tgharib, how did you get your G430 working?

closed 2019-07-30 by tgharib