DIGImend/digimend-kernel-drivers #62: Can't install from .deb on Linux Mint 18.1 (initialization from incompatible pointer types)

qwertystop, 2017-03-21:
Attempting `sudo dpkg --install digimend-dkms_6_all.deb` produces:
```
Selecting previously unselected package digimend-dkms.
(Reading database ... 440347 files and directories currently installed.)
Preparing to unpack digimend-dkms_6_all.deb ...
Unpacking digimend-dkms (6) ...
Setting up digimend-dkms (6) ...
Loading new digimend-6 DKMS files...
First Installation: checking all kernels...
Building only for 4.8.0-42-generic
Building initial module for 4.8.0-42-generic
Error! Bad return status for module build on kernel: 4.8.0-42-generic (x86_64)
Consult /var/lib/dkms/digimend/6/build/make.log for more information.
```
The indicated log file contains:
```
DKMS make.log for digimend-6 for kernel 4.8.0-42-generic (x86_64)
Tue Mar 21 10:26:24 EDT 2017
make -C /lib/modules/4.8.0-42-generic/build SUBDIRS=/var/lib/dkms/digimend/6/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.8.0-42-generic'
  CC [M]  /var/lib/dkms/digimend/6/build/hid-kye.o
  CC [M]  /var/lib/dkms/digimend/6/build/hid-uclogic.o
/var/lib/dkms/digimend/6/build/hid-uclogic.c:1101:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .input_configured = uclogic_input_configured,
                      ^
/var/lib/dkms/digimend/6/build/hid-uclogic.c:1101:22: note: (near initialization for ‘uclogic_driver.input_configured’)
cc1: some warnings being treated as errors
scripts/Makefile.build:295: recipe for target '/var/lib/dkms/digimend/6/build/hid-uclogic.o' failed
make[2]: *** [/var/lib/dkms/digimend/6/build/hid-uclogic.o] Error 1
Makefile:1491: recipe for target '_module_/var/lib/dkms/digimend/6/build' failed
make[1]: *** [_module_/var/lib/dkms/digimend/6/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.8.0-42-generic'
Makefile:9: recipe for target 'modules' failed
make: *** [modules] Error 2
```

Solutions? Next steps?

spbnick, 2017-03-22:
@qwertystop Yes, this was fixed in master, but I haven't got around to making a release of that yet.

Please download and build latest master instead:
https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip

spbnick, 2018-02-17:
This should be fixed with [v7 release](https://github.com/DIGImend/digimend-kernel-drivers/releases/tag/v7).

closed 2018-02-17 by spbnick