DIGImend/digimend-kernel-drivers #63: Huion h610 pro no evdev properties Fedora 25

RomuloPBenedetti, 2017-04-26:
I've installed the drive with some (SSL errors...) 

```
make -C /lib/modules/4.10.11-200.fc25.x86_64/build SUBDIRS=/home/rbenedetti/Downloads/digimend-kernel-drivers modules_install
make[1]: Entering directory '/usr/src/kernels/4.10.11-200.fc25.x86_64'
  INSTALL /home/rbenedetti/Downloads/digimend-kernel-drivers/hid-kye.ko
At main.c:158:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:182
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/rbenedetti/Downloads/digimend-kernel-drivers/hid-polostar.ko
At main.c:158:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:182                                                         
sign-file: certs/signing_key.pem: No such file or directory       
  INSTALL /home/rbenedetti/Downloads/digimend-kernel-drivers/hid-uclogic.ko                                                         
At main.c:158:                                                    
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175                                                 
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:182                                                         
sign-file: certs/signing_key.pem: No such file or directory       
  DEPMOD  4.10.11-200.fc25.x86_64                                 
make[1]: Leaving directory '/usr/src/kernels/4.10.11-200.fc25.x86_64'                                                               
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf      
depmod -a                                                         
install hid-rebind /sbin/hid-rebind                               
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules                                                           
udevadm control --reload
```
xinput list:

```
Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ HUION PenTablet Pad                       id=12   [slave  pointer  (2)]
⎜   ↳ HUION PenTablet Consumer Control          id=14   [slave  pointer  (2)]
⎜   ↳ Logitech USB Receiver                     id=15   [slave  pointer  (2)]
⎜   ↳ HUION PenTablet Pen Pen (0)               id=18   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Power Button                              id=8    [slave  keyboard (3)]
    ↳ Sleep Button                              id=9    [slave  keyboard (3)]
    ↳ SONiX USB Keyboard                        id=10   [slave  keyboard (3)]
    ↳ SONiX USB Keyboard                        id=11   [slave  keyboard (3)]
    ↳ HUION PenTablet Pen                       id=13   [slave  keyboard (3)]
    ↳ Eee PC WMI hotkeys                        id=16   [slave  keyboard (3)]
    ↳ HUION PenTablet Consumer Control          id=17   [slave  keyboard (3)]
```

modinfo:
```
modinfo hid-uclogic
filename:       /lib/modules/4.10.11-200.fc25.x86_64/extra/hid-uclogic.ko
version:        7
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     94C7F6380836A8EC7EFE1EB
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        
vermagic:       4.10.11-200.fc25.x86_64 SMP mod_unload 

```
devices props:

```
Device 'HUION PenTablet Pad':
        Device Enabled (152):   1
        Coordinate Transformation Matrix (154): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Send Events Modes Available (273):     1, 0
        libinput Send Events Mode Enabled (274):        0, 0
        libinput Send Events Mode Enabled Default (275):        0, 0
        libinput Left Handed Enabled (287):     0
        libinput Left Handed Enabled Default (288):     0
        Device Node (276):      "/dev/input/event7"
        Device Product ID (277):        9580, 110

Device 'HUION PenTablet Consumer Control':
        Device Enabled (152):   1
        Coordinate Transformation Matrix (154): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Natural Scrolling Enabled (295):       0
        libinput Natural Scrolling Enabled Default (296):       0
        libinput Send Events Modes Available (273):     1, 0
        libinput Send Events Mode Enabled (274):        0, 0
        libinput Send Events Mode Enabled Default (275):        0, 0
        Device Node (276):      "/dev/input/event10"
        Device Product ID (277):        9580, 110
        libinput Drag Lock Buttons (297):       <no items>
        libinput Horizontal Scroll Enabled (298):       1

Device 'HUION PenTablet Pen Pen (0)':
        Device Enabled (152):   1
        Coordinate Transformation Matrix (154): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        libinput Accel Speed (299):     0.000000
        libinput Accel Speed Default (300):     0.000000
        libinput Left Handed Enabled (287):     0
        libinput Left Handed Enabled Default (288):     0
        Device Node (276):      "/dev/input/event6"
        Device Product ID (277):        9580, 110

```

As you can see, there are no evdev properties, and I need to calibrate the proportion, I know it has, from the time I used it in ubuntu, but for some reason, it is not available here.

Thanks to the attention!

spbnick, 2017-04-26:
Hi Romulo,

There are no evdev properties, because you're using the xf86-input-libinput driver. Still, you can use the xinput tool to change the proportions by modifying the coordinate transformation matrix. You can also install xf86-input-evdev, but it's going away, and it should work with xf86-input-libinput.

RomuloPBenedetti, 2017-04-28:
You're right, many thanks for the suggestion, I've managed to calibrate it with this matrix:
1   0    0
0 2.22 0
0   0    1
It is not much important but is there a way to activate middle button scroll with libinput?

spbnick, 2017-04-28:
Glad to hear you figured it out. Unfortunately, I'm not sure how wheel emulation should be set up with the libinput driver, or if it's even supported. You'll have to raise this with support for your distro, or libinput developers.

RomuloPBenedetti, 2017-05-02:
Ok, many thanks @spbnick!