DIGImend/digimend-kernel-drivers #64: xp-pen star05

ilvinc3nt, 2017-05-13:
$ lsusb
Bus 002 Device 005: ID 28bd:0071  

$ xinput list
 ↳ XP-PEN STAR 05                          	id=15	[slave  pointer  (2)]

$ cat /proc/bus/input/devices
I: Bus=0003 Vendor=28bd Product=0071 Version=0100
N: Name="XP-PEN STAR 05"
P: Phys=usb-0000:00:1d.0-1.1/input0
S: Sysfs=/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:28BD:0071.0002/input/input15
U: Uniq=
H: Handlers=event14 mouse2 
B: PROP=0
B: EV=1f
B: KEY=400 3f0000 0 0 0 0
B: REL=303
B: ABS=1000003
B: MSC=10

$ xinput list-props 15
Device 'XP-PEN STAR 05':
	Device Enabled (139):	1
	Coordinate Transformation Matrix (141):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Calibration Matrix (624):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Calibration Matrix Default (625):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	libinput Accel Speed (275):	0.000000
	libinput Accel Speed Default (276):	0.000000
	libinput Accel Profiles Available (277):	1, 1
	libinput Accel Profile Enabled (278):	1, 0
	libinput Accel Profile Enabled Default (279):	1, 0
	libinput Natural Scrolling Enabled (280):	0
	libinput Natural Scrolling Enabled Default (281):	0
	libinput Send Events Modes Available (260):	1, 0
	libinput Send Events Mode Enabled (261):	0, 0
	libinput Send Events Mode Enabled Default (262):	0, 0
	libinput Left Handed Enabled (282):	0
	libinput Left Handed Enabled Default (283):	0
	libinput Scroll Methods Available (284):	0, 0, 1
	libinput Scroll Method Enabled (285):	0, 0, 0
	libinput Scroll Method Enabled Default (286):	0, 0, 0
	libinput Button Scrolling Button (287):	2
	libinput Button Scrolling Button Default (288):	2
	libinput Middle Emulation Enabled (289):	0
	libinput Middle Emulation Enabled Default (290):	0
	Device Node (263):	"/dev/input/event14"
	Device Product ID (264):	10429, 113
	libinput Drag Lock Buttons (291):	<no items>
	libinput Horizontal Scroll Enabled (292):	1

it doesn't work (arch linux) even if i compiled the last kernel-drivers

ilvinc3nt, 2017-05-14:
the problem i think is in xorg. i don't know. the pen cannot write. (i'm on arch linux)

i added this to /etc/X11/xorg.conf.d/52-tablet.conf
 
Section "InputClass"
       Identifier "XP-PEN STAR 05"
       Driver "evdev"
       MatchProduct "XP-PEN STAR 05"
       MatchDevicePath "/dev/input/event*"
EndSection

Xorg.0.log:
41.374] (II) config/udev: Adding input device XP-PEN STAR 05 (/dev/input/event5)
[    41.374] ( ** ) XP-PEN STAR 05: Applying InputClass "evdev pointer catchall"
[    41.374] ( ** ) XP-PEN STAR 05: Applying InputClass "libinput pointer catchall"
[    41.374] ( ** ) XP-PEN STAR 05: Applying InputClass "XP-PEN STAR 05"
[    41.374] (II) LoadModule: "evdev"
[    41.375] (II) Loading /usr/lib/xorg/modules/input/evdev_drv.so
[    41.383] (II) Module evdev: vendor="X.Org Foundation"
[    41.383] 	compiled for 1.19.1, module version = 2.10.5
[    41.383] 	Module class: X.Org XInput Driver
[    41.383] 	ABI class: X.Org XInput driver, version 24.1
[    41.384] (II) Using input driver 'evdev' for 'XP-PEN STAR 05'
[    41.384] ( ** ) XP-PEN STAR 05: always reports core events
[    41.384] ( ** ) evdev: XP-PEN STAR 05: Device: "/dev/input/event5"
[    41.440] (--) evdev: XP-PEN STAR 05: Vendor 0x28bd Product 0x71
[    41.440] (--) evdev: XP-PEN STAR 05: Found 10 mouse buttons
[    41.440] (--) evdev: XP-PEN STAR 05: Found scroll wheel(s)
[    41.440] (--) evdev: XP-PEN STAR 05: Found relative axes
[    41.440] (--) evdev: XP-PEN STAR 05: Found x and y relative axes
[    41.440] (--) evdev: XP-PEN STAR 05: Found absolute axes
[    41.440] (--) evdev: XP-PEN STAR 05: Found x and y absolute axes
[    41.440] (--) evdev: XP-PEN STAR 05: Found absolute touchpad.
[    41.440] (II) evdev: XP-PEN STAR 05: Configuring as touchpad
[    41.440] (II) evdev: XP-PEN STAR 05: Adding scrollwheel support
[    41.440] ( ** ) evdev: XP-PEN STAR 05: YAxisMapping: buttons 4 and 5
[    41.440] ( ** ) evdev: XP-PEN STAR 05: EmulateWheelButton: 4, EmulateWheelInertia: 10, EmulateWheelTimeout: 200
[    41.440] ( ** ) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:28BD:0071.0001/input/input6/event5"
[    41.440] (II) XINPUT: Adding extended input device "XP-PEN STAR 05" (type: TOUCHPAD, id 11)
[    41.440] (WW) evdev: XP-PEN STAR 05: touchpads, tablets and touchscreens ignore relative axes.
[    41.440] (II) evdev: XP-PEN STAR 05: initialized for absolute axes.
[    41.440] ( ** ) XP-PEN STAR 05: (accel) keeping acceleration scheme 1
[    41.440] ( ** ) XP-PEN STAR 05: (accel) acceleration profile 0
[    41.440] ( ** ) XP-PEN STAR 05: (accel) acceleration factor: 2.000
[    41.440] ( ** ) XP-PEN STAR 05: (accel) acceleration threshold: 4
[    41.441] (II) config/udev: Adding input device XP-PEN STAR 05 (/dev/input/mouse0)
[    41.441] (II) No input driver specified, ignoring this device.
[    41.441] (II) This device may have been added with another device file.

spbnick, 2017-05-15:
Hi @ilvinc3nt, did you try this HOWTO: http://digimend.github.io/support/howto/trbl/locating_failure/ ?

ilvinc3nt, 2017-05-15:
> did you try this HOWTO: http://digimend.github.io/support/howto/trbl/locating_failure/ ?

thanks for replying to me!

first of all 28bd:0071 is ugee ex07 (searched in Tablet repo)

second:

> # sudo usbhid-dump -es -m 28bd:0071
i got stream

> [root@vinc-lnx 0003:13EE:0001.0004]# cat events
no straming (only when i move a usb mouse)

> # evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	Power Button
/dev/input/event1:	Lid Switch
/dev/input/event2:	Power Button
/dev/input/event3:	Video Bus
/dev/input/event4:	AT Translated Set 2 keyboard
/dev/input/event5:	XP-PEN STAR 05
/dev/input/event6:	MOON AND 
/dev/input/event7:	HP WMI hotkeys
/dev/input/event8:	HDA Digital PCBeep
/dev/input/event9:	HDA Intel MID Mic
/dev/input/event10:	HDA Intel MID Headphone
/dev/input/event11:	HDA Intel MID HDMI/DP,pcm=3
/dev/input/event12:	SynPS/2 Synaptics TouchPad
/dev/input/event13:	HP Webcam-101
/dev/input/event14:	GIGABYTE U7300
Select the device event number [0-14]: 

> 5
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x28bd product 0x71 version 0x100
Input device name: "XP-PEN STAR 05"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
    Event code 277 (BTN_FORWARD)
    Event code 330 (BTN_TOUCH)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 8 (REL_WHEEL)
    Event code 9 (REL_MISC)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value  24818
      Min        0
      Max    32767
    Event code 1 (ABS_Y)
      Value  21419
      Min        0
      Max    32767
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     1023
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
[... 132 lines cut ...]
Event: time 1494833262.394398, type 3 (EV_ABS), code 1 (ABS_Y), value 11426
Event: time 1494833262.394398, -------------- SYN_REPORT ------------
Event: time 1494833262.398377, type 3 (EV_ABS), code 0 (ABS_X), value 2541
Event: time 1494833262.398377, type 3 (EV_ABS), code 1 (ABS_Y), value 11457
Event: time 1494833262.398377, -------------- SYN_REPORT ------------
Event: time 1494833262.402397, type 3 (EV_ABS), code 0 (ABS_X), value 2573
Event: time 1494833262.402397, type 3 (EV_ABS), code 1 (ABS_Y), value 11472
Event: time 1494833262.402397, -------------- SYN_REPORT ------------
Event: time 1494833265.666411, type 3 (EV_ABS), code 0 (ABS_X), value 29385
Event: time 1494833265.666411, type 3 (EV_ABS), code 1 (ABS_Y), value 26119
Event: time 1494833265.666411, -------------- SYN_REPORT ------------
Event: time 1494833265.698402, type 3 (EV_ABS), code 0 (ABS_X), value 29412
Event: time 1494833265.698402, -------------- SYN_REPORT ------------
Event: time 1494833265.702393, type 3 (EV_ABS), code 0 (ABS_X), value 29433
Event: time 1494833265.702393, -------------- SYN_REPORT ------------
Event: time 1494833265.706443, type 3 (EV_ABS), code 0 (ABS_X), value 29459
Event: time 1494833265.706443, type 3 (EV_ABS), code 1 (ABS_Y), value 26144
Event: time 1494833265.706443, -------------- SYN_REPORT ------------
Event: time 1494833265.710436, type 3 (EV_ABS), code 0 (ABS_X), value 29492
Event: time 1494833265.710436, -------------- SYN_REPORT ------------
Event: time 1494833265.714399, type 3 (EV_ABS), code 0 (ABS_X), value 29528
Event: time 1494833265.714399, type 3 (EV_ABS), code 1 (ABS_Y), value 26165
Event: time 1494833265.714399, -------------- SYN_REPORT ------------
Event: time 1494833265.718401, type 3 (EV_ABS), code 0 (ABS_X), value 29575
Event: time 1494833265.718401, -------------- SYN_REPORT ------------

ilvinc3nt, 2017-05-15:
ok i noticed that button 8 was clicked even if no button i touched. so i add this:

> Option "ButtonMapping" "1 2 3 4 5 6 7 0 9 10 11 12 13 14"

to 52-tablet.conf

spbnick, 2017-05-16:
Alright. The input stack actually seems to be dealing with this mostly OK. Except it doesn't notice the in-range event and produces some spurious button events. Nothing that can't be fixed with a proper report descriptor.

However, the tablet also crops the horizontal resolution (horizontal extent is 32767, but should be 40640). Apparently because of the limited generic HID driver in Windows. This is a recurring pattern.
You can try to hunt down the initialization sequence of the tablet to enable full resolution.

I suggest you try with [uclogic-tools](https://github.com/DIGImend/uclogic-tools) first. Run `uclogic-probe | uclogic-decode` on it and see if it manages to get any useful data.

ilvinc3nt, 2017-05-16:
> uclogic-probe 2 4 | uclogic-decode

Manufacturer: XP-PEN
       Product: STAR 05
         Max X: 40640
         Max Y: 25400
  Max pressure: 2047
    Resolution: 5080
Internal model: U608i
Buttons status: U608i

spbnick, 2017-05-17:
[Ah, so it's UC-Logic after all, great! Then this could be easy. Let me spend some time on this. Perhaps I'll have a driver to test for you tomorrow.

ilvinc3nt, 2017-05-17:
thank Nikolai for this giant support!

but my big problem is that the distance movement works before the pen touches the table. 
When i touch the table, the mouse pointer stucks on the screen when i take a short distance (distance movement is lost). evtest works (i mean the distance stream) but the pointer stucks.
it's like some buttons stay in one click status even if i release the pen from table.
of course the pointer starts moving when i move the pen on the table as it should be.

spbnick, 2017-05-17:
Let's see what happens when we enable the fully-functional mode with the driver. Perhaps the behavior will change and we'll go from there.

ilvinc3nt, 2017-05-18:
yes, it's a good starting point. maybe everything is gonna be fixed with driver.

thanks so much

ilvinc3nt, 2017-05-18:
added this:

90-hid-rehind.rules
> SUBSYSTEM=="hid", ACTION=="add", ENV{HID_ID}=="0003:000028BD:00000071", \
	RUN+="/sbin/hid-rebind"

hid-ids.h
> #define USB_VENDOR_ID_XPPEN			0x28bd
#define USB_DEVICE_ID_XPPEN_STAR_05		0x0071

hid-uclogic.c
>  /*
		*XP-PEN STAR 05			
 */
	case USB_DEVICE_ID_XPPEN_STAR_05:	
		if (intf->cur_altsetting->desc.bInterfaceNumber == 1) {
			rc = uclogic_probe_tablet(hdev);
			if (rc) {
				hid_err(hdev, "tablet enabling failed\n");
				return rc;
			}
			drvdata->invert_pen_inrange = true;
		} else {
			drvdata->ignore_pen_usage = true;
		}
		break;

It works with (52-tablet.conf):
> Driver "wacom"

now the iusse is with hard button on tablet.

spbnick, 2017-05-19:
It turns out that the upstream kernel already has the tablet enabled: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=fe1a83b43869790501ed82b58dc8229023c69f74

I'll merge that into the driver package.

Meanwhile could you please collect buttons diagnostics as described here:
http://digimend.github.io/support/howto/trbl/diagnostics/

spbnick, 2017-05-19:
Also, can I ask you to make an entry on your tablet in the [tablets repo](https://github.com/DIGImend/tablets), so that we have info on it on the website?

spbnick, 2017-05-19:
Alright, pushed. Could you please try the latest master?

ilvinc3nt, 2017-05-19:
with your last commits everything is buggy:

> It turns out that the upstream kernel already has the tablet enabled: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=fe1a83b43869790501ed82b58dc8229023c69f74
I'll merge that into the driver package.

Xorg.0.log:
> 41.184] (II) config/udev: Adding input device XP-PEN STAR 05 (/dev/input/event5)
[    41.184] ( ** ) XP-PEN STAR 05: Applying InputClass "evdev pointer catchall"
[    41.184] ( ** ) XP-PEN STAR 05: Applying InputClass "libinput pointer catchall"
[    41.184] ( ** ) XP-PEN STAR 05: Applying InputClass "XP-PEN STAR 05"
[    41.184] (II) LoadModule: "wacom"
[    41.184] (II) Loading /usr/lib/xorg/modules/input/wacom_drv.so
[    41.202] (II) Module wacom: vendor="X.Org Foundation"
[    41.202] 	compiled for 1.19.2, module version = 0.34.2
[    41.202] 	Module class: X.Org XInput Driver
[    41.202] 	ABI class: X.Org XInput driver, version 24.1
[    41.202] (II) wacom: Driver for Wacom graphics tablets: PenPartner, Graphire,
	Graphire2 4x5, Graphire2 5x7, Graphire3 4x5, Graphire3 6x8,
	Graphire4 4x5, Graphire4 6x8, BambooFun 4x5, BambooFun 6x8,
	Bamboo1 Medium, Graphire4 6x8 BlueTooth, CTL-460, CTH-461, CTL-660,
	CTL-461/S, Bamboo Touch, CTH-460/K, CTH-461/S, CTH-661/S1, CTH-461/L,
	CTH-661/L, Intuos 4x5, Intuos 6x8, Intuos 9x12, Intuos 12x12,
	Intuos 12x18, PTU600, PL400, PL500, PL600, PL600SX, PL550, PL800,
	PL700, PL510, PL710, DTI520, DTF720, DTF720a, DTF521, DTU1931,
	DTU2231, DTU1631, Intuos2 4x5, Intuos2 6x8, Intuos2 9x12,
	Intuos2 12x12, Intuos2 12x18, Intuos2 6x8 , Volito, PenStation,
	Volito2 4x5, Volito2 2x3, PenPartner2, Bamboo, Bamboo1, Bamboo1 4x6,
	Bamboo1 5x8, Intuos3 4x5, Intuos3 6x8, Intuos3 9x12, Intuos3 12x12,
	Intuos3 12x19, Intuos3 6x11, Intuos3 4x6, Intuos4 4x6, Intuos4 6x9,
	Intuos4 8x13, Intuos4 12x19, Intuos4 WL USB Endpoint,
	Intuos4 WL Bluetooth Endpoint, Intuos5 touch S, Intuos5 touch M,
	Intuos5 touch L, Intuos5 S, Intuos5 M, Intuos Pro S, Intuos Pro M,
	Intuos Pro L, Cintiq 21UX, Cintiq 20WSX, Cintiq 12WX, Cintiq 21UX2,
	Cintiq 24HD, Cintiq 22HD, Cintiq 24HD touch (EMR digitizer),
	Cintiq 13HD, DTK2241, DTH2242, Cintiq 22HDT, TabletPC 0x90,
	TabletPC 0x93, TabletPC 0x97, TabletPC 0x9A, CapPlus  0x9F,
	TabletPC 0xE2, TabletPC 0xE3, TabletPC 0xE5, TabletPC 0xE6,
	TabletPC 0xEC, TabletPC 0xED, TabletPC 0xEF, TabletPC 0x100,
	TabletPC 0x101, TabletPC 0x10D, TabletPC 0x116, TabletPC 0x12C,
	TabletPC 0x4001, TabletPC 0x4004, TabletPC 0x5000, TabletPC 0x5002,
	usb:172f:0024, usb:172f:0025, usb:172f:0026, usb:172f:0027,
	usb:172f:0028, usb:172f:0030, usb:172f:0031, usb:172f:0032,
	usb:172f:0033, usb:172f:0034, usb:172f:0035, usb:172f:0036,
	usb:172f:0037, usb:172f:0038, usb:172f:0039, usb:172f:0051,
	usb:172f:0052, usb:172f:0053, usb:172f:0054, usb:172f:0055,
	usb:172f:0056, usb:172f:0057, usb:172f:0058, usb:172f:0500,
	usb:172f:0501, usb:172f:0502, usb:172f:0503, usb:1b96:0001,
	usb:17ef:6004
[    41.204] (II) Using input driver 'wacom' for 'XP-PEN STAR 05'
[    41.204] ( ** ) XP-PEN STAR 05: always reports core events
[    41.204] ( ** ) Option "Device" "/dev/input/event5"
[    41.260] (EE) XP-PEN STAR 05: Invalid type 'stylus' for this device.
[    41.260] (EE) XP-PEN STAR 05: Invalid type 'eraser' for this device.
[    41.260] (EE) XP-PEN STAR 05: Invalid type 'cursor' for this device.
[    41.260] (II) XP-PEN STAR 05: type not specified, assuming 'touch'.
[    41.260] (II) XP-PEN STAR 05: other types will be automatically added.
[    41.260] (--) XP-PEN STAR 05 touch: maxX=32767 maxY=32767 maxZ=1023 resX=0 resY=0 
[    41.260] (II) XP-PEN STAR 05 touch: hotplugging dependent devices.
[    41.260] (EE) XP-PEN STAR 05 touch: Invalid type 'stylus' for this device.
[    41.260] (EE) XP-PEN STAR 05 touch: Invalid type 'eraser' for this device.
[    41.260] (EE) XP-PEN STAR 05 touch: Invalid type 'cursor' for this device.
[    41.260] (II) XP-PEN STAR 05 touch: hotplugging completed.
[    41.306] ( ** ) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:28BD:0071.0001/input/input6/event5"
[    41.306] (II) XINPUT: Adding extended input device "XP-PEN STAR 05 touch" (type: TOUCH, id 11)
[    41.307] ( ** ) XP-PEN STAR 05 touch: (accel) keeping acceleration scheme 1
[    41.307] ( ** ) XP-PEN STAR 05 touch: (accel) acceleration profile 0
[    41.307] ( ** ) XP-PEN STAR 05 touch: (accel) acceleration factor: 2.000
[    41.307] ( ** ) XP-PEN STAR 05 touch: (accel) acceleration threshold: 4
[    41.364] (II) config/udev: Adding input device XP-PEN STAR 05 (/dev/input/mouse0)
[    41.364] (II) No input driver specified, ignoring this device.
[... 93 lines cut ...]
[    44.517] (II) Using input driver 'wacom' for 'XP-PEN STAR 05 Pen'
[    44.517] ( ** ) XP-PEN STAR 05 Pen: always reports core events
[    44.517] ( ** ) Option "Device" "/dev/input/event5"
[    44.573] (II) XP-PEN STAR 05 Pen: type not specified, assuming 'stylus'.
[    44.573] (II) XP-PEN STAR 05 Pen: other types will be automatically added.
[    44.573] (--) XP-PEN STAR 05 Pen stylus: using pressure threshold of 873 for button 1
[    44.573] (--) XP-PEN STAR 05 Pen stylus: maxX=40640 maxY=25400 maxZ=2047 resX=200000 resY=200000  tilt=enabled
[    44.573] (II) XP-PEN STAR 05 Pen stylus: hotplugging dependent devices.
[    44.573] (EE) XP-PEN STAR 05 Pen stylus: Invalid type 'eraser' for this device.
[    44.573] (EE) XP-PEN STAR 05 Pen stylus: Invalid type 'cursor' for this device.
[... cut ...]
[    44.670] (II) config/udev: Adding input device XP-PEN STAR 05 Pen (/dev/input/mouse0)
[    44.670] (II) No input driver specified, ignoring this device.
[    44.670] (II) This device may have been added with another device file.

now tablet works again (the buttons issue is still there, but i'm gonna try this: http://digimend.github.io/support/howto/trbl/diagnostics/ and then report here).
Maybe Ugee EX07 and XP-PEN STAR05 are different in something and they needs different lines in the code.

spbnick, 2017-05-19:
Right, I forgot the rebinding piece, as the upstream commit doesn't need/have it. Could you please try the latest master again? I pushed a fix.

ilvinc3nt, 2017-05-19:
now it works!
buttons bug as it was.

sudo usbhid-dump -es -m 28bd:0071 | tee frame_buttons.txt
> 002:003:001:STREAM             1495213146.125198
 06 00 00 00 40

> 002:003:001:STREAM             1495213146.217195
 06 00 00 00 00

> 002:003:001:STREAM             1495213147.301212
 06 00 00 00 80

> 002:003:001:STREAM             1495213147.401158
 06 00 00 00 00

> 002:003:001:STREAM             1495213148.423233
 06 01 00 00 00

> 002:003:001:STREAM             1495213148.523224
 06 00 00 00 00

> 002:003:001:STREAM             1495213149.493233
 06 02 00 00 00

> 002:003:001:STREAM             1495213149.607230
 06 00 00 00 00

> 002:003:001:STREAM             1495213150.691260
 06 04 00 00 00

> 002:003:001:STREAM             1495213150.783245
 06 00 00 00 00

> 002:003:001:STREAM             1495213151.813256
 06 08 00 00 00

> 002:003:001:STREAM             1495213151.927270
 06 00 00 00 00

ilvinc3nt, 2017-05-21:
@spbnick look at this:

xinput
>↳ XP-PEN STAR 05 Mouse touch              	id=11	[slave  pointer  (2)]
  ↳ XP-PEN STAR 05 Pen stylus               	id=12	[slave  pointer  (2)]

xsetwacom --list
> XP-PEN STAR 05 Mouse touch      	id: 11	type: TOUCH     
   XP-PEN STAR 05 Pen stylus          	id: 12	type: STYLUS    

so: the buttons on stylus works. maybe 'mouse touch" should be 'pad' to make buttons works?

leoorshii, 2018-01-08:
I followed the instructions http://digimend.github.io/support/howto/drivers/wizardpen/ set-up XP-Pen star06, the tablet is writing but frame buttons are not working,

i ran the command

`xsetwacom --list`

It returned empty line

Please how can i map the buttons?

aliuadepoju, 2018-01-08:
I followed the instructions http://digimend.github.io/support/howto/drivers/wizardpen/ to set-up XP-Pen star06, the tablet is writing but frame buttons are not working,

i ran the command

`xsetwacom --list`

It returned empty line

Please how can i map the buttons?