DIGImend/digimend-kernel-drivers #69: H420 recognized but not responding

StrayLamb2, 2017-05-23:
I work on Xubuntu 16.04 kernel 4.4.0.78, installed the .deb and everything was fine. A day passed, and now the tablet is not responding. Everything except xinput sees the tablet, but xinput refuses to recognize it... Any ideas? Tried to: reinstall driver, rmmod/modprob, kernel reinstall, plug/repug, start with unplugged device/start with plugged device.

StrayLamb2, 2017-05-23:
by the way, I have to report that hid-huion seems to not be loaded, but the installation on my kernel was fine from dpkg...

StrayLamb2, 2017-05-23:
not sure how and why, but running the software updater fixed it... :/

spbnick, 2017-05-23:
`dmesg` output could have helped. Also, upgrade output. Perhaps the driver failed to build.

StrayLamb2, 2017-05-23:
dmesg saw a huion device, so I guess it recognized it...

spbnick, 2017-05-23:
Sure. The question is which driver handled it.

StrayLamb2, 2017-05-23:
I think it was uc-logic, but it doesn't matter now that I can't reproduce it I guess... Anyways, as a contribution, so that this thread is not completely useless, if anyone can't get the drivers to work, try the software updates on settings after installing the .deb. :D

closed 2017-12-05 by spbnick