DIGImend/digimend-kernel-drivers #70: Tablet stops working after suspending (Manjaro KDE)

Wahffle, 2017-05-30:
Hi there, everytime I suspend my computer and wake it up, I end up with a non-functional tablet and having to manually disconnect and reconnect the usb ports.

I used to have this problem under Ubuntu 16.04, but it got patched (https://github.com/DIGImend/digimend-kernel-drivers/issues/52). Now that I'm using Manjaro with the KDE desktop I ended up with the same problem as before. I got my both my stylus and pad working by installing the most updated drivers from the github source and creating a file under /etc/X11/xorg.conf.d/52-tablet.conf (source: #26)

spbnick, 2017-05-30:
Hmmm. The latest master should have this fixed. Could you please verify that you're using the digimend driver, and not the upstream kernel's driver?

Please run `/sbin/modinfo hid-uclogic` and post the output here.

Wahffle, 2017-05-30:
`filename:       /lib/modules/4.9.30-1-MANJARO/kernel/drivers/hid/hid-uclogic.ko.gz
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid,usbcore
intree:         Y
vermagic:       4.9.30-1-MANJARO SMP preempt mod_unload modversions 
`

spbnick, 2017-05-30:
Right, this is from the kernel in your distribution. The filename doesn't have "extra" in the path, and the module description doesn't have a "version" field. Could you please try (re)installing the driver from master?

spbnick, 2017-05-30:
If that still doesn't help, please provide the output from the build and installation process.

Wahffle, 2017-05-30:
It worked! Thanks again and sorry for bothering you, my kernel must have been updated right after manually installing the drivers and I didn't notice it.

closed 2017-05-30 by Wahffle

spbnick, 2017-05-30:
Glad to hear it helped!