DIGImend/digimend-kernel-drivers #72: XP-Pen Artist 22HD support

abisaro, 2017-07-23:
Hi, I recently bought a XP-Pen Artist 22HD, as I'm a developer (not C, but at least I know a bit where to put my hands in C sources) and that looks like there are no more maintainers for this project I thought to try and follow the hints in the wiki.
So I got the descriptor, transformed it, arranged where it needed, tweaked hid-uclogic.c (it's a XP-Pen, but lsusb reports it as UC-Logic) and hid-ids.h, the result is not that bad, though I see some warning in X logs that seems to be related to libinput and tablet, it's overall working, including pressure (tested with Gimp), just a little parallax problem I've not yet got rid of (I've not been able yet to find the right xorg.conf configuration).

As I read around some other people trying to have this display tablet working, I thought to send you the changes I added, so maybe you can review and improve it and push it into the kernel tree.

If you need anything (logs, descriptors, etc) just let me know.

One more thing: debugging the driver I noticed that when configuring interface 1 (mouse) the descriptor is not retrieved from the device and the fixed one I added in the code is used, for interface 0 (pen) it is instead read correctly from the device.

cheers

[xppen_artist22hd.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/1168116/xppen_artist22hd.zip)

spbnick, 2017-08-03:
Hi @abisaro, thank you very much for sharing your work. However, I have very little time recently to work on the project, as you probably read in the README.md and nobody has stepped in to help yet.

If you could shape your changes into a pull request that might speed it up. Thank you.

abisaro, 2017-08-05:
Hi @spbnick, yes, I read the README.md, that's why I thought to try to work on it on my own and contribute it (I just created the pull request).
Please note: implementation is quite rough and will need some rework and improvement, but as I told you I'm not much into C and kernel dev anymore, and I too have not much free time to dedicate to this implementation for the time being, but at least it's better than nothing.

BTW thank you for your great job ;-)