DIGImend/digimend-kernel-drivers #75: Express Keys on Huion H610 Pro

rhg, 2017-08-25:
Hi, I was able to get the hardware buttons working by using the latest drivers here and using the X wacom driver but I can't get the express keys working. I have no idea how to map them.

spbnick, 2017-08-25:
Express keys don't actually exist. They are just areas on the tablet surface that the windows driver dedicates to translating into key-presses. They're mostly a useless gimmick. You can't press them without looking at the tablet, wasting your attention. That's why none of the serious tablets actually have them. So instead just bind some actions to the hardware buttons and use either a proper keyboard, or on-screen buttons instead :)

rhg, 2017-08-25:
I see, I just wasn't sure.

- Ricardo Gomez

spbnick, 2017-08-26:
No problem :)

closed 2017-08-26 by spbnick