DIGImend/digimend-kernel-drivers #76: Huion Q11K not detected

mifth, 2017-09-28:
Hi.
I have Linux 64bit, Ubuntu 17.04.
I have compiled and installed Digimend but nothing works. (((
My Huion Q11K Inspiroy is not recognized and not seen in xinput --list. My Pen does not work at all. I even cannot move a pointer. ((
It seems this is the same issue as here https://github.com/DIGImend/digimend-kernel-drivers/issues/59

https://pastebin.com/MJA0vEsa
https://pastebin.com/33S7rT7w
https://pastebin.com/Dy9G4pMW
https://pastebin.com/1SdDM2VL - dmesg

I also tried to set my conf but it did not help:
https://pastebin.com/05E9hQQW

Please, help me if you can!
I just need to get my pen get working with 2 its buttons.

Thank you.

spbnick, 2017-09-29:
This might be something new from Huion, even though they reuse the product ID, again. Could you please run [uclogic-probe](https://github.com/DIGImend/uclogic-tools) on your tablet and post its output here?

spbnick, 2017-09-29:
Also, `lsusb -v` and `usbhid-dump` output would be useful.

mifth, 2017-09-29:
Hi!
Thanks for the reply. Yep, it seems they bypass manufacturer name.
https://pastebin.com/FsnbBzas here is uclogic-probe. I hope I did it right
https://pastebin.com/w34dqZe6 - lsusb -v
https://pastebin.com/FR4w1nRu - usbhid-dump

I would be happy if I could just draw and press rightclick and middleclick.

Thanks a lot!

mifth, 2017-09-29:
https://pastebin.com/sWh9Bzhh  - I also tried with decode

I have these libs installed
https://packages.ubuntu.com/search?keywords=libusb-0.1-4
[image]

spbnick, 2017-09-29:
Thank you.
Could you please edit your comments to have the text from pastebin embedded, and not linked, so we don't lose it?

Also, could you please build and try to use the latest [uclogic-tools master](https://github.com/DIGImend/uclogic-tools)? That might give us a bit more data. Thank you.

mifth, 2017-09-29:
Ok, I did it.

[text.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1343781/text.txt) - uclogic tools master

[w34dqZe6.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1343784/w34dqZe6.txt) - lsusb

[FR4w1nRu.txt](https://github.com/DIGImend/digimend-kernel-drivers/files/1343786/FR4w1nRu.txt) - usbhid-dump

uclogic tools still soes not work. May be I have a problem with libusb? 
[image]

spbnick, 2017-09-29:
What do you mean by "does not work"?

mifth, 2017-09-29:
It writes
"Failed to get manufacturer string descriptor: Pipe error
Failed to get product string descriptor: Pipe error"

Is it ok?

spbnick, 2017-09-29:
Right, this shows us that the tablet is *very* different from what was using that VID:PID before and that more needs to be done to make it work. Now, please do full diagnostics as described at http://digimend.github.io/support/howto/trbl/diagnostics/ and post the results here, either as is or in an archive.

mifth, 2017-09-29:
Good. Here is my test. Buttons et the tablet did not write anything itnto the file.
[test.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/1344174/test.zip)

spbnick, 2017-09-29:
Thank you. So, this case is indeed similar to #59. Without discovering the initialization sequence which the Windows driver sends, we can make the movement, pressure, and pen buttons work. However, spatial resolution will be lower than advertised, and frame buttons will not work.

The frame wheel dump seems to contain a lot of reports from the pen. Does this tablet actually have a wheel? It seems it doesn't.

If you're willing to drive this further, then please capture and post some USB traffic on Windows while the tablet is being plugged in.

mifth, 2017-09-29:
Thank you too! 
That's goot that its possible to make the pen get working.
There is no wheel at the tablet. Only buttons.
Also, there is no tilt. As far as i know.

I'm ready to go feather. I have windows7 vm. Also, i have some friends with windows os.
What should i do to get the usb traffic?

spbnick, 2017-09-29:
The VM should be enough and might even be preferred. Install Wireshark on the host, start Windows VM, install tablet drivers on it, start recording USB traffic on all interfaces, connect your tablet, and connect it to the VM. Wait a little, then disconnect the tablet, then stop recording. That should catch it. Save the file, compress, and post it here.

For safety, though, I would filter out the traffic from other devices before posting the capture on the web. Or at least making sure you're do not have any devices connected that receive or transmit any sensitive data during the capture. Such as a webcam or a microphone, or a keyboard where you type your passwords.

mifth, 2017-09-29:
Ok, I did it. I hope I did it correctly.
[test_shark.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/1344543/test_shark.zip)

spbnick, 2017-09-29:
Thank you, this seems to be what we need. I'll try to update uclogic-tools to support your tablet, if it makes sense, or make another small tool, on the weekend. Once we have your tablet outputting full-featured reports under Linux we'll be able to make the driver work.

mifth, 2017-09-29:
Oooooooooooo!!!!!!!! Thank you bratan!
Be the force with you!

spbnick, 2017-10-01:
Alright, I pushed probing and decoding your tablet parameters to the latest [uclogic-tools master](https://github.com/DIGImend/uclogic-tools/archive/master.zip). Please download, build, run it on your tablet, then (if it doesn't fail), **without unplugging** your tablet, collect diagnostics samples for pen coordinates, pressure, and buttons, plus frame buttons. Please post results here.

mifth, 2017-10-01:
Hi!
Thanks for the code! I did it:

mifth@mifth:~$ lsusb
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 003: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 002: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 005: ID 256c:006e  
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 09da:000a A4Tech Co., Ltd. Optical Mouse Opto 510D / OP-620D
Bus 004 Device 002: ID 172f:0034 Waltop International Corp. Slim Tablet 12.1"
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
mifth@mifth:~$ sudo uclogic-probe 5 5 | uclogic-decode
Failed to get manufacturer string descriptor: Pipe error
Failed to get product string descriptor: Pipe error
                 Max X: 55880
                 Max Y: 34925
          Max pressure: 8191
            Resolution: 5080
     Unknown string #1: HUION_T164_170520
 Internal manufacturer: HUION Animation Technology Co.,ltd

spbnick, 2017-10-01:
Yes, that's working fine. Please proceed to collecting diagnostics, after running the new uclogic-probe and without reconnecting your tablet.

mifth, 2017-10-01:
Oh, sorry i forgot to get diagnostics. I'll do it asap when i get to the computer.

mifth, 2017-10-01:
Hi again!
I did the new test.
[test2.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/1347386/test2.zip)

spbnick, 2017-10-01:
That doesn't seem to have worked, but let's check. Did you run the new `uclogic-probe` right before collecting these new files?

mifth, 2017-10-02:
I used uclogic-tools from your post https://github.com/DIGImend/digimend-kernel-drivers/issues/76#issuecomment-333357249

mifth, 2017-10-02:
Ok, I did another test with latest master code.
[test3.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/1348012/test3.zip)

mifth, 2017-10-02:
I rebooted my machine and did the test again.
[test4.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/1348017/test4.zip)

spbnick, 2017-10-02:
Thank you. These also show the old reports. What we should be seeing in `usbhid-dump` output, when pen is moved is reports starting with "08", instead of "0A". We also should be getting something when frame buttons are pressed. I'll take a closer look at the windows driver trace and try to figure out the initialization sequence. I will update uclogic-tools with it and update you. However, this is not likely to happen today, perhaps tomorrow.

mifth, 2017-10-02:
Thank you. I'll wait for sure.

spbnick, 2017-10-04:
Alright. I couldn't see anything more special in the Windows trace than what we already do in uclogic-probe. However, I'm not sure what exactly happens with the tablet between running uclogic-probe, and usbhid-dump. I suspect that the driver reinitializes the tablet when uclogic-probe finishes.

To check that, and to see if there is something else that could be interfering, please capture tablet USB traffic on Linux:

1. Start the capture
2. Plug in the tablet
3. Run `uclogic-probe`
4. Start `usbhid-dump -es`
5. Do a stroke with the pen
6. Stop `usbhid-dump`
7. Stop the capture
8. Make sure nothing sensitive got into the capture
9. Post the capture here

mifth, 2017-10-04:
Ok, i'll do it this evening when get back home.

ashwanipirate11, 2017-10-05:
i might be random but you seem like you know a lot about linux.
iam new to linux and a noob , i accidentally deleted my partitions while installing linux( 18.1) mint
and ended up with 1drive for everything .
my hdd is 500gb and i have 100GB of important data and cannot afford to lose it. can you shed some light on this lost soul.

and i also ordered huion q11k inspiroy , i hope i dont get any problems like this.

spbnick, 2017-10-06:
@ashwanipirate11, saving data is important and deserves close attention. If your budget is low, see if you can find any hackerspaces nearby, perhaps they can help you, otherwise find a data recovery professional. If it doesn't involve physical data recovery (like from a broken drive) it might be affordable. I unfortunately lack time to help you.

Regarding the tablet, until we or somebody else fixes the drivers, you are going to have a problem with your Huion Inspiroy Q11K.

ashwanipirate11, 2017-10-06:
thank you spbnick.   actually i did not lose my data.

mifth, 2017-10-06:
Hi!
I have just tried to Start the capture. But I failed to do it as i'm no an advanced linux user.
I installed Wireshark.
But when I run "sudo usbhid-dump -es" - my mouse and keyboard will be disabled. And I cannot do anything. I have to reboot my machine.

Without "sudo" I get this:
mifth@mifth:~$ usbhid-dump -es
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/002/002: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
Failed to find and open the devices: Access denied (insufficient permissions)

mifth, 2017-10-06:
Also, should I filter wireshark usb device? As it records all USB inputs.

spbnick, 2017-10-06:
@mifth, tell usbhid-dump which device to dump. Run `usbhid-dump --help` to see the options. You can use the `-m VID:PID` option, for example.

You can record all devices with Wireshark, then filter out just the one you need, and then export only displayed packets.

mifth, 2017-10-07:
Thanks a lot! I'll try to do it at monday.

spbnick, 2017-10-07:
Feel free to ask here if you get stuck, I'll try to answer quickly, so you can complete this on Monday.

konachan700, 2017-10-12:
**mifth**, please try https://github.com/konachan700/Q11K_Driver

spbnick, 2017-10-13:
@konachan700, perhaps *you* could help us and do a USB traffic dump following the instructions in https://github.com/DIGImend/digimend-kernel-drivers/issues/76#issuecomment-334066906 ?

If you can find time to do this, and use the latest uclogic-tools master, we might be able to find the initialization sequence and make the tablet work with digimend-kernel-drivers with full resolution.

mifth, 2017-10-13:
Hi guys.

@spbnick sorry I'll make the test today. I had some problems at my job. ((
@konachan700 I have installed your driver and rebooted my machine. But still nothing works. ((
Here is my install https://pastebin.com/8HwRBEPE

Here are my logs:
[boot.log](https://github.com/DIGImend/digimend-kernel-drivers/files/1382179/boot.log)

[Xorg.0.log](https://github.com/DIGImend/digimend-kernel-drivers/files/1382180/Xorg.0.log)
[kern.log](https://github.com/DIGImend/digimend-kernel-drivers/files/1382182/kern.log)

I hope it will help.

mifth, 2017-10-13:
@konachan700  
Looking at Xorg.0... It looks like the device is recognized as keyboard. May be that's a problem at my machine?

[    23.426] (II) config/udev: Adding input device Huion Q11K Keyboard (/dev/input/event20)
[    23.426] (**) Huion Q11K Keyboard: Applying InputClass "libinput keyboard catchall"
[    23.426] (II) Using input driver 'libinput' for 'Huion Q11K Keyboard'
[    23.426] (**) Huion Q11K Keyboard: always reports core events
[    23.426] (**) Option "Device" "/dev/input/event20"
[    23.426] (**) Option "_source" "server/udev"
[    23.426] (II) input device 'Huion Q11K Keyboard', /dev/input/event20 is tagged by udev as: Keyboard
[    23.426] (II) input device 'Huion Q11K Keyboard', /dev/input/event20 is a keyboard
[    23.456] (**) Option "config_info" "udev:/sys/devices/virtual/input/input22/event20"
[    23.456] (II) XINPUT: Adding extended input device "Huion Q11K Keyboard" (type: KEYBOARD, id 15)
[    23.456] (**) Option "xkb_model" "pc105"
[    23.456] (**) Option "xkb_layout" "us,ru"
[    23.456] (**) Option "xkb_variant" ","
[    23.456] (**) Option "xkb_options" "grp:alt_shift_toggle,grp_led:scroll"
[    23.456] (II) input device 'Huion Q11K Keyboard', /dev/input/event20 is tagged by udev as: Keyboard
[    23.456] (II) input device 'Huion Q11K Keyboard', /dev/input/event20 is a keyboard

konachan700, 2017-10-13:
mifth, please, download the current version - i add a conflicted driver to blacklist now.

mifth, 2017-10-13:
OMG!!!! It works!!!! It's alive!!! It's alive!!
[image]

The pressure is recognized in Krita.

But the pressure is not recognized in Blender.

[image]

mifth, 2017-10-13:
May be I have to add something to "/usr/share/X11/xorg.conf.d"? A config file.

mifth, 2017-10-13:
My xinput now looks like this

mifth@mifth:~$ xinput --list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Microsoft Wired Keyboard 600              id=10   [slave  pointer  (2)]
⎜   ↳ Wacom Intuos4 6x9 Pen stylus              id=11   [slave  pointer  (2)]
⎜   ↳ Wacom Intuos4 6x9 Pad pad                 id=12   [slave  pointer  (2)]
⎜   ↳ A4TECH USB Device                         id=13   [slave  pointer  (2)]
⎜   ↳ A4TECH USB Device                         id=14   [slave  pointer  (2)]
⎜   ↳ Huion Q11K Tablet stylus                  id=15   [slave  pointer  (2)]
⎜   ↳ Wacom Intuos4 6x9 Pen eraser              id=18   [slave  pointer  (2)]
⎜   ↳ Wacom Intuos4 6x9 Pen cursor              id=19   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ Power Button                              id=8    [slave  keyboard (3)]
    ↳ Microsoft Wired Keyboard 600              id=9    [slave  keyboard (3)]
    ↳ Huion Q11K Keyboard                       id=16   [slave  keyboard (3)]
    ↳ Microsoft Wired Keyboard 600              id=17   [slave  keyboard (3)]
    ↳ A4TECH USB Device                         id=20   [slave  keyboard (3)]

konachan700, 2017-10-13:
>>Wacom Intuos4 6x9 Pen eraser id=18 [slave pointer (2)]
may be, try to disconnect a wacom...

mifth, 2017-10-13:
You were right. I disconnected the Wacom and Blender now gets the Huion pressure!!!! I love you man!!!

[image]

mifth, 2017-10-13:
The only program which does not recognize the Huion is 3DCoat.
[image]

Wacom and my old G-Pen works ok with 3DCoat.
Could you, please check it?
You can download the 3dcoat here http://3dcoat.com/forum/index.php?/topic/21433-3d-coat-48-beta-testing-thread/

Or this is 3DCoat has a problem?

mifth, 2017-10-13:
Could you, please say is it possible to add RightClick and MiddleClick for the pen?
I always use these keys with the pen to move/rotate 2d/3d canvas.

konachan700, 2017-10-13:
Problem with driver. It use a wacom emulation =) and wacom xorg driver, but tablet not wacom and may be to send partitally incomplete\incorrect data to userspace.

> add RightClick and MiddleClick for the pen
 reassign a keys in the software is not will be a more simple way?

mifth, 2017-10-13:
"reassign a keys in the software is not will be a more simple way?"

I reassign nothing nothing with wacom and g-pen. Their pen buttons are recognized. It will be pain to reassign keys for all software. ((

mifth, 2017-10-13:
I think my G-pen uses wacom xorg driver too.

mifth, 2017-10-17:
Hi.
@spbnick i tried to record all things yesterday but i failed.
usbhid-dump hangs all my usb devices ((
I don't know what to do. I'm not so advanced user. ((

spbnick, 2017-10-17:
@mifth, `usbhid-dump` takes over all your devices, because you don't tell it which one to dump. Run it this way to dump only one device and prevent it from taking over: `sudo usbhid-dump -m 256c:006e -es`. Write if you need more help with getting this recording.

mifth, 2017-10-17:
Thank you for the reply. It helped! )

[q11k_capture_linux.zip](https://github.com/DIGImend/digimend-kernel-drivers/files/1390445/q11k_capture_linux.zip)
Here is my capture.
The only thing is I don't know how to filter in wireshark. I just selected usbmon3.
[image]

spbnick, 2017-10-18:
Thank you, that seems to be enough. It looks like the tablet is getting the old initialization sequence after usbhid-dump stops, after all, so I'll try to update the driver with the new initialization sequence and we'll see how it works. Will post here when done.

mifth, 2017-10-18:
Thank you too! I'll wait for sure.

spbnick, 2017-11-11:
Sorry for the delay, we just moved house and I won't be able to resume work on this for a few more weekends.

mifth, 2017-11-11:
No problem man. Solve your life issues. We will wait!

ashwanipirate11, 2017-11-11:
just download win 7 pirated* and it works perfectly.... i had problems with
linux so i did the same

[45 more replies]