DIGImend/digimend-kernel-drivers #77: Huion New 1060 PLUS not working

jeremyawhitman, 2017-10-01:
I picked up a new Huion 1060 plus and can't get it to work basically at all. It shows up in xinput as such:
```

⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Razer Razer BlackWidow Ultimate         	id=9	[slave  pointer  (2)]
⎜   ↳ Razer Razer BlackWidow Ultimate         	id=10	[slave  pointer  (2)]
⎜   ↳ Microsoft Microsoft® Nano Transceiver v1.0	id=12	[slave  pointer  (2)]
⎜   ↳ Microsoft Microsoft® Nano Transceiver v1.0	id=13	[slave  pointer  (2)]
**⎜   ↳ HID 256c:006e Pad                       	id=15	[slave  pointer  (2)]**
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Razer Razer BlackWidow Ultimate         	id=8	[slave  keyboard (3)]
    ↳ Microsoft Microsoft® Nano Transceiver v1.0	id=11	[slave  keyboard (3)]
    ↳ Razer Razer BlackWidow Ultimate         	id=16	[slave  keyboard (3)]
    ↳ Microsoft Microsoft® Nano Transceiver v1.0	id=17	[slave  keyboard (3)]
    ↳ Microsoft Microsoft® Nano Transceiver v1.0	id=18	[slave  keyboard (3)]
    **↳ HID 256c:006e Pen                       	id=14	[slave  keyboard (3)]**
```
and half shows in lsusb, along with its flash drive:

```
**Bus 001 Device 025: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader**
**Bus 001 Device 024: ID 256c:006e**  
**Bus 001 Device 023: ID 058f:6254 Alcor Micro Corp. USB Hub**
Bus 001 Device 004: ID 0409:005a NEC Corp. HighSpeed Hub
Bus 001 Device 003: ID 0409:005a NEC Corp. HighSpeed Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 003: ID 045e:0745 Microsoft Corp. Nano Transceiver v1.0 for Bluetooth
Bus 002 Device 002: ID 1532:011a Razer USA, Ltd 
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```
I can get it to prove it's taking input as such:

```
$ sudo usbhid-dump -es -m 256c:006e
Starting dumping interrupt transfer stream
with 1 minute timeout.

001:024:000:STREAM             1506893800.951797
 08 80 2B 5F A2 31 00 00 00 00 00 00

001:024:000:STREAM             1506893800.953801
 08 81 2B 5F A2 31 79 00 00 00 00 00

001:024:000:STREAM             1506893800.957769
 08 81 2B 5F A2 31 4A 01 00 00 00 00

001:024:000:STREAM             1506893800.961769
 08 81 2B 5F A2 31 E8 01 00 00 00 00

001:024:000:STREAM             1506893800.965768
 08 81 2B 5F A2 31 78 02 00 00 00 00

001:024:000:STREAM             1506893800.967762
 08 81 2B 5F A2 31 FA 02 00 00 00 00

```

But, I can't get it to give any input back to EV test, or  when I cat the events file in /sys/kernel/debug/hid/0003:256C:006E.001A or /sys/kernel/debug/hid/0003:256C:006E.001B after loggin in as root. 

The rdesc files for A and B are 

a: cat rdesc 
```
05 0d 09 02 a1 01 85 0a 09 20 a1 00 09 42 09 44 09 45 09 3c 09 43 09 44 15 00 25 01 75 01 95 06 81 02 09 32 75 01 95 01 81 02 81 03 05 01 09 30 09 31 55 0d 65 33 26 ff 7f 35 00 46 00 08 75 10 95 02 81 02 05 0d 09 30 26 ff 07 75 10 95 01 81 02 c0 c0 

  INPUT(10)[INPUT]
    Field(0)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(6)
        Digitizers.TipSwitch
        Digitizers.BarrelSwitch
        Digitizers.Eraser
        Digitizers.Invert
        Digitizers.SecondaryTipSwitch
        Digitizers.BarrelSwitch
      Logical Minimum(0)
      Logical Maximum(1)
      Report Size(1)
      Report Count(6)
      Report Offset(0)
      Flags( Variable Absolute )
    Field(1)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        Digitizers.InRange
      Logical Minimum(0)
      Logical Maximum(1)
      Report Size(1)
      Report Count(1)
      Report Offset(6)
      Flags( Variable Absolute )
    Field(2)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(2)
        GenericDesktop.X
        GenericDesktop.Y
      Logical Minimum(0)
      Logical Maximum(32767)
      Physical Minimum(0)
      Physical Maximum(2048)
      Unit Exponent(-3)
      Unit(English Linear : Inch^3)
      Report Size(16)
      Report Count(2)
      Report Offset(8)
      Flags( Variable Absolute )
    Field(3)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        Digitizers.TipPressure
      Logical Minimum(0)
      Logical Maximum(2047)
      Physical Minimum(0)
      Physical Maximum(2048)
      Unit Exponent(-3)
      Unit(English Linear : Inch^3)
      Report Size(16)
      Report Count(1)
      Report Offset(40)
      Flags( Variable Absolute )

Digitizers.TipSwitch ---> Sync.Report
Digitizers.BarrelSwitch ---> Sync.Report
Digitizers.Eraser ---> Sync.Report
Digitizers.Invert ---> Sync.Report
Digitizers.SecondaryTipSwitch ---> Sync.Report
Digitizers.BarrelSwitch ---> Sync.Report
Digitizers.InRange ---> Sync.Report
GenericDesktop.X ---> Sync.Report
GenericDesktop.Y ---> Sync.Report
Digitizers.TipPressure ---> Sync.Report

```

and b: 
```

05 0d 09 02 a1 01 85 07 09 20 a0 14 25 01 75 01 09 42 09 44 09 46 95 03 81 02 95 03 81 03 09 32 95 01 81 02 95 01 81 03 75 10 95 01 a4 05 01 65 13 55 fd 34 09 30 27 40 9c 00 00 47 10 27 00 00 81 02 09 31 27 a8 61 00 00 47 6a 18 00 00 81 02 b4 09 30 27 ff 1f 00 00 81 02 c0 c0 05 01 09 07 a1 01 85 f7 05 0d 09 39 a0 05 09 75 01 95 18 81 03 19 01 29 08 95 08 81 02 c0 c0 

  INPUT(7)[INPUT]
    Field(0)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(3)
        Digitizers.TipSwitch
        Digitizers.BarrelSwitch
        Digitizers.TabletPick
      Logical Minimum(0)
      Logical Maximum(1)
      Report Size(1)
      Report Count(3)
      Report Offset(0)
      Flags( Variable Absolute )
    Field(1)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        Digitizers.InRange
      Logical Minimum(0)
      Logical Maximum(1)
      Report Size(1)
      Report Count(1)
      Report Offset(6)
      Flags( Variable Absolute )
    Field(2)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        GenericDesktop.X
      Logical Minimum(0)
      Logical Maximum(40000)
      Physical Minimum(0)
      Physical Maximum(10000)
      Unit Exponent(-3)
      Unit(English Linear : Inch)
      Report Size(16)
      Report Count(1)
      Report Offset(8)
      Flags( Variable Absolute )
    Field(3)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        GenericDesktop.Y
      Logical Minimum(0)
      Logical Maximum(25000)
      Physical Minimum(0)
      Physical Maximum(6250)
      Unit Exponent(-3)
      Unit(English Linear : Inch)
      Report Size(16)
      Report Count(1)
      Report Offset(24)
      Flags( Variable Absolute )
    Field(4)
      Physical(Digitizers.Stylus)
      Application(Digitizers.Pen)
      Usage(1)
        Digitizers.TipPressure
      Logical Minimum(0)
      Logical Maximum(8191)
      Report Size(16)
      Report Count(1)
      Report Offset(40)
      Flags( Variable Absolute )
  INPUT(247)[INPUT]
    Field(0)
      Physical(Digitizers.TabletFunctionKey)
      Application(GenericDesktop.Keypad)
      Usage(8)
        Button.0001
        Button.0002
        Button.0003
        Button.0004
        Button.0005
        Button.0006
        Button.0007
        Button.0008
      Logical Minimum(0)
      Logical Maximum(8191)
      Report Size(1)
      Report Count(8)
      Report Offset(24)
      Flags( Variable Absolute )

Digitizers.TipSwitch ---> Key.Touch
Digitizers.BarrelSwitch ---> Key.Stylus
Digitizers.TabletPick ---> Key.Stylus2
Digitizers.InRange ---> Key.ToolPen
GenericDesktop.X ---> Absolute.X
GenericDesktop.Y ---> Absolute.Y
Digitizers.TipPressure ---> Absolute.Pressure
Button.0001 ---> Key.Btn0
Button.0002 ---> Key.Btn1
Button.0003 ---> Key.Btn2
Button.0004 ---> Key.Btn3
Button.0005 ---> Key.Btn4
Button.0006 ---> Key.Btn5
Button.0007 ---> Key.Btn6
Button.0008 ---> Key.Btn7

```

mifth, 2017-10-02:
It looks like the same issue as there:
https://github.com/DIGImend/digimend-kernel-drivers/issues/76
https://github.com/DIGImend/digimend-kernel-drivers/issues/59

spbnick, 2017-10-02:
Hi Jeremy,

You don't see any input, because the tablet is sending reports with IDs different from the report descriptor, most likely. What's interesting is that it is already sending report IDs which we are only trying to enable for Huion Inspiroy Q11K in #76.

Anyway, could you please send the output of `dmesg` after you plug the tablet in? Also, please download, build and run `uclogic-probe` from the latest [uclogic-tools master](https://github.com/DIGImend/uclogic-tools/archive/master.zip) on your tablet, and post your output here.

Thank you.

jeremyawhitman, 2017-10-02:
[dmesg](https://pastebin.com/mtjNcWkS) without tablet 
[dmesg](https://pastebin.com/SkEWLVYz) with tablet (did and unplug and replug as it was complaining about the usb drive endlessly) 

uclogic-probe: 

```
10/02/17 13:11:32 $ lsusb
Bus 001 Device 012: ID 058f:6366 Alcor Micro Corp. Multi Flash Reader
Bus 001 Device 011: ID 256c:006e  
Bus 001 Device 010: ID 058f:6254 Alcor Micro Corp. USB Hub
Bus 001 Device 006: ID 045e:0745 Microsoft Corp. Nano Transceiver v1.0 for Bluetooth
Bus 001 Device 005: ID 0409:005a NEC Corp. HighSpeed Hub
Bus 001 Device 004: ID 0409:005a NEC Corp. HighSpeed Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 003: ID 045e:0800 Microsoft Corp. 
Bus 002 Device 002: ID 1532:011a Razer USA, Ltd 
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

jeremy@desktop:~ [2.4.2p198]
10/02/17 13:11:49 $ sudo uclogic-probe 1 11
Failed to get manufacturer string descriptor: Pipe error

```

spbnick, 2017-10-03:
Thank you, Jeremy.

I see there's something wrong with tablet initialization. Could you please download, build, and install [this latest development version of the driver](https://github.com/DIGImend/digimend-kernel-drivers/archive/master.zip), and then please post the output of `/sbin/modinfo hid-uclogic` and `dmesg` after plugging in the tablet.

Also please download, build, and install [this latest development version of uclogic-tools](https://github.com/DIGImend/uclogic-tools/archive/master.zip), and then run it for your tablet, and post results here. You seem to be using an older version in the output you posted above.

Thank you.

jeremyawhitman, 2017-10-03:
```
sudo uclogic-probe 01 011
[sudo] password for jeremy: 
Failed to get manufacturer string descriptor: Pipe error
Failed to get product string descriptor: Pipe error
S 64 0E 03 40 9C A8 61 03 00 FF 1F A0 0F 08 00
S 65 04 03 20 A0
S 6E 04 03 00 30
S 79 14 03 48 00 41 00 36 00 30 00 2D 00 46 00 34 00 30 00 30 00
S 7A 08 03 01 0C 00 00 00 00
S 7B 0C 03 48 00 4B 00 20 00 4F 00 6E 00
S C8 12 03 70 C6 00 06 7C 00 FF 1F D8 13 03 0C 10 00 01 00
S C9 24 03 48 00 55 00 49 00 4F 00 4E 00 5F 00 54 00 31 00 37 00 34 00 5F 00 31 00 37 00 30 00 35 00 31 00 39 00
S CA 46 03 48 00 55 00 49 00 4F 00 4E 00 20 00 41 00 6E 00 69 00 6D 00 61 00 74 00 69 00 6F 00 6E 00 20 00 54 00 65 00 63 00 68 00 6E 00 6F 00 6C 00 6F 00 67 00 79 00 20 00 43 00 6F 00 2E 00 2C 00 6C 00 74 00 64 00
```

/sbin/modinfo hid-uclogic

```
filename:       /lib/modules/4.13.3-1-ARCH/extra/hid-uclogic.ko.gz
version:        7
license:        GPL
author:         Nikolai Kondrashov
author:         Martin Rusko
srcversion:     D7014AC838FC65F6579FF69
alias:          hid:b0003g*v000028BDp00000075
alias:          hid:b0003g*v000028BDp00000071
alias:          hid:b0003g*v00002179p00000053
alias:          hid:b0003g*v00005543p00003031
alias:          hid:b0003g*v00005543p00000045
alias:          hid:b0003g*v00005543p00000081
alias:          hid:b0003g*v00005543p0000004D
alias:          hid:b0003g*v00005543p0000006E
alias:          hid:b0003g*v0000256Cp0000006E
alias:          hid:b0003g*v00005543p00000781
alias:          hid:b0003g*v00005543p00000522
alias:          hid:b0003g*v00005543p00000064
alias:          hid:b0003g*v00005543p00000005
alias:          hid:b0003g*v00005543p00000004
alias:          hid:b0003g*v00005543p00000003
alias:          hid:b0003g*v00005543p00000042
depends:        hid,usbcore
name:           hid_uclogic
vermagic:       4.13.3-1-ARCH SMP preempt mod_unload modversions 

```
```
[    0.000000] random: get_random_bytes called from start_kernel+0x42/0x4b7 with crng_init=0
[    0.000000] Linux version 4.13.3-1-ARCH (builduser@tobias) (gcc version 7.2.0 (GCC)) #1 SMP PREEMPT Thu Sep 21 20:33:16 CEST 2017
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-linux root=/dev/mapper/AntergosVG-AntergosRoot rw quiet resume=UUID=00a0c7b4-b11d-45ac-ad9d-3c9878c71553
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ebff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000dfe0abff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dfe0ac00-0x00000000dfe5cbff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dfe5cc00-0x00000000dfe5ebff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dfe5ec00-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffb00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000081fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] random: fast init done
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: Dell Inc. Precision WorkStation T5400  /0RW203, BIOS A11 04/30/2012
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x820000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[... 1894 lines cut ...]
[40565.229750] usb 1-6.1.1: input irq status -75 received
[40565.245771] usb 1-6.1.1: input irq status -75 received
[40565.259732] usb 1-6.1.1: input irq status -75 received
[40565.273747] usb 1-6.1.1: input irq status -75 received
[40565.289723] usb 1-6.1.1: input irq status -75 received
[40565.303753] usb 1-6.1.1: input irq status -75 received
[40565.317741] usb 1-6.1.1: input irq status -75 received
[40565.333745] usb 1-6.1.1: input irq status -75 received
[40565.343740] usb 1-6.1.1: input irq status -75 received
[40565.351738] usb 1-6.1.1: input irq status -75 received
[40565.357729] usb 1-6.1.1: input irq status -75 received
[40565.365740] usb 1-6.1.1: input irq status -75 received
[40565.371747] usb 1-6.1.1: input irq status -75 received
[40565.379736] usb 1-6.1.1: input irq status -75 received
[40565.385735] usb 1-6.1.1: input irq status -75 received
[40565.393740] usb 1-6.1.1: input irq status -75 received
[40565.401724] usb 1-6.1.1: input irq status -75 received
[40565.407740] usb 1-6.1.1: input irq status -75 received
[40565.415731] usb 1-6.1.1: input irq status -75 received
[40565.421729] usb 1-6.1.1: input irq status -75 received
[40565.429728] usb 1-6.1.1: input irq status -75 received
[40565.435740] usb 1-6.1.1: input irq status -75 received
[40565.443728] usb 1-6.1.1: input irq status -75 received
[40565.449728] usb 1-6.1.1: input irq status -75 received
```

jeremyawhitman, 2017-10-03:
dmesg after reboot

    0.000000] random: get_random_bytes called from start_kernel+0x42/0x4b7 with crng_init=0
[    0.000000] Linux version 4.13.3-1-ARCH (builduser@tobias) (gcc version 7.2.0 (GCC)) #1 SMP PREEMPT Thu Sep 21 20:33:16 CEST 2017
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-linux root=/dev/mapper/AntergosVG-AntergosRoot rw quiet resume=UUID=00a0c7b4-b11d-45ac-ad9d-3c9878c71553
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ebff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000dfe0abff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dfe0ac00-0x00000000dfe5cbff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dfe5cc00-0x00000000dfe5ebff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dfe5ec00-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffb00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000081fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] random: fast init done
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: Dell Inc. Precision WorkStation T5400  /0RW203, BIOS A11 04/30/2012
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x820000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-D3FFF write-protect
[    0.000000]   D4000-EFFFF uncachable
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0000000000 mask 2000000000 write-back
[    0.000000]   1 base 00DFF00000 mask 3FFFF00000 uncachable
[    0.000000]   2 base 00E0000000 mask 3FE0000000 uncachable
[    0.000000]   3 base 1FF0000000 mask 3FF0000000 uncachable
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] e820: update [mem 0xdff00000-0xffffffff] usable ==> reserved
[    0.000000] e820: last_pfn = 0xdfe0a max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x000fe710-0x000fe71f] mapped at [ffff9defc00fe710]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff9defc0098000] 98000 size 24576
[    0.000000] BRK [0x4e6d5e000, 0x4e6d5efff] PGTABLE
[    0.000000] BRK [0x4e6d5f000, 0x4e6d5ffff] PGTABLE
[    0.000000] BRK [0x4e6d60000, 0x4e6d60fff] PGTABLE
[    0.000000] BRK [0x4e6d61000, 0x4e6d61fff] PGTABLE
[    0.000000] BRK [0x4e6d62000, 0x4e6d62fff] PGTABLE
[    0.000000] BRK [0x4e6d63000, 0x4e6d63fff] PGTABLE
[    0.000000] BRK [0x4e6d64000, 0x4e6d64fff] PGTABLE
[    0.000000] BRK [0x4e6d65000, 0x4e6d65fff] PGTABLE
[    0.000000] BRK [0x4e6d66000, 0x4e6d66fff] PGTABLE
[    0.000000] RAMDISK: [mem 0x36815000-0x37401fff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000FEBF0 000024 (v02 DELL  )
[    0.000000] ACPI: XSDT 0x00000000000FCD4E 00006C (v01 DELL   B9K      00000015 ASL  00000061)
[    0.000000] ACPI: FACP 0x00000000000FCE7E 0000F4 (v03 DELL   B9K      00000015 ASL  00000061)
[    0.000000] ACPI: DSDT 0x00000000FFF65BCA 003A12 (v01 DELL   dt_ex    00001000 INTL 20050624)
[    0.000000] ACPI: FACS 0x00000000DFE0AC00 000040
[    0.000000] ACPI: FACS 0x00000000DFE0AC00 000040
[    0.000000] ACPI: SSDT 0x00000000FFF696FD 0000AC (v01 DELL   st_ex    00001000 INTL 20050624)
[    0.000000] ACPI: APIC 0x00000000000FCF72 0000AA (v01 DELL   B9K      00000015 ASL  00000061)
[    0.000000] ACPI: BOOT 0x00000000000FD01C 000028 (v01 DELL   B9K      00000015 ASL  00000061)
[    0.000000] ACPI: ASF! 0x00000000000FD044 000096 (v32 DELL   B9K      00000015 ASL  00000061)
[    0.000000] ACPI: MCFG 0x00000000000FD0DA 00003E (v01 DELL   B9K      00000015 ASL  00000061)
[    0.000000] ACPI: HPET 0x00000000000FD118 000038 (v01 DELL   B9K      00000015 ASL  00000061)
[    0.000000] ACPI: TCPA 0x00000000000FD374 000032 (v01 DELL   B9K      00000015 ASL  00000061)
[    0.000000] ACPI: DMAR 0x00000000000FD3A6 0000C8 (v01 DELL   B9K      00000015 ASL  00000061)
[... 1391 lines cut ...]
[   57.318228] usb 1-6.1-port4: Cannot enable. Maybe the USB cable is bad?
[   57.322475] usb 1-6.1-port4: cannot disable (err = -71)
[   57.326731] usb 1-6.1-port4: cannot reset (err = -71)
[   57.330963] usb 1-6.1-port4: cannot reset (err = -71)
[   57.335221] usb 1-6.1-port4: cannot reset (err = -71)
[   57.336594] usb 1-6.1: USB disconnect, device number 6
[   57.336597] usb 1-6.1.1: USB disconnect, device number 8
[   57.339467] usb 1-6.1-port4: cannot reset (err = -71)
[   57.373039] usb 1-6.1.4: USB disconnect, device number 9
[   65.495859] usb 1-6.1: new high-speed USB device number 10 using ehci-pci
[   65.610132] hub 1-6.1:1.0: USB hub found
[   65.610223] hub 1-6.1:1.0: 4 ports detected
[   65.906033] usb 1-6.1.1: new full-speed USB device number 11 using ehci-pci
[... cut ...]
[   66.063116] uclogic 0003:256C:006E.000D: hidraw10: USB HID v1.11 Device [HID 256c:006e] on usb-0000:00:1d.7-6.1.1/input1
[   66.152805] usb 1-6.1.4: new high-speed USB device number 12 using ehci-pci
[   66.268635] usb-storage 1-6.1.4:1.0: USB Mass Storage device detected
[   66.268834] scsi host8: usb-storage 1-6.1.4:1.0
[   67.290886] scsi 8:0:0:0: Direct-Access     Multi    Flash Reader     1.00 PQ: 0 ANSI: 0
[   67.291263] sd 8:0:0:0: Attached scsi generic sg4 type 0
[   67.292860] sd 8:0:0:0: [sde] Attached SCSI removable disk
[   82.125738] usb 1-6.1.1: input irq status -75 received

spbnick, 2017-10-03:
Thank you, this now is right, although the tablet behavior itself is confusing. E.g. it reports different resolution using two different methods. Still, we can probably figure out how to make it work. Please collect diagnostics following this HOWTO: http://digimend.github.io/support/howto/trbl/diagnostics/ but please run the new `uclogic-probe` before that and do not reconnect the tablet.

spbnick, 2017-10-03:
Please post the diagnostics here.

jeremyawhitman, 2017-10-03:
Ok, I'll have to do that tomorrow. It's pretty late here. I appreciate all your help very much.

jeremyawhitman, 2017-10-03:
Ok, All of that is now located [here](https://github.com/jeremyawhitman/Huion_New_1060_Plus_diagnostics10.17). 

- [lsusb](https://github.com/jeremyawhitman/Huion_New_1060_Plus_diagnostics10.17/blob/master/lsusb.txt)
- [uclogic](https://github.com/jeremyawhitman/Huion_New_1060_Plus_diagnostics10.17/blob/master/uclogic_probe.txt)
- [descriptors](https://github.com/jeremyawhitman/Huion_New_1060_Plus_diagnostics10.17/blob/master/descriptors.txt)
- [frame buttons](https://github.com/jeremyawhitman/Huion_New_1060_Plus_diagnostics10.17/blob/master/frame_buttons.txt)
- [frame scrolling](https://github.com/jeremyawhitman/Huion_New_1060_Plus_diagnostics10.17/blob/master/frame_wheel_srolling.txt)
- (hid report)[https://github.com/jeremyawhitman/Huion_New_1060_Plus_diagnostics10.17/blob/master/hid_report_descriptors.txt]
- (pen buttons)[https://github.com/jeremyawhitman/Huion_New_1060_Plus_diagnostics10.17/blob/master/pen_buttons.txt]
- (pen corrds)[https://github.com/jeremyawhitman/Huion_New_1060_Plus_diagnostics10.17/blob/master/pen_coords.txt]
- (pen pressure)[https://github.com/jeremyawhitman/Huion_New_1060_Plus_diagnostics10.17/blob/master/pen_pressure.txt]

spbnick, 2017-10-04:
Thank you, Jeremy. This should be enough to make a driver. However, that needs time and I'm not sure when I'll have it. If anybody wishes to do it, I'll be glad to provide some assistance.

Aside from the code itself, and HOWTOs on the site, I also recorded a bunch of videos working on two tablets, which might help: https://www.youtube.com/channel/UCioG-vCm2WB6nSz0Y0JTXoA/playlists
I will be recording more with time.

Lastly, I'm considering setting up a Patreon account to help finance my work, and if I get enough support I'll be able to dedicate more time to drivers.

For now, though, you'll have to wait.

jeremyawhitman, 2017-10-05:
You should definitely set up a Patrion. I'm sure many people would prefer to use Huion tablets on Linux, as they are sometimes half the price of the Wacom counterparts when you get into the high end and would support development.

spbnick, 2017-10-05:
Thanks, Jeremy, that's encouraging. Just need to think my rewards through.

jeremyawhitman, 2017-10-06:
I have no experience with patrion, but believe in a case like this people would be willing to support it without direct rewards. If you set it up I'll sign up and mention it on some reddits.

spbnick, 2017-10-06:
Perhaps you're right. Thanks, Jeremy.

RJQuiralta, 2017-10-06:
How much difference is there between the plus and the pro? my 1060 pro+ its been working fine since I got it more than a year ago, digimend drivers allowed to remap buttons as mentioned [here](https://github.com/DIGImend/digimend-kernel-drivers/issues/18#issuecomment-242892045)

@spbnick I'm not sure how much relation is between your work and that of the libinput team (if any), I've been using only libinput for all this year without a problem, no buttons remapping though, my thinking is that maybe the code used there could be of use for the digimend? Probably not much, but I think is worth the comment. : )

jeremyawhitman, 2017-10-09:
Huion sent me a link to their [Linux drivers](
https://www.dropbox.com/s/kd0sdj15irnlr9n/linux-3.10.2_HuionTablet.tar.gz?dl=0 ). 

Haven't had a chance to attempt to install it yet.

spbnick, 2017-10-10:
@RJQuiralta, I haven't seen much information on Huion 1060 pro+ yet. I suspect there is considerable difference, since it is working and the 1060 PLUS doesn't. Glad to hear it worked for you, BTW.

I know a few of the libinput guys, and I try to follow their work. However, libinput is a userspace library for abstracting various input devices, used both in X.org and Wayland, and I suspect possible to use for other purposes. It was also made to replace the xf86-input-evdev driver. As such, libinput is using whatever digimend-kernel-drivers and their upstream parts are providing from the kernel.

@jeremyawhitman, I would not recommend spending time on Huion "drivers". They never worked, and now they are giving you a link to a whole outdated kernel tree. You really don't want that.

They approached me a while ago with the interest of financing the driver work, but after a while disappeared. I might try reaching them again.

mifth, 2017-10-10:
Hi.
I have installed these drivers some time ago. They do not work with Huion q11k tablet. The only hope for linux is @spbnick .
@spbnick please, contact them again. As we need good and stable linux support. :)

RJQuiralta, 2017-10-15:
@spbnick I see now, about the libinput and how it works with what the kernel includes from your work, is a good thing as it brings basic functionality to devices like some of these tablets out of the box.

Btw I notice the 1060 pro and the 1060 plus (first version) are physically identical, the only difference is that they advertise extra shortcut buttons (12 instead of 8, although both have 12), there is a new model for the 1060 plus though, it looks a bit different from the other two, I guess the mechanics of it may be also different as you suggested and that that new model is the one in question here.

@jeremyawhitman then I'm guessing the steps I mentioned on the previous link don work for you either at all? (just to confirm we are talking about two completely different tablets)

jsprada, 2017-11-10:
Hi All, I just bought a Huion 1060 Plus (new) and am now in the same boat as you guys.

ralleal, 2017-11-23:
Exactly the same problem here with Huion 1060 Plus (new, just bought it) and Ubuntu 17.10.

ralleal, 2017-11-24:
Huium support said to me: "Kindly note that currently we don't provide official Linux driver, but we are contacting them to fix the compatible issue. You may pay attention to Facebook, if the new version of the linux of the linux driver comes, it will have the information from here. Thank you for your cooperaion and understanding."

mifth, 2017-11-24:
I had the same answer. This is just copy paste. :)
My Q11k still does not work too. ((

mhagger, 2017-12-02:
I spent a chunk of time last weekend and yesterday bumbling around, with the hope of getting a Huion 1060+ new (my daughter's birthday present) working. Let me say what I have found out (not much) and hopefully get some feedback about whether I'm even on the right track...

The raw data that I see agrees with what @jeremyawhitman reported [here](https://github.com/DIGImend/digimend-kernel-drivers/issues/77#issuecomment-333952967), except that one hid report descriptor differs in one byte (the one with asterisks around it below):

```
001:003:001:DESCRIPTOR         1512150058.858251
 05 0D 09 02 A1 01 85 0A 09 20 A1 00 09 42 09 44
 09 45 09 3C 09 43 09 44 15 00 25 01 75 01 95 06
 81 02 09 32 75 01 95 01 81 02 81 03 05 01 09 30
 09 31 55 0D 65 33 26 FF 7F 35 00 46 00 08 75 10
 95 02 81 02 05 0D 09 30 26 FF*1F*75 10 95 01 81
 02 C0 C0

001:003:000:DESCRIPTOR         1512150058.860057
 06 00 FF 09 01 A1 01 85 08 75 38 95 01 09 01 81
 02 C0
```

(This seems to correspond to "Logical Maximum" of 8191 instead of 2047 for "Tip Pressure", and indeed the tip pressure numbers reported by `usbhid-dump -m 256c:006e -es` seem to max out at 0x1fff on my device.)

I also note that I'm getting lots of kernel messages like this whenever generating events on the tablet:

```
Dec  1 12:34:45 michael kernel: [11162.508859] usb 1-2.1: input irq status -75 received
Dec  1 12:34:45 michael kernel: [11162.514858] usb 1-2.1: input irq status -75 received
Dec  1 12:34:45 michael kernel: [11162.522860] usb 1-2.1: input irq status -75 received
Dec  1 12:34:45 michael kernel: [11162.528860] usb 1-2.1: input irq status -75 received
[...]
```

Here is how I currently understand how things work and the scope of the challenge. Please correct me if I'm wrong.

* HI devices report the format of the events that they send via "report descriptors". These are in a binary format that seems to be documented [here](http://www.usb.org/developers/hidpage/HID1_11.pdf).
* The report descriptors, ideally, should make the device more or less self-documenting. But manufacturers don't necessarily ensure that their devices generate them correctly. A large part of the job of digimend is to report "fixed" report descriptors (?)
* The fixed report descriptors are consumed further downstream, by the xinput driver I guess, and used to pick apart the events from the device as, for example, pen position and pressure values. I'm still pretty hazy about this part.

I'm very encouraged (in fact, it's almost my only ray of hope) that the output of `sudo usbhid-dump -m 256c:006e -es` seems to properly reflect x, y, pressure, and all of the hardware buttons. This tells me that it should be possible to get the device working at some rudimentary level without having to worry about any additional device-specific initialization.

So as I understand it, the main task is to construct a report descriptor that reflects the true structure of the events, and then (somehow?) the events will magically be understood and things will work.

Though actually, I haven't yet grokked how the work is divided between the kernel driver and the X driver, so it could be that there is more work to get X to recognize the events even after the report descriptors are correct.

...more to come...

mhagger, 2017-12-02:
It seems that report descriptor "001:003:001:DESCRIPTOR" is the important one for this device, based mainly on the fact that the other one is very small. In "code" format (coverted using `hidrd-convert -i hex -o code original1.txt original1.c`), it looks like

```
0x05, 0x0D,         /*  Usage Page (Digitizer),                 */
0x09, 0x02,         /*  Usage (Pen),                            */
0xA1, 0x01,         /*  Collection (Application),               */
0x85, 0x0A,         /*      Report ID (10),                     */
0x09, 0x20,         /*      Usage (Stylus),                     */
0xA1, 0x00,         /*      Collection (Physical),              */
0x09, 0x42,         /*          Usage (Tip Switch),             */
0x09, 0x44,         /*          Usage (Barrel Switch),          */
0x09, 0x45,         /*          Usage (Eraser),                 */
0x09, 0x3C,         /*          Usage (Invert),                 */
0x09, 0x43,         /*          Usage (Secondary Tip Switch),   */
0x09, 0x44,         /*          Usage (Barrel Switch),          */
0x15, 0x00,         /*          Logical Minimum (0),            */
0x25, 0x01,         /*          Logical Maximum (1),            */
0x75, 0x01,         /*          Report Size (1),                */
0x95, 0x06,         /*          Report Count (6),               */
0x81, 0x02,         /*          Input (Variable),               */
0x09, 0x32,         /*          Usage (In Range),               */
0x75, 0x01,         /*          Report Size (1),                */
0x95, 0x01,         /*          Report Count (1),               */
0x81, 0x02,         /*          Input (Variable),               */
0x81, 0x03,         /*          Input (Constant, Variable),     */
0x05, 0x01,         /*          Usage Page (Desktop),           */
0x09, 0x30,         /*          Usage (X),                      */
0x09, 0x31,         /*          Usage (Y),                      */
0x55, 0x0D,         /*          Unit Exponent (13),             */
0x65, 0x33,         /*          Unit (Inch^3),                  */
0x26, 0xFF, 0x7F,   /*          Logical Maximum (32767),        */
0x35, 0x00,         /*          Physical Minimum (0),           */
0x46, 0x00, 0x08,   /*          Physical Maximum (2048),        */
0x75, 0x10,         /*          Report Size (16),               */
0x95, 0x02,         /*          Report Count (2),               */
0x81, 0x02,         /*          Input (Variable),               */
0x05, 0x0D,         /*          Usage Page (Digitizer),         */
0x09, 0x30,         /*          Usage (Tip Pressure),           */
0x26, 0xFF, 0x1F,   /*          Logical Maximum (8191),         */
0x75, 0x10,         /*          Report Size (16),               */
0x95, 0x01,         /*          Report Count (1),               */
0x81, 0x02,         /*          Input (Variable),               */
0xC0,               /*      End Collection,                     */
0xC0                /*  End Collection                          */
```

If I understand correctly, in an ideal world I could more or less copy-paste this as `static const __u8 huion_new_1060_plus_rdesc_template[]`, wire that up in `uclogic_probe()` to be passed to `uclogic_probe_tablet()`, and done. Needless to say, that doesn't work.

No matter what I've done so far, the only signs of life are:

1. `/var/log/messages` shows the device was plugged in. I added some `printk()`s so I'm sure that the expected branch of the code in `uclogic_probe()` is being executed. It also looks like X kindof recognizes the device:

    <details><summary>`/var/log/messages` output</summary><p>

    ```
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1997]: (II) config/udev: Adding input device HID 256c:006e Pen (/dev/input/mouse2)
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1997]: (II) No input driver specified, ignoring this device.
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1997]: (II) This device may have been added with another device file.
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1854]: (II) config/udev: Adding input device HID 256c:006e Pen (/dev/input/mouse2)
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1854]: (II) No input driver specified, ignoring this device.
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1854]: (II) This device may have been added with another device file.
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1854]: (II) config/udev: Adding input device HID 256c:006e Pad (/dev/input/js1)
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1854]: (II) No input driver specified, ignoring this device.
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1854]: (II) This device may have been added with another device file.
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1997]: (II) config/udev: Adding input device HID 256c:006e Pad (/dev/input/js1)
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1997]: (II) No input driver specified, ignoring this device.
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1997]: (II) This device may have been added with another device file.
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1997]: (II) config/udev: Adding input device HID 256c:006e Pen (/dev/input/mouse3)
    Dec  2 06:11:04 michael /usr/lib/gdm3/gdm-x-session[1997]: (II) No input driver specified, ignoring this device.
[... cut ...]
        ↳ Dell WMI hotkeys                          id=14   [slave  keyboard (3)]
    ```

In particular, I've never seen **any** output from `xinput test` for any of the ids 15, 16, 17, or 18.

I'm still not sure what part of the data pipeline is broken, but my working hypothesis is that I need to handcraft a report descriptor that matches the structure observed in the event stream emitted by `usbhid-dump -m 256c:006e -es`. But I'm not looking forward to reading the 97 page specification :scream:

Any feedback or tips would be very welcome!

mhagger, 2017-12-02:
Our pen button measurements are surprising to me. Looking at byte 1 of the `usbhid-dump -m 256c:006e -es` output, I see the following:

| pen height | button | value | pressure value |
| ----- | ----- | ----- | ----- |
| out of range | don't care | 0x80 | zero |
| in range, not touching | none | 0x80 | zero |
| in range, not touching | lower button | 0x82 | zero |
| in range, not touching | upper button | 0x84 | zero |
| touching | none | 0x81 | available |
| touching | lower button | 0x84 | zero |
| touching | upper button | 0x80 | zero |

What's strange to me is that (1) no pressure data is available when either button is pressed while the pen is touching the tablet, and (2) the meaning of the 0x02 and 0x04 bits seems to be reversed when the pen is touching the tablet.

The end effect is that I don't see a way to distinguish, using the data reported by the tablet, between

* "pen in range but not touching, no buttons pressed" vs. "pen touching, upper button pressed"
* "pen in range but not touching, upper button pressed" vs. "pen touching, lower button pressed"

It seems that means that you should only press the buttons while the pen is *not* touching the tablet, which surprises me. Wouldn't it have made a lot more sense for the last two rows of that table to return 0x83 and 0x85, respectively?

asbaklm, 2017-12-02:
Patreon Page here for pledges to Digimend project. also see comment:

https://github.com/DIGImend/digimend-kernel-drivers/issues/82#issuecomment-348677724

https://www.patreon.com/spbnick

mhagger, 2017-12-03:
I just became a patron and encourage others to do so, too.

My latest puzzle is that I added a `printk()` to `uclogic_raw_event()`, but it doesn't seem to ever get called. I assumed that it would get called for every event. It seems that the HUION will need something like that to separate motion events from tablet button events, given that both types of events have the same report id (0x08) but different data formats.

I've kindof figured out how the report descriptors work, and it seems to me that the one reported for the HUION is roughly correct for motion events, except that (if I understand correctly) it describes report id 0x0a whereas the actual events have report id 0x08.

My next plan is to try to figure out what the `UCLOGIC_PH()` macro is for and whether it is needed for the HUION. Does this tablet even report device parameters the way that `uclogic_probe_tablet()` expects?

asbaklm, 2017-12-03:
I wrote a message to Huion to request financial & technical assistance for Nikolai. I suggest that everybody participating on the DIGImend lists do the same, because ultimately the more these companies (Huion and other manufacturers) are subjected to customer feedback requesting Linux support, the more likely it is that they will eventually cave in and do something about it.

Their email address appears to be here:

https://www.huiontablet.com/about-us/
service@huiontablet.com

Here is an example of what I wrote to them, however I would suggest that you don't copy and paste this text and that you re-write it in your own words.

> Hello dear Huion
> 
> I am a Linux user who bought your very excellent NEW 1060 PLUS tablet
> in the hope that it would work with the DIGImend project drivers but
> unfortunately this model is not supported yet so the tablet does not
> work in Linux.
> 
> DIGImend tablet Linux driver Project
> https://github.com/DIGImend/digimend-kernel-drivers
> 
> List of supported models
> https://digimend.github.io/tablets/
> 
> 
> The developer Nikolai Kondrashov requires some financial and technical
> support from companies such as Huion to develop additional drivers for
> new tablet models. He already has a proven track record in providing
> Linux drivers for older tablet devices, including a number of older
> model Huion tablets.
> 
> More people around the world are switching to Linux to free themselves
> from vendor lock-in by large IT Corporations such as Apple & Microsoft.
> The list of amazing and capable software packages available on Linux
> continues to grow. Unfortunately the situation with hardware device
> support is not yet as good as we hope it to be.
> 
> Would you therefore please consider also providing some financial and
> technical assistance and cooperation to Nikolai, so that he can make
> more of your excellent new products available to Linux users (and Huion
> customers) around the world?
>

spbnick, 2017-12-03:
@mhagger, thank you for your support!

First of all, you're on the right track, yes. Here are some responses to your comments:

> I also note that I'm getting lots of kernel messages like this whenever generating events on the tablet:

>     Dec  1 12:34:45 michael kernel: [11162.508859] usb 1-2.1: input irq status -75 received

This might be a hardware issue, or a protocol mismatch. Try different USB port or cable first. I'll need to dig into the kernel code and the USB spec to answer properly.

> A large part of the job of digimend is to report "fixed" report descriptors (?)

Not really, it's just easier to simply write a suitable report descriptor instead of writing code parsing the input reports, so that's what we mostly do, adding C code where necessary only. Lately it's been getting more hairy, though.

> The fixed report descriptors are consumed further downstream, by the xinput driver I guess, and used to pick apart the events from the device as, for example, pen position and pressure values. I'm still pretty hazy about this part.

The report descriptors are processed in the kernel by the hid-core driver, which runs the specific hid drivers. See `drivers/hid/hid-core.c`, `drivers/hid/hid-input.c` and `drivers/hid/usbhid` (at least). See also `/sys/kernel/debug/hid`.

> So as I understand it, the main task is to construct a report descriptor that reflects the true structure of the events, and then (somehow?) the events will magically be understood and things will work.

It often is sufficient, but IIRC these new tablets need initialization to report things properly. You might be able to make some sense from their reports in just the default mode, at least as the first step.

> Though actually, I haven't yet grokked how the work is divided between the kernel driver and the X driver, so it could be that there is more work to get X to recognize the events even after the report descriptors are correct.

The interface between the kernel and the X.org and Wayland drivers (nowadays both are libinput-based) is called "evdev". See `/usr/include/linux/input.h` and `/usr/include/linux/input-event-codes.h`. The specific ways tablets use that interface is yet another matter, though.

> It also looks like X kindof recognizes the device

The input drivers are trying to be very tolerant to the devices as there are so many of them, and they vary in so many ways. So they'll try to handle anything and make sense of it, which works in many cases, but far from all.

> Doesn't the 08 at the beginning of all of these events indicate the "Report ID"? If so, that's strange, because the descriptor reported by usbhid-dump seems to be for report ID 0x0a. I've tried hand-editing the report descriptor to change the report ID from 0x0a to 0x08, but it didn't seem to help.

Report descriptors often include report IDs that only appear in actual reports after certain initialization/request is done to the tablet, or are dedicated to feature reports, which are used to send the data *to* the device, or are simply there for legacy reasons. Our job is to find what the device actually reports, or what we can make it report, and to write a corresponding report descriptor.

> In particular, I've never seen any output from xinput test for any of the ids 15, 16, 17, or 18.
> I'm still not sure what part of the data pipeline is broken, but my working hypothesis is that I need [...]

Basically, if you have HID_QUIRK_MULTI_INPUT set for your HID driver (and it is set for all devices in hid-uclogic.c, IIRC), then you get an input device for each report ID in the descriptor, unless they're suppressed in the driver somehow (we do some of that in hid-uclogic.c).

> What's strange to me is that (1) no pressure data is available when either button is pressed while the pen is touching the tablet, and (2) the meaning of the 0x02 and 0x04 bits seems to be reversed when the pen is touching the tablet.

Welcome to the wonderful world of graphics tablets :)! Seriously, though, from quickly going through your description, this one seems to be extra messed-up.

> My latest puzzle is that I added a printk() to uclogic_raw_event(), but it doesn't seem to ever get called. I assumed that it would get called for every event.

Check out `raw_event` documentation in `include/linux/hid.h` in the kernel, that might help. I would expect it to be called all the time too. Also read the HID core sources I mentioned above, where they call `raw_event`.

> I've kindof figured out how the report descriptors work, and it seems to me that the one reported for the HUION is roughly correct for motion events, except that (if I understand correctly) it describes report id 0x0a whereas the actual events have report id 0x08.

Now, this, and all the other problems you're having trying to make sense of what the tabelt reports might just get solved once we figure out how to reliably enable the fully-functional mode on those "new" tablets. Checkout #76 for my efforts so far.

BTW, It would be great if you could contribute information on your tablet to the [tablets](https://github.com/DIGImend/tablets) repo, which is also used to publish the tablet database on the website.

spbnick, 2017-12-03:
@asbaklm, thank you for reaching out to Huion and for encouraging others! I don't have the time to do all the things this projects has to do and I appreciate all the help you're giving.

spbnick, 2017-12-03:
>     Dec  1 12:34:45 michael kernel: [11162.508859] usb 1-2.1: input irq status -75 received

Right, @mhagger, this is because the kernel requests reports with different size than the device expects. Check what it outputs with usbhid-dump, and compare it to what your report descriptor amounts to. They sizes should match.

mhagger, 2017-12-03:
> Right, @mhagger, this is because the kernel requests reports with different size than the device expects. Check what it outputs with usbhid-dump, and compare it to what your report descriptor amounts to. The sizes should match.

Ahh that was a key tip :heart: I had naively assumed that the useless bytes at the end of the events didn't need to be accounted for in the report descriptor.

I fixed the size, and now I'm seeing at least some xinput events, and also I'm seeing that `uclogic_raw_event()` is being called as expected. Apparently the fact that the size was incorrect was causing something upstream of the uclogic driver to drop the events entirely.

spbnick, 2017-12-03:
@mhagger, I just made the same mistake while recording my work on another tablet today, and remembered your case :)

mhagger, 2017-12-04:
It's a bit off-topic, but I noticed that [this comment](https://github.com/torvalds/linux/blob/ae64f9bd1d3621b5e60d7363bc20afb46aede215/include/linux/hid.h#L694-L695) seems no longer to be true after torvalds/linux@b1a1442a23776756b254b69786848a94d92445ba. After that commit, `ret == 1` is treated identically to `ret == 0`. I'm surprised that hasn't broken a bunch of stuff.

spbnick, 2017-12-04:
@mhagger, to me the [use of raw_event in the latest code](https://github.com/torvalds/linux/blob/ae64f9bd1d3621b5e60d7363bc20afb46aede215/drivers/hid/hid-core.c#L1586-L1590) looks like anything >=0 is simply ignored. Either that comment is overdue for update, or I'm missing something.

mhagger, 2017-12-04:
I'd say that if the `raw_event` callback returns 1, then control falls through to the call to `hid_report_raw_event()`, the same as if the callback had returned zero. This leaves no way for the `raw_event` callback to tell hid_core that an event should be dropped, which seems like a regression. But presumably that commit was hoping to fix some other problem, so switching back to the old behavior is not a no-brainer, either.

spbnick, 2017-12-04:
Perhaps we need to ask Jiri Kosina on linux-input@vger.kernel.org about this.

mhagger, 2017-12-04:
Regarding Huion support, I now have crude support for x, y, pressure, and at least one of the pen buttons working. But it's fragile; every time I click one of the tablet buttons, X gets in some weird state where all input events start behaving weirdly until I restart X.
[image]

I just pushed the state of my code to [the `huion-new-1060-plus-experiments` branch in my fork](https://github.com/mhagger/digimend-kernel-drivers/tree/huion-new-1060-plus-experiments) in case anybody else wants to play along. This is really just for playing; it's still quite broken and it also breaks the support for some other existing Huion tablet since they use the same vendor and product IDs.

asbaklm, 2017-12-04:
Well done mhagger

unnamed81, 2017-12-09:
Hi people
I just got the Huion New 1060 plus, mhagger i tried your experimental module and like you said it sort of works but X weirds out until relogin, but i found if you set the driver options in X conf (52-tablet.conf) to wacom X seems to behave its self, its also creates 4 devices 3 for the pen and 1 for the tablet.

 xxxx@xxxx:~> xsetwacom --list devices

HID 256c:006e Pen stylus                id: 9   type: STYLUS    
HID 256c:006e Pen eraser                id: 10  type: ERASER    
HID 256c:006e Pen pad                   id: 13  type: PAD       
HID 256c:006e Pad pad                   id: 14  type: PAD

I did muck around with the xsetwacom --get  and xsetwacom --set eg.
xsetwacom --get  -s "HID 256c:006e Pen stylus" all
xsetwacom set "HID 256c:006e Pen stylus" "PressureCurve" "0 0 100 100" (that is the default, i changed it to something else eg. "0 0 0 100")
they seemed to work, but someone who knows more about this stuff should give it a try,

mhagger, 2017-12-09:
I just pushed a couple more commits to my branch that make the pen work a little bit more stably.

> i found if you set the driver options in X conf (52-tablet.conf) to wacom X seems to behave its self, its also creates 4 devices 3 for the pen and 1 for the tablet.

@unnamed81 That's interesting. I haven't tried yet much mucking with the X driver and settings. If you could give step by step info that would help, because I don't know my way around x settings that well, either.

I haven't experimented very much, but the status quo is more or less as follows:

The pen position and the tip switch seem to work pretty reliably. This is enough that my daughter can start having some fun with her present :confetti_ball:

The two switches on the side of the pen pretty reliably do something, but what they do seems to depend on whether the pen is in contact with the surface or not (see https://github.com/DIGImend/digimend-kernel-drivers/issues/77#issuecomment-348677565). I suspect that the tablet might need some setup commands to make the pen buttons behave less ridiculously.

The tablet buttons misbehave badly. When I click one of the tablet buttons, an event is registered, but it usually causes the cursor to jump up to the top or top-left corner of the screen at the same time. Moreover, once I've pressed a tablet button, it seems that the pen stops working correctly (e.g., the tip button stops working) and even the normal mouse seems a bit weirded. I have to restart X to get things working again. I was going to disable the buttons temporarily for testing so that it doesn't matter if I hit them accidentally, but the problem I mentioned [here](https://github.com/DIGImend/digimend-kernel-drivers/issues/77#issuecomment-348893587) seems to preclude doing that in the obvious way.

**Request for help**

I don't have any copies of Windows or OSX around, so I can't try the Huion driver to see what kind of setup sequence it sends to the tablet. If somebody has the ability to do that and upload the results here, that might help a lot. It could very well be that the tablet doesn't function correctly if it hasn't been set up, in which case I'm wasting my time. One of Nikolai's videos shows how he uses wireshark to snoop on the connection between a windows VM and (a different) tablet.

unnamed81, 2017-12-10:
here is the wireshark usb capture, i think its ok, i compared it to another wireshark file in different thread and it looks similar.  hope it helps
btw. this was from a win7 VM.
[new1060plus.pcapng.tar.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/1545386/new1060plus.pcapng.tar.gz)

mhagger, 2017-12-10:
Thanks @unnamed81, I'll look at it later.

I just pushed some more commits. The tablet buttons now don't misbehave so much, though I haven't gotten them to do anything useful yet, either. `uclogic_raw_event()` was looking for the wrong report id so it wasn't doing anything.

mhagger, 2017-12-11:
Below is a wireshark capture of the traffic to the tablet when it is plugged in under Linux. This is my first time working with dumps like this, but ISTM that this one seems to cover the same bases as the Windows dump reported by @unnamed81 [here](https://github.com/DIGImend/digimend-kernel-drivers/issues/77#issuecomment-350530769). I mostly focused on the `SET CONFIGURATION` and `SET IDLE` requests, since the others seem to be the host querying the tablet as opposed to the host configuring the tablet.

@spbnick, if you have a moment to glance at these and see if anything jumps out at you, it would be much appreciated.

[usb-capture-linux.pcapng.gz](https://github.com/DIGImend/digimend-kernel-drivers/files/1546692/usb-capture-linux.pcapng.gz)

spbnick, 2017-12-11:
@mhagger No matter how weird, that's exactly what's happening: Those Huion tablets (and others supposedly based on similar or the same UC-Logic chipsets) are configured by *requesting* string descriptors. It was one string descriptor for a while, now those new tablets seem to be using another. When you request this string descriptor, you get response with tablet parameters, and you also enable the fully-functional mode. Check out the driver source for the old way, and  uclogic-tools source for both the old and (supposedly) the new way.

spbnick, 2017-12-11:
No need to mention, of course, that that is going kind of against the USB spec and what good engineers believe in, but that's how it works.

unnamed81, 2017-12-13:
i changed the report ID in __u8 huion_new_1060_plus_buttonpad_rdesc to 0x85, 0x0A, and it seemed to 
fix the X freezing thing, but i dont have the buttons on the tablet working now, but their not going wacko either like the were before, one step forward two back i guess.

spbnick, 2017-12-13:
@kolrabi has submitted #85 that goes some way towards enabling New 1060 PLUS support as well. Could you work together and figure out what's needed to get both of your tablets working?

spbnick, 2018-02-13:
Closing as a duplicate of #82. Reopen if you think this should still stay open.

closed 2018-02-13 by spbnick