DIGImend/digimend-kernel-drivers #78: HELP! Huion gt-191 pen display tablet, not getting it to work on Fedora 25

macpenguini, 2017-10-03:
Hello everyone, I have recently bought the huion kamvas gt-191 graphic pen display tablet. Not too long ago I started making my transition into the wonderful world of Linux using mainly Fedora and CentOs. I have managed thus far to get all my programs, hardware etc… working great, but for the life of me I am having no luck with the huion tablet, and my research as led me with a few tries and nothing! As I am still pretty new to the Linux world, I am not 100% sure what would be needed for one of you genius minds to help me, so I will share in here some of the troubleshooting I have done and let me know if any one can help, cause don´t make go into Windows to use my tablet cause every-time I need to go there tears fall down slowly as my soul is crushed!

I downloaded and built the DIGImend kernel drivers, using dkms from: https://github.com/DIGImend/digimend-kernel-drivers

Next I followed some of the troubleshooting suggestions found here: http://digimend.github.io/support/howto/trbl/locating_failure/

Everything seems to work according to the outputs I was getting but when I run: sudo uclogic-probe 3 9
I get this: Failed to get manufacturer string descriptor: Pipe error
I got this step from: https://github.com/DIGImend/uclogic-tools

I did notice when I boot into Fedora I get errors at start see below:

usbhid 5-2:1.2: couldn´t find an input interrup endpoint
uclogic 003:256C:006E.0003: device paramiters not found
uclogic 003:256C:006E.0003: tablet enabling failed
uclogic 003:256C:006E.0004: No inputs registered, leaving

Do let me know if you need any further info from my side and any help would oh so appreciated.

Thank you again,

MaC

spbnick, 2017-10-06:
Hi MaC, could you please try downloading, building and running the [latest development version of uclogic-tools](https://github.com/DIGImend/uclogic-tools/archive/master.zip) instead?

macpenguini, 2017-10-08:
Hi spbnick,

Thank you so much for getting back to me, I have downloaded and built the latest uclogic-tools. I know get the bellow outputs.

$sudo ./uclogic-probe 3 8
Failed to get manufacturer string descriptor: Pipe error
Failed to get product string descriptor: Pipe error
S C8 13 03 BA 53 01 88 BA 00 FF 1F D8 13 03 00 00 01 02 00 40
S C9 24 03 48 00 55 00 49 00 4F 00 4E 00 5F 00 4D 00 31 00 36 00 38 00 5F 00 31 00 37 00 30 00 35 00 32 00 30 00
S CA 46 03 48 00 55 00 49 00 4F 00 4E 00 20 00 41 00 6E 00 69 00 6D 00 61 00 74 00 69 00 6F 00 6E 00 20 00 54 00 65 00 63 00 68 00 6E 00 6F 00 6C 00 6F 00 67 00 79 00 20 00 43 00 6F 00 2E 00 2C 00 6C 00 74 00 64 00

$ sudo ./uclogic-probe 3 8 | uclogic-decode
Failed to get manufacturer string descriptor: Pipe error
Failed to get product string descriptor: Pipe error
                 Max X: 86970
                 Max Y: 47752
          Max pressure: 8191
            Resolution: 5080
     Unknown string #1: HUION_M168_170520
 Internal manufacturer: HUION Animation Technology Co.,ltd

Do let me know if you need anything else from my side.

Thank you gaian,

Mac

benthor, 2017-10-21:
Hi spbnick,

I've also recently bought a Huion GT-191 and can confirm macpenguini's findings.

 My **digimend-kernel-drivers are built from master** (81f8c71, Sep 30 2017).

After plugging it in, `dmesg` spits out the following:

> [20455.639991] usb 1-10: new full-speed USB device number 12 using xhci_hcd
[20455.771414] usb 1-10: config 1 has an invalid interface number: 1 but max is 0
[20455.771418] usb 1-10: config 1 has no interface number 0
[20455.774210] input: NUVOTON WPM USB as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:0416:A316.001A/input/input31
[20455.830188] hid-generic 0003:0416:A316.001A: input,hidraw3: USB HID v1.10 Device [NUVOTON WPM USB] on usb-0000:00:14.0-10/input1
[20459.015340] usb 1-10: USB disconnect, device number 12
[20459.336635] usb 1-10: new full-speed USB device number 13 using xhci_hcd
[20459.574667] uclogic 0003:256C:006E.001B: device parameters not found
[20459.574670] uclogic 0003:256C:006E.001B: tablet enabling failed
[20459.574785] uclogic 0003:256C:006E.001C: No inputs registered, leaving

`sudo ./uclogic-probe 1 13 | ./uclogic-decode`  (**uclogic-tools built from master** (87a7f6, Oct 14 2017)) gives me

>  c8 |        Params block #2: ????????
    |                          Max X: 86970
    |                          Max Y: 47752
    |                   Max pressure: 8191
    |                     Resolution: 5080
    |
 c9 |      Unknown string #1: HUION_M168_170727
 ca |  Internal manufacturer: HUION Animation Technology Co.,ltd

`usbhid-dump` shows me pointer, pressure and button events. I've uploaded the dumps as per the [instructions on the diagnostics page](https://digimend.github.io/support/howto/trbl/diagnostics/)

* [descriptors.txt](https://gist.github.com/benthor/34af075d8d80ea8db0c9269017f59b0e)
* [hid_report_descriptors.txt](https://gist.github.com/benthor/d292c66fcb179a2bba147d9edea72375)
* [pen_coords.txt](https://gist.github.com/benthor/7c29b7d2fe7f82b0a4616d58d4de52e9)
* [pen_pressure.txt](https://gist.github.com/benthor/8c6a7a6f6ed4a693125daa4dda245a68)
* [pen_buttons.txt](https://gist.github.com/benthor/8753ff7c3cd0689f9484fc488b405fa8)

I've had a look at the code to find out why the device parameters are not found. Seems to me the driver quits after -EPIPE is thrown in [line 922 of hid-uclogic.c](https://github.com/DIGImend/digimend-kernel-drivers/blob/master/hid-uclogic.c#L922). Am I correct in assuming that I'd need to somehow properly translate the descriptors I found for my new device into a new static constant in `hid-uclogic.c` to make this work?

benthor, 2017-10-22:
FYI: I wrote a very quick and dirty user space evdev module to support this device
 * [HuionKamvasGT191LinuxDriver](https://github.com/benthor/HuionKamvasGT191LinuxDriver)

insomniacUNDERSCORElemon, 2017-12-06:
Anything on this? Can't get my 191 to work on Arch Linux. Neither Digimend nor Benthor's user space driver works.

Worked oOtB with outdated Win7, although pen pressure came after installing the Huion driver. I don't want to need to reboot into Win7 though (besides none of my apps, it's also very slow).

TheJackiMonster, 2018-01-23:
I have setup a custom user space driver in C, which also gets the correct monitor automaticly via EDID through Xrandr and stuff. I am currently struggling with a correct calibration for the device because the cursor resolution is much higher than the display.

https://github.com/TheJackiMonster/HuionGT191LinuxDriver

aigoia, 2018-02-15:
i install huion kamvas gt-191 driver made by bentheor...
it works only well in script print... 
so i checked uclogic-probe... but... i showed this

aigoia@aigoia:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 256c:006e 
Bus 001 Device 004: ID 1a2c:2d23 China Resource Semico Co., Ltd 
Bus 001 Device 003: ID 056a:033b Wacom Co., Ltd 
Bus 001 Device 002: ID 056e:0107 Elecom Co., Ltd 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
aigoia@aigoia:~$ sudo uclogic-probe 1 5
[sudo] aigoia의 암호: 
Failed to get manufacturer string descriptor: Pipe error (ERROR_PIPE)

i think that 256c:006e is huion... it works well window 10... 
i dont go ahead any more... that point in ubuntu...

LiamKane, 2018-03-17:
@aigoia I get the same error fro uclogic-probe, if I run lsusb or uclogic-probe within ~3 seconds of plugging in the USB fromthe tablet it comes up under a different ID with a name.