DIGImend/digimend-kernel-drivers #79: Problem with pressing buttons

CaramelHeaven, 2017-10-11:
Hello everyone. I have a tablet UGEE M708 and I cannot setting buttons on Krita. How can this be remedied? The rest of the functions work. And before in Windows, I could rotate the tablet on 180 degrees - for lefties, but here I can't do this feature.

spbnick, 2017-10-13:
Hello,

Have you tried setting up your tablet with the Wacom driver for X.org? It support mapping the buttons and rotating the tablet relatively easily, using the xsetwacom tool.

CaramelHeaven, 2017-10-16:
Thanks for the answer. So, I have ubuntu 16.04. I tried enter command: xsetwacom --list devices - it's isn't working. Nothing outputting. Then I take lsusb - so I get `Bus 001 Device 007: ID 5543:0081 UC-Logic Technology Corp.` - probably it's my tablet. 
So, I downloaded digimend-kernel-drivers and installed:
`sudo apt-get install linux-headers-generic build-essential git`
`mkdir build`
`cd build`
`git clone https://github.com/DIGImend/digimend-kernel-drivers.git`
`cd digimend-kernel-drivers`
`make`
`sudo make install`
`cd`

Then, I reconnected my table and runed this commands to restart the drivers:

`sudo modprobe -r hid-kye`
`sudo modprobe -r hid-uclogic`
`sudo modprobe -r hid-huion`

I entered this:
`lsmod | grep hid`

This showed me:
`hid_kye                16384  0`
`hid_uclogic            16384  0`
`hid_polostar           16384  0`
`mac_hid                16384  0`
`hid_generic            16384  0`
`usbhid                 53248  0`
`i2c_hid                20480  0`
`hid                   118784  6 i2c_hid,hid_generic,usbhid,hid_polostar,hid_kye,hid_uclogic`

But when I tried to check again `xsetwacom --list` - also nothing outputting.

I checked command - `modinfo hid-uclogic` :
`filename:       /lib/modules/4.10.0-37-generic/extra/hid-uclogic.ko`
`version:        7`
`license:        GPL`
`author:         Nikolai Kondrashov`
`author:         Martin Rusko`
`srcversion:     D7014AC838FC65F6579FF69`
`alias:          hid:b0003g*v000028BDp00000075`
`alias:          hid:b0003g*v000028BDp00000071`
`alias:          hid:b0003g*v00002179p00000053`
`alias:          hid:b0003g*v00005543p00003031`
`alias:          hid:b0003g*v00005543p00000045`
`alias:          hid:b0003g*v00005543p00000081`
`alias:          hid:b0003g*v00005543p0000004D`
`alias:          hid:b0003g*v00005543p0000006E`
`alias:          hid:b0003g*v0000256Cp0000006E`
`alias:          hid:b0003g*v00005543p00000781`
`alias:          hid:b0003g*v00005543p00000522`
`alias:          hid:b0003g*v00005543p00000064`
`alias:          hid:b0003g*v00005543p00000005`
`alias:          hid:b0003g*v00005543p00000004`
`alias:          hid:b0003g*v00005543p00000003`
`alias:          hid:b0003g*v00005543p00000042`
`depends:        hid`
`vermagic:       4.10.0-37-generic SMP mod_unload`

spbnick, 2017-10-16:
You need to make X.org use the wacom driver for your tablet. Take a look here for the general idea: https://github.com/DIGImend/digimend-kernel-drivers/issues/26 and/or watch a live video recording where I configure Ugee EX07 to use the wacom driver (you'll have to skip quite a bit there, I expect).

CaramelHeaven, 2017-10-16:
Thanks for your reply. I will try.

CaramelHeaven, 2017-10-16:
Okay, I watched some tutorials and now I can pressing buttons on my tablet. I learned #26 there they did:
```
cd /etc/X11
sudo mkdir xorg.conf.d 
cd xorg.conf.d 
sudo gedit 52-tablet.conf 
```
But, I have this file in this catalog:` /usr/share/X11/xorg.conf.d/70-wacom.conf`, not in this  `/etc/X11/xorg.conf.d.` So, pressing the buttons works. But I can't override them in the application and in the consol.. And `xsetwacom --list` not working again. Nothing outputing:
```
li3@OS:~$ xsetwacom --list
li3@OS:~$ 
```
I tried to add some code from `#26`  in file 70-wacom.conf 
```
Section "InputClass"
        Identifier "Wacom USB tablet class"
        MatchProduct "UGEE"
        MatchDevicePath "/dev/input/event*"
        MatchIsTablet "on"
        Driver "wacom"
EndSection
```
But not working. In theory, it should show the tablet after the command `xsetwacom --list` but doesn't show. This file contains:
```
# Some of the below input classes appear 3x times, once for each of
# "tablet", "touchscreen", and "touchpad" to ensure that the Wacom
# driver is not accidentally bound to other types of hardware that
# Wacom has made which are not handled by the wacom driver (e.g the
# Wacom Bluetooth Keyboard)
#
# https://sourceforge.net/p/linuxwacom/bugs/294/

Section "InputClass"
        Identifier "Wacom USB tablet class"
        MatchUSBID "056a:*"
        MatchDevicePath "/dev/input/event*"
        MatchIsTablet "true"
        Driver "wacom"
EndSection

Section "InputClass"
        Identifier "Wacom USB touchscreen class"
        MatchUSBID "056a:*"
        MatchDevicePath "/dev/input/event*"
        MatchIsTouchscreen "true"
        Driver "wacom"
EndSection

Section "InputClass"
        Identifier "Wacom USB touchpad class"
        MatchUSBID "056a:*"
        MatchDevicePath "/dev/input/event*"
        MatchIsTouchpad "true"
        Driver "wacom"
EndSection

Section "InputClass"
	Identifier "Wacom tablet class"
	MatchProduct "Wacom|WACOM|PTK-540WL|ISD-V4"
	MatchDevicePath "/dev/input/event*"
	MatchIsTablet "true"
	Driver "wacom"
EndSection

Section "InputClass"
	Identifier "Wacom touchscreen class"
	MatchProduct "Wacom|WACOM|PTK-540WL|ISD-V4"
	MatchDevicePath "/dev/input/event*"
	MatchIsTouchscreen "true"
	Driver "wacom"
EndSection

Section "InputClass"
	Identifier "Wacom touchpad class"
	MatchProduct "Wacom|WACOM|PTK-540WL|ISD-V4"
	MatchDevicePath "/dev/input/event*"
	MatchIsTouchpad "true"
	Driver "wacom"
EndSection

# Serial Wacom devices should always be one of tablet, touchscreen, or
# touchpad so we can safely get away with just one match section in
# these cases
Section "InputClass"
        Identifier "Wacom PnP device class"
        MatchPnPID "WACf*|WCOM*|WACM*|FUJ02e5|FUJ02e7|FUJ02e9"
        MatchDevicePath "/dev/input/event*"
        Driver "wacom"
EndSection

Section "InputClass"
	Identifier "Wacom serial class"
	MatchProduct "Serial Wacom Tablet"
	Driver "wacom"
EndSection

Section "InputClass"
        Identifier "Wacom serial class identifiers"
        MatchProduct "WACf|FUJ02e5|FUJ02e7|FUJ02e9"
        Driver "wacom"
EndSection

# Hanwang tablets
Section "InputClass"
	Identifier "Hanwang class"
	MatchProduct "Hanwang"
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
EndSection

# Waltop tablets
Section "InputClass"
	Identifier "Waltop class"
	MatchProduct "WALTOP"
	MatchIsTablet "on"
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
EndSection

# N-Trig Duosense Electromagnetic Digitizer
Section "InputClass"
	Identifier "Wacom N-Trig class"
	MatchProduct "HID 1b96:0001|N-Trig Pen|N-Trig DuoSense"
	MatchDevicePath "/dev/input/event*"
	Driver "wacom"
	Option "Button2" "3"
EndSection

```

spbnick, 2017-10-16:
First, in general, you shouldn't touch `/usr/share/X11/xorg.conf.d/70-wacom.conf` - it's coming with the driver package. You need only to touch `/etc/X11/xorg.conf.d` - that directory is for local configuration. Next, I think you need to drop `MatchIsTablet` line, because you also need the Wacom driver to handle your buttons ("pad") device. Last, if that doesn't work, please post the output of `xinput list` and the contents of your `/var/log/Xorg.0.log` file after you plug the tablet in.

CaramelHeaven, 2017-10-16:
I tried to use #, but without success. 
```
Section "InputClass"
    Identifier "Ugee on wacom"
    # MatchIsTablet "on"
    MatchProduct "UGEE M708"
    MatchDevicePath "/dev/input/event*"
    Driver "wacom"
EndSection
```

Okay.
```
xinput list
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ PixArt Dell MS116 USB Optical Mouse       id=9    [slave  pointer  (2)]
⎜   ↳ MOSART Semi. 2.4G Keyboard Mouse          id=11   [slave  pointer  (2)]
⎜   ↳ UC-LOIC TABLET 1060 Pad                   id=15   [slave  pointer  (2)]
⎜   ↳ UC-LOIC TABLET 1060 Pen                   id=16   [slave  pointer  (2)]
⎜   ↳ UC-LOIC TABLET 1060 Mouse                 id=17   [slave  pointer  (2)]
⎜   ↳ UC-LOIC TABLET 1060 Consumer Control      id=19   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Power Button                              id=7    [slave  keyboard (3)]
    ↳ Sleep Button                              id=8    [slave  keyboard (3)]
    ↳ MOSART Semi. 2.4G Keyboard Mouse          id=10   [slave  keyboard (3)]
    ↳ Dell KB216 Wired Keyboard                 id=12   [slave  keyboard (3)]
    ↳ Dell KB216 Wired Keyboard                 id=13   [slave  keyboard (3)]
    ↳ UC-LOIC TABLET 1060 Keyboard              id=14   [slave  keyboard (3)]
    ↳ UC-LOIC TABLET 1060 System Control        id=18   [slave  keyboard (3)]
```
[Xorg.0.log](https://github.com/DIGImend/digimend-kernel-drivers/files/1387874/Xorg.0.log)

CaramelHeaven, 2017-10-16:
Okay, I added MathcUSBID in this:
```
Section "InputClass"
    Identifier "Ugee on wacom"
    MatchUSBID "5543:0081"
    MatchProduct "UGEE M708"
    MatchDevicePath "/dev/input/event*"
    Driver "wacom"
EndSection
```
So, when I rebooted OS, `xsetwacom --list` isn't show me devices again. Okay, I tried to use this command `lsusb -v | grep 0x0081` but is it  show me this:
```
lsusb -v | grep 0x0081
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
  idProduct          0x0081 
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
Couldn't open device, some information will be missing
```

spbnick, 2017-10-16:
Yes, you need to use `MatchUSBID` instead of `MatchProduct`. I.e. remove `MatchProduct` completely, leave `MatchUSBID` only, because the USB ID database is missing information on your tablet and its "product" string is empty.

spbnick, 2017-10-16:
Great, glad to hear you got it working :) No need to edit the configuration file, you can just [use xsetwacom for rotation](http://linuxwacom.sourceforge.net/wiki/index.php/Rotation).

CaramelHeaven, 2017-10-16:
So, I dont quite understand, should this ( on 180 )
```
    case "$rotation" in
    normal)
    # rotate to the left
    xrandr -o left
    xsetwacom set stylus rotate ccw
    xsetwacom set eraser rotate ccw
    xsetwacom set touch rotate ccw
    ;;
    left)
    # rotate to inverted
    xrandr -o inverted
    xsetwacom set stylus rotate half
    xsetwacom set eraser rotate half
    xsetwacom set touch rotate half
    ;;
```
in my file ` 52-tablet.conf` ? Right?

closed 2017-10-20 by CaramelHeaven